pyocd/pyOCD #1054: st-link not found after upgrade

nerdralph, 2021-01-17:
I tried using a st-link (part of a Nucleo board), and pyocd would list the adapter.  If I tried to connect, I got an error saying the firmware was too old.  I upgraded to the latest firmware (V2J37M27), and now pyocd can't even find the device.  openocd is still able to use the st-link.  This is on Win7E.
```
$ openocd -f stm32.cfg
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev (2020-10-13-17:29) Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 100 kHz
Info : STLINK V2J37M27 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.266403
Warn : UNEXPECTED idcode: 0x0bb11477
```

```
$ pyocd list
No available debug probes are connected
```

flit, 2021-01-17:
Thanks for reporting this. I'll try to reproduce it.

What version of pyocd—0.28.3?
Do you have libusb installed for the version of Python you are using? (I assume so since you said pyocd listed the STLink before the firmware upgrade.)

nerdralph, 2021-01-18:
Yes, I'm using 0.28.3.  And yes, libusb is installed.  pyocd not only recognized the STlink before the firmware upgrade, but it also communicates fine with the ch551/ch552 CMSIS-DAP adapter I've been working on.
https://github.com/nerdralph/ch554_sdcc/tree/master/examples/CMSIS_DAP
```
$ pyocd list
  #   Probe                 Unique ID
---------------------------------------
  0   WCH CH55x CMSIS-DAP   ?
```

flit, 2021-02-03:
Unfortunately, I'm not able to reproduce this on macOS or Win10 after updating STLinkV2-1 firmware to the same V2J37M27 version (on an STM32L475VG IOT01A board).

Would you mind running `pyocd cmd -vv`? That will include some debug log messages for looking up the device in Windows.

nerdralph, 2021-02-03:
I did some more digging on my end.  For Windoze, I'm using Win7E, which may explain the different behavior.  'pyocd cmd -vv' only output "Waiting for a debug probe to be connected...", which got me thinking pyocd isn't getting any results from libusb.  When I looked in the device manager, I saw that the upgrade removed the association between ST-Link and libusb.  Under Universal Serial Bus devices I could see: "STMicroelectronics STLink dongle".
I used Zadig to set the driver for "ST-Link Debug (Interface 0) " to libusb-win32, and now pyocd works fine.

closed 2021-02-03 by nerdralph

flit, 2021-02-03:
Now that I think about it, being on Win7 makes sense. STLink uses WinUSB to avoid requiring a kernel driver on Windows, but this is only supported on Win 8 and above.