pyocd/pyOCD #1056: No cores were discovered with LPC845

jpmzometa, 2021-01-20:
I'm getting the error 'No cores were discovered'. This happens after successfully flashing a valid program.
If I erase the chip (using MCUXpresso, the official IDE for LPC845), the pyocd is able to detect the cores again.

This might be related to issue #743 . But this an NXP LPC845, so the solution proposed there does not work here. 

```
pyocd erase -vv --chip -t lpc845 -O reset_type=hw
0000439:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=1bcf PID=2c94). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000441:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=05f3 PID=0007). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000441:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=145f PID=0263). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000442:DEBUG:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=1bcf PID=2c94). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000443:DEBUG:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=05f3 PID=0007). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000443:DEBUG:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=145f PID=0263). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000768:DEBUG:session:Project directory: /home/pablo/pyocd2
0000936:WARNING:cmsis_pack:Overlapping memory regions in file <zipfile.ZipFile filename='/home/pablo/.local/share/cmsis-pack-manager/NXP/LPC845_DFP/12.3.0.pack' mode='r'>, deleting outer region. Further warnings will be suppressed for this file.
0000956:INFO:board:Target type is lpc845
0000957:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=1bcf PID=2c94). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000959:DEBUG:pyusb_backend:Detaching Kernel Driver of Interface 0 from USB device (VID=1fc9 PID=0090).
0000965:DEBUG:dap_access_cmsis_dap:CMSIS-DAP probe I3F4M0GS firmware version: 1.10
0000971:DEBUG:sequencer:Running task load_svd
0001033:DEBUG:sequencer:Running task pre_connect
0001033:DEBUG:sequencer:Running task dp_init
0001033:DEBUG:sequencer:Running task lock_probe
0001033:DEBUG:sequencer:Running task get_probe_capabilities
0001038:DEBUG:sequencer:Running task connect
0001118:DEBUG:dap:Default wire protocol selected; using SWD
0001124:DEBUG:swj:Sending deprecated SWJ sequence to select SWD
0001134:INFO:dap:DP IDR = 0x0bc11477 (v1 MINDP rev0)
0001134:DEBUG:sequencer:Running task clear_sticky_err
0001134:DEBUG:sequencer:Running task power_up_debug
0001135:DEBUG:sequencer:Running task check_version
0001135:DEBUG:sequencer:Running task unlock_probe
0001135:DEBUG:sequencer:Running task create_discoverer
0001135:DEBUG:sequencer:Running task discovery
0001135:DEBUG:sequencer:Running task find_aps
0001139:DEBUG:sequencer:Running task create_aps
0001139:DEBUG:sequencer:Running task create_ap.0
0001140:INFO:ap:AHB-AP#0 IDR = 0x04770031 (AHB-AP var3 rev0)
0001142:DEBUG:ap:AHB-AP#0 default HPROT=3 HNONSEC=0
0001143:DEBUG:ap:AHB-AP#0 implemented HPROT=3 HNONSEC=0
0001144:DEBUG:sequencer:Running task find_components
0001144:DEBUG:sequencer:Running task init_ap.0
0001149:WARNING:rom_table:Invalid coresight component, cidr=0x0
0001149:DEBUG:sequencer:Running task create_cores
0001149:DEBUG:sequencer:Running task set_default_reset_type
0001149:DEBUG:sequencer:Running task create_components
0001149:DEBUG:sequencer:Running task check_for_cores
0001150:DEBUG:session:uninit session <pyocd.core.session.Session object at 0x7fd405854470>
0001150:DEBUG:pyusb_backend:closing interface
0001152:CRITICAL:__main__:No cores were discovered!
```

flit, 2021-01-24:
Hi @jpmzometa. This is an interesting one. The key log entry is `0001149:WARNING:rom_table:Invalid coresight component, cidr=0x0`.

Does the code you flashed happen to change power modes, especially enter a low power mode? The invalid CoreSight component ID is a clue that part of the chip is asleep, but it could be something else.

You can try connecting with `--connect=under-reset` to see if that helps. This is very often required to fully wake up devices in low power states.

jpmzometa, 2021-01-27:
Hi @flit thanks for your reply.

`pyocd erase -vv --chip -t lpc845  -O reset_type=hw --connect=under-reset`
results in a different error
`0001156:CRITICAL:__main__:No ACK received`.

I've been testing a bit, and I always get that warning `WARNING:rom_table:Invalid coresight component, cidr=0x0` even if the action is successful. See for example succesfully erasing the chip (and later on succesfully flashing the chip):
```
pyocd erase -vv --chip -t lpc845
0000489:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=1bcf PID=2c94). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000489:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=05f3 PID=0007). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000491:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=145f PID=0263). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000492:DEBUG:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=1bcf PID=2c94). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000492:DEBUG:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=05f3 PID=0007). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000493:DEBUG:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=145f PID=0263). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0000841:DEBUG:session:Project directory: /home/pablo/pyocd2
0001055:WARNING:cmsis_pack:Overlapping memory regions in file <zipfile.ZipFile filename='/home/pablo/.local/share/cmsis-pack-manager/NXP/LPC845_DFP/12.3.0.pack' mode='r'>, deleting outer region. Further warnings will be suppressed for this file.
0001076:INFO:board:Target type is lpc845
0001076:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=1bcf PID=2c94). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0001077:DEBUG:pyusb_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=05f3 PID=0007). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.
0001079:DEBUG:pyusb_backend:Detaching Kernel Driver of Interface 0 from USB device (VID=1fc9 PID=0090).
0001087:DEBUG:dap_access_cmsis_dap:CMSIS-DAP probe I3F4M0GS firmware version: 1.10
0001092:DEBUG:sequencer:Running task load_svd
0001174:DEBUG:sequencer:Running task pre_connect
0001174:DEBUG:sequencer:Running task dp_init
0001174:DEBUG:sequencer:Running task lock_probe
0001174:DEBUG:sequencer:Running task get_probe_capabilities
0001179:DEBUG:sequencer:Running task connect
0001275:DEBUG:dap:Default wire protocol selected; using SWD
0001275:DEBUG:swj:Sending deprecated SWJ sequence to select SWD
0001278:INFO:dap:DP IDR = 0x0bc11477 (v1 MINDP rev0)
0001278:DEBUG:sequencer:Running task clear_sticky_err
0001279:DEBUG:sequencer:Running task power_up_debug
0001279:DEBUG:sequencer:Running task check_version
0001279:DEBUG:sequencer:Running task unlock_probe
0001279:DEBUG:sequencer:Running task create_discoverer
0001280:DEBUG:sequencer:Running task discovery
0001280:DEBUG:sequencer:Running task find_aps
0001283:DEBUG:sequencer:Running task create_aps
0001283:DEBUG:sequencer:Running task create_ap.0
0001284:INFO:ap:AHB-AP#0 IDR = 0x04770031 (AHB-AP var3 rev0)
0001285:DEBUG:ap:AHB-AP#0 default HPROT=3 HNONSEC=0
0001286:DEBUG:ap:AHB-AP#0 implemented HPROT=3 HNONSEC=0
0001288:DEBUG:sequencer:Running task find_components
0001288:DEBUG:sequencer:Running task init_ap.0
0001293:INFO:rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0xf0000000 (designer=501 part=000)
0001296:WARNING:rom_table:Invalid coresight component, cidr=0x0
0001296:INFO:rom_table:[0]<15000c000: cidr=0, pidr=0, component invalid>
0001298:INFO:rom_table:[1]<e00ff000:ROM class=1 designer=43b part=4c0>
0001298:INFO:rom_table:  AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c0)
0001300:INFO:rom_table:  [0]<e000e000:SCS-M0+ class=14 designer=43b part=008>
0001303:INFO:rom_table:  [1]<e0001000:DWT-M0+ class=14 designer=43b part=00a>
0001305:INFO:rom_table:  [2]<e0002000:BPU class=14 designer=43b part=00b>
0001305:DEBUG:sequencer:Running task create_cores
0001305:DEBUG:discovery:Creating SCS-M0+ component
0001307:INFO:cortex_m:CPU core #0 is Cortex-M0+ r0p1
0001308:DEBUG:sequencer:Running task set_default_reset_type
0001308:DEBUG:sequencer:Running task create_components
0001308:DEBUG:discovery:Creating DWT-M0+ component
0001311:INFO:dwt:2 hardware watchpoints
0001312:DEBUG:discovery:Creating BPU component
0001313:INFO:fpb:4 hardware breakpoints, 0 literal comparators
0001313:DEBUG:fpb:fpb has been disabled
0001313:DEBUG:sequencer:Running task check_for_cores
0001313:DEBUG:sequencer:Running task halt_on_connect
[... cut ...]
0006798:INFO:loader:Erased 32768 bytes (32 sectors), programmed 32768 bytes (128 pages), skipped 0 bytes (0 pages) at 5.76 kB/s
0006798:DEBUG:session:uninit session <pyocd.core.session.Session object at 0x7fcb63ad6630>
0006798:DEBUG:board:uninit board <pyocd.board.board.Board object at 0x7fcb63a92978>
0006800:DEBUG:cortex_m:resuming core 0
0006800:DEBUG:manager:added=[] removed=[]
0006800:DEBUG:manager:bps after flush={}
0006804:DEBUG:pyusb_backend:closing interface
```

Attempting to flash the chip again (using the same command) gives the error show in the OP.

Hoohaha, 2021-02-04:
@jpmzometa Could you try the latest code? 
Same warning in there but the erase and flash works fine.
`0001404:WARNING:rom_table:Invalid coresight component, cidr=0x0`

jpmzometa, 2021-02-05:
I have tried the latest code (0.29.1.dev16), but the issue remains.

flit, 2021-02-05:
Thanks for testing. I'm ordering an LPC845-BRK board so I can try to reproduce it.

flit, 2021-02-05:
Looking at the user manual (UM11029) explains a few things.

Connect under-reset doesn't work because boundary scan mode is enabled when the nRESET pin is asserted; SWD is disabled while the part is held in reset. See section 32.5.3. This is an unfortunate design. It means the only way to recover from a low power mode or bad configuration is to use the ISP pin, which is usually not accessible by the debugger.

My guess is that the "invalid coresight component" is caused by two different components in the working and non-working case. In the working case, it's probably the MTB; section 32.3 says, "In order to use the micro-trace buffer for debugging, enable the MTB clock in the SYSAHBCLKCTRL register". And the MTB is not otherwise listed in the ROM table.

In the non-working case from your first log, it looks like the root ROM table itself is returning invalid IDs.

@jpmzometa What does the code you are flashing do? Does it put the device in a low power state?

jpmzometa, 2021-02-05:
Hi. The code does not put the device in a low power mode. 
The interesting thing is that the MCUXpresso IDE is able to flash/erase the LPC845 every time without problems using the same setup, independently of the state of the MCU (valid code, low power mode or not). So, my guess is that it is possible to do it via software (i.e. without the need for ISP pin), but no idea how.

flit, 2021-02-05:
Ok, thanks for confirming that. And it's good to know that it's MCUX can do it without the ISP pin—the LPC845-BRK schematic confirms there isn't an ISP pin connection to the on-board debug probe like there is on some other LPC boards.

Hoohaha, 2021-03-11:
@fli @jpmzometa 
I can reproduce this issue in my local. Board cannot be flashed when an application is running which used "WFI()" to wait for interrupt. IAR and MCUXpressoIDE both can connect to even target is sleeping. See below picture:
[image]

flit, 2021-03-11:
Hi @Hoohaha, thanks for taking a look at this and providing the IAR log. I did get an LPC845-BRK board, so I should be able to debug it. Are you using one of the MCUX demos for testing?

Hoohaha, 2021-03-12:
- Example name: pint_pattern_match 
- Example project in package: ./boards/lpc845breakout/driver_examples/pint/pattern_match/
- Compiled by MCUXpressoIDE:

[lpc845breakout_pint_pattern_match.zip](https://github.com/pyocd/pyOCD/files/6127309/lpc845breakout_pint_pattern_match.zip)

Test command: 
```
pyocd flash --format elf --target lpc845  <path/to/elf>
```

flit, 2021-03-12:
Thanks @Hoohaha !

flit, 2021-03-20:
Ok, I can reproduce it now. For some reason, immediately after I flash the test app it was working fine. It may have required a power cycle to reproduce.

flit, 2021-03-30:
The lpc845breakout_pint_pattern_match test image above reproduces the "No cores found" error. Using the "under-reset" connect mode, e.g., `--connect=under-reset`, seems to resolve all issues. Erasing and programming flash work fine, and connecting Commander works fine. (I'm not sure why you'd get a "No ACK" error, since I'm not getting that.)

Definitely seems to be related to low power mode. If I load the test image but leave the core halted, I can always connect fine. But if the core is running when attempting to connect, it fails. Also, resuming the CPU from within Commander results in a "Device entered sleep" message.

Pyocd fails to find any cores because the root ROM table reads as all zeroes when the device is in a low power state. IAR works because it apparently doesn't perform CoreSight discovery, and the CPUID and other core debug registers read correctly when the core is asleep. 

Btw, the reason a fully erased device works without errors is because it stays in the boot ROM and does not enter sleep.

I'll continue to explore this a bit and see if there's anything pyocd is missing or doing wrong.

Hoohaha, 2021-03-31:
@flit Thanks for so detailed analysis. 

As you know, there are many people were confused the exception "No cores were discovered". And I know somes are relate to specific soc design. Is possible that pyocd can automatically do something then to reconnect like you mentioned with a under-reset? This would probably improve the user experience.

flit, 2021-03-31:
You're right about the message being difficult to understand for many people. A big part is the lack of documentation. I'm working on a pyocd.io website and will be writing (hopefully a lot) more documentation to try to address this.

For most debuggers like Keil MDK, you just have to "know" which connect mode to use. But they tend to have pretty good documentation.

Automatically switching to under-reset a good idea. One problem is that not all MCUs can be connected under-reset. For instance the LPC55xx family cannot be connected at all while reset is asserted (because debug is completely disabled until mid-way through boot ROM execution). But, we could have a target flag that enables auto-reconnect-under-reset.

MiroHibler, 2022-03-24:
Hey guys, I'm now having the same problem with pyOCD v0.33.1 and LPC-Link2. Did you have any progress in the meantime?

flit, 2022-05-01:
@MiroHibler Thank you for all the details. Not sure when I'll be able to get back to this issue, but I haven't forgotten about it.

danieltecheyra, 2022-08-20:
If you would do me the favor of helping me, I have to use the lpc845brk with the esp232(bluethoo) and the ESP8266(WIFI), if someone has any file or tutorial, I would appreciate it. Thank you