pyocd/pyOCD #1057: Exception error when using Microchip Pack

sslupsky, 2021-01-21:
I encountered a problem when using the Microchip PACK for the atsamd51 series MCU's.

The first issue appears to be that pyOCD cannot determine the RAM layout.

The second problem causes an exception.

I tried with two different probes.  A JLink and a Particle Debugger (CMSIS-DAP) and the errors are similar.

Here is the output when using the JLink:

```
0010018:WARNING:cmsis_pack:Overlapping memory regions in file <zipfile.ZipFile filename='/Users/stevenslupsky/Documents/source/CMSIS-Packs/Microchip.SAMD51_DFP.3.3.76.atpack' mode='r'>, deleting outer region. Further warnings will be suppressed for this file.
0010054:WARNING:cmsis_pack:CMSIS-Pack device ATSAMD51J20A has no default RAM defined, cannot program flash
0010057:INFO:board:Target type is atsamd51j20a
0010612:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0010616:ERROR:discovery:Exception while probing AP#2: Unspecified error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/pyocd/probe/jlink_probe.py", line 318, in read_ap
    value = self._link.coresight_read((addr & self.A32) // 4, ap=True)
  File "/usr/local/lib/python3.9/site-packages/pylink/jlink.py", line 204, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/pylink/jlink.py", line 3361, in coresight_read
    raise errors.JLinkException(res)
pylink.errors.JLinkException: Unspecified error.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/pyocd/coresight/discovery.py", line 128, in _find_aps
    isValid = AccessPort.probe(self.dp, apsel)
  File "/usr/local/lib/python3.9/site-packages/pyocd/coresight/ap.py", line 304, in probe
    idr = dp.read_ap((ap_num << APSEL_SHIFT) | AP_IDR)
  File "/usr/local/lib/python3.9/site-packages/pyocd/coresight/dap.py", line 703, in read_ap
    result_cb = self.probe.read_ap(addr, now=False)
  File "/usr/local/lib/python3.9/site-packages/pyocd/probe/jlink_probe.py", line 320, in read_ap
    six.raise_from(self._convert_exception(exc), exc)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.ProbeError: Unspecified error.
0010619:INFO:ap:AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0010625:INFO:ap:AHB-AP#1 IDR = 0x74770001 (AHB-AP var0 rev7)
0010647:INFO:rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0x41003000 (designer=01f part=cd0)
0010659:INFO:rom_table:[0]<-1ff01000:ROM class=1 designer=43b part=4c4>
0010659:INFO:rom_table:  AHB-AP#0 Class 0x1 ROM table #1 @ 0x-1ff01000 (designer=43b part=4c4)
0010678:INFO:rom_table:  [0]<-1fff2000:SCS-M4 class=14 designer=43b part=00c>
0010686:INFO:rom_table:  [1]<-1ffff000:DWT class=14 designer=43b part=002>
0010695:INFO:rom_table:  [2]<-1fffe000:FPB class=14 designer=43b part=003>
0010706:INFO:rom_table:  [3]<-20000000:ITM class=14 designer=43b part=001>
0010716:INFO:rom_table:  [4]<-1ffc0000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=ca1:0:0>
0010724:INFO:rom_table:  [5]<-1ffbf000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0010734:INFO:rom_table:  [6]<-1ffbe000:ETB class=9 designer=43b part=907 devtype=21 archid=0000 devid=0:0:0>
0010736:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
0010743:INFO:cortex_m:FPU present: FPv4-SP-D16-M
0010746:INFO:dwt:4 hardware watchpoints
0010751:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0010772:INFO:server:Semihost server started on port 4444 (core 0)
0010776:INFO:gdbserver:GDB server started on port 3333 (core 0)
```

Here is the output when using the Particle Debugger:

```
0001084:WARNING:cmsis_pack:Overlapping memory regions in file <zipfile.ZipFile filename='/Users/stevenslupsky/Documents/source/CMSIS-Packs/Microchip.SAMD51_DFP.3.3.76.atpack' mode='r'>, deleting outer region. Further warnings will be suppressed for this file.
0001120:WARNING:cmsis_pack:CMSIS-Pack device ATSAMD51J20A has no default RAM defined, cannot program flash
0001123:INFO:board:Target type is atsamd51j20a
0001272:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0001310:ERROR:discovery:Exception while probing AP#2: Memory transfer fault
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/pyocd/probe/cmsis_dap_probe.py", line 286, in read_ap_result_callback
    return result()
  File "/usr/local/lib/python3.9/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 847, in read_reg_cb
    res = transfer.get_result()
  File "/usr/local/lib/python3.9/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 144, in get_result
    self.daplink.flush()
  File "/usr/local/lib/python3.9/site-packages/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 672, in flush
    self._read_packet()
  File "/usr/local/lib/python3.9/site-packages/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 932, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "/usr/local/lib/python3.9/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 452, in decode_data
    data = self._decode_transfer_data(data)
  File "/usr/local/lib/python3.9/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 359, in _decode_transfer_data
    self._check_response(data[2])
  File "/usr/local/lib/python3.9/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 338, in _check_response
    raise DAPAccessIntf.TransferFaultError()
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferFaultError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/pyocd/coresight/discovery.py", line 128, in _find_aps
    isValid = AccessPort.probe(self.dp, apsel)
  File "/usr/local/lib/python3.9/site-packages/pyocd/coresight/ap.py", line 304, in probe
    idr = dp.read_ap((ap_num << APSEL_SHIFT) | AP_IDR)
  File "/usr/local/lib/python3.9/site-packages/pyocd/coresight/dap.py", line 729, in read_ap
    return read_ap_cb()
  File "/usr/local/lib/python3.9/site-packages/pyocd/coresight/dap.py", line 717, in read_ap_cb
    result = result_cb()
  File "/usr/local/lib/python3.9/site-packages/pyocd/probe/cmsis_dap_probe.py", line 288, in read_ap_result_callback
    six.raise_from(self._convert_exception(error), error)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.TransferFaultError: Memory transfer fault
0001345:INFO:ap:AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0001355:INFO:ap:AHB-AP#1 IDR = 0x74770001 (AHB-AP var0 rev7)
0001375:INFO:rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0x41003000 (designer=01f part=cd0)
0001382:INFO:rom_table:[0]<-1ff01000:ROM class=1 designer=43b part=4c4>
0001382:INFO:rom_table:  AHB-AP#0 Class 0x1 ROM table #1 @ 0x-1ff01000 (designer=43b part=4c4)
0001389:INFO:rom_table:  [0]<-1fff2000:SCS-M4 class=14 designer=43b part=00c>
0001394:INFO:rom_table:  [1]<-1ffff000:DWT class=14 designer=43b part=002>
0001399:INFO:rom_table:  [2]<-1fffe000:FPB class=14 designer=43b part=003>
0001404:INFO:rom_table:  [3]<-20000000:ITM class=14 designer=43b part=001>
0001409:INFO:rom_table:  [4]<-1ffc0000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=ca1:0:0>
0001414:INFO:rom_table:  [5]<-1ffbf000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0001419:INFO:rom_table:  [6]<-1ffbe000:ETB class=9 designer=43b part=907 devtype=21 archid=0000 devid=0:0:0>
0001422:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
0001430:INFO:cortex_m:FPU present: FPv4-SP-D16-M
0001435:INFO:dwt:4 hardware watchpoints
0001438:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0001457:INFO:server:Semihost server started on port 4444 (core 0)
0001460:INFO:gdbserver:GDB server started on port 3333 (core 0)
```

This is the contents of pyocd.yaml:
```
pack:
  - /Users/stevenslupsky/Documents/source/CMSIS-Packs/Microchip.SAMD51_DFP.3.3.76.atpack
  - /Users/stevenslupsky/Documents/source/CMSIS-Packs/Microchip.SAMD21_DFP.3.3.98.atpack

```

flit, 2021-01-23:
Thanks for the report. I'll have to look at the reference manual for the SAMD51 parts. The fact that it has a second AHB-AP is kind of surprising since it's a single core device.

I'll also take a look at the CMSIS Pack memory region definitions. Given the warnings from pyocd, it seems like there are problems that Microchip will have to fix.

The J-Link "unspecified error" is really the same transfer fault, J-Link just doesn't return a meaningful error. I've created #1063 to convert it to a transfer fault, about the best that can be done.

sslupsky, 2021-01-23:
The exception says it crashed while probing AP#2.  The numbering in the dump suggests they start at 0 so AP#2 would be the third port?  The info dump lists AP#0 and AP#1 but there is no rom_table info for AP#1.

Looking at the datasheet I cannot really tell if it has more than one AHB-AP.  It seems to me there is only one but there is a TPIU port and a SWD port so perhaps somehow they enumerate as separate AP's?

flit, 2021-01-23:
Oh, I didn't even catch that about probing AP#2! That means that is got a fault result from attempting to read the ID register of AP#2. This is against the Arm ADI specification, which requires that reads of nonexistent AP registers return 0 (RAZWI in Arm terminology).

laurencedv, 2023-04-11:
I get the same error with mcu ATSAME54P20A which is on the microchip devkit for same54 family. Is there any work around using the microchip pack? I am trying to setup a toolchain using pyocd as the chip handling layer but this seems to be a blocking problem doesn't it?