pyocd/pyOCD #1066: memory leak when calling ConnectHelper.get_all_connected_probes()

maxschuh, 2021-01-27:
**Issue:**
When calling ConnectHelper.get_all_connected_probes() multiple times memory consumption increases indefinitely.

**System:**
- Linux
- python3.8, python3.9
- pyocd-0.28.3

**Steps to reproduce:**
`>>> from pyocd.core.helpers import ConnectHelper`
`>>> while True:`
`>>> ... ConnectHelper.get_all_connected_probes(blocking=False)`

I also tried to call garbage the collector manually, however this didn't help. Presumably there is still a reference somewhere to something?! The background of the hole thing is, that I have a daemon that checks periodically for new probes and after some time the daemon occupies the whole memory.

flit, 2021-01-31:
Thanks for reporting. Definitely an issue if you need to continually check for new probes!

I agree there's probably a reference left dangling somewhere. Hopefully it's not somewhere in pyusb that can't be easily fixed.

flit, 2021-01-31:
What type(s) of probe(s) are you using? That could narrow down the possibilities.

maxschuh, 2021-02-01:
The memory leak seems to be independent from the type of probe as there is a leak even if no probe is attached. However typically I use J-Link and/or ST-Link probes where the error also occurs.

flit, 2021-02-01:
Very interesting, I'm surprised it leaks with no probe attached. That's very helpful.

flit, 2021-02-07:
I can reproduce it on macOS 10.15 with one CMSIS-DAPv1 (hidapi), one CMSIS-DAPv2 (libusb), and an STLinkV2-1 (libusb).

It doesn't seem to leak when no probes are attached. I need to try it on Linux and Windows next, and start trying to narrow down the root cause.

joncutting, 2023-11-09:
I also confirm this is happening, I have 3 Raspberry Pi Debug Probes attached and after making O(10) consecutive connections on a Raspberry Pi 3 I see the same MemoryError:

Traceback (most recent call last):
  File "/home/pi/sram/swdtest.py", line 366, in cycle_dut
    session = ConnectHelper.session_with_chosen_probe(unique_id = debug_id,
  File "/home/pi/sram/.env/lib/python3.9/site-packages/pyocd/core/helpers.py", line 257, in session_with_chosen_probe
    probe = ConnectHelper.choose_probe(
  File "/home/pi/sram/.env/lib/python3.9/site-packages/pyocd/core/helpers.py", line 154, in choose_probe
    allProbes = ConnectHelper.get_all_connected_probes(blocking=blocking, unique_id=unique_id)
  File "/home/pi/sram/.env/lib/python3.9/site-packages/pyocd/core/helpers.py", line 94, in get_all_connected_probes
    allProbes = DebugProbeAggregator.get_all_connected_probes(unique_id=unique_id)
  File "/home/pi/sram/.env/lib/python3.9/site-packages/pyocd/probe/aggregator.py", line 57, in get_all_connected_probes
    probe = cls.get_probe_with_id(unique_id, is_explicit)
  File "/home/pi/sram/.env/lib/python3.9/site-packages/pyocd/probe/jlink_probe.py", line 84, in get_probe_with_id
    jlink = cls._get_jlink()
  File "/home/pi/sram/.env/lib/python3.9/site-packages/pyocd/probe/jlink_probe.py", line 58, in _get_jlink
    return pylink.JLink(
  File "/home/pi/sram/.env/lib/python3.9/site-packages/pylink/jlink.py", line 293, in __init__
    lib = library.Library(use_tmpcpy=use_tmpcpy)
  File "/home/pi/sram/.env/lib/python3.9/site-packages/pylink/library.py", line 293, in __init__
    self.load_default()
  File "/home/pi/sram/.env/lib/python3.9/site-packages/pylink/library.py", line 351, in load_default
    return self.load(path)
  File "/home/pi/sram/.env/lib/python3.9/site-packages/pylink/library.py", line 401, in load
    outputfile.write(inputfile.read())
MemoryError