pyocd/pyOCD #1075: module import error

nerdralph, 2021-02-13:
When I try to run pyocd.py on Debian10 from virtualenv or after running setup.py build, I get this error:
```
Traceback (most recent call last):
  File "./pyocd.py", line 23, in <module>
    from .. import __version__
ValueError: attempted relative import beyond top-level package
```
The problem also occurs with Win7/mingw.

Is this the same as issue #1004?

flit, 2021-02-13:
Thanks.

How are you invoking pyocd? Normally `setup.py` will create an executable called `pyocd` with no extension. This is the executable that should be run. Alternatively you can run it as a module via `python -mpyocd`, which will call the `pyocd.__main__` module.

The scripts under `pyocd/tools/` are deprecated and should normally not be used. Even so, `setup.py` is still set up to create executables for these scripts: `pyocd-tool`, `pyocd-flash`, and `pyocd-gdbserver`. Pretty soon these will be dropped, probably the first two very soon, maybe next release. (`pyocd-gdbserver` is still called by the cortex-debug plugin for VSCode; I need to create a PR to update it.)

nerdralph, 2021-02-13:
I had run `setup.py build` then tried running `build/lib/pyocd/tools/pyocd.py`.  My intention was to be able to run and debug pyocd.py, and not to replace the pyocd version I had installed via pip.
When I try to run `setup.py install`, I get permission errors because it tries to install globally.  I checked the help options, and I don't see a flag to install locally, the way pip will if it isn't run as root.

flit, 2021-02-14:
You can use pip instead of running `setup.py`. Like `pip install --user .` from the pyocd root.

The `-e` option of pip lets you install pyocd as "editable", meaning a reference to the pyocd code in your working copy is installed. The pyocd executables are created as usual when you install. Any changes you make will immediately be reflected when you run the executables.

If you install via the `dev-requirements.txt` file as recommended in the [developer's guide](https://github.com/pyocd/pyOCD/blob/master/docs/developers_guide.md), pyocd will automatically be installed as editable.

The best way to deal with all this is to create a virtualenv and install pyocd into it. This will create a new virtualenv named `.venv`, activate it, then install pyocd as editable and all its dependencies.

```
$ python3 -mvenv .venv
$ source .venv/bin/activate
$ pip install -r dev-requirements.txt
```

(The above uses python3's built-in venv module. You can also use the `virtualenv` package to the same effect.)

flit, 2021-03-20:
Closing this issue. If you still have trouble, please feel free to reopen it.

closed 2021-03-20 by flit