pyocd/pyOCD #1089: document probe & target detection algorithm

nerdralph, 2021-02-24:
I recently tried changing the serial string reported by my DAP firmware to "123456", and was surprised that pyOCD then listed the probe as "u-blox-C027".  After looking at the pyOCD source, I found a list of serial strings that are associated with different boards.
https://github.com/pyocd/pyOCD/blob/18252c8b78252448fe193bee26c774850db34f2c/pyocd/board/board_ids.py#L191

I found an old issue regarding board IDs, but nothing seems to have been done to document how board IDs are assigned and maintained.
https://github.com/pyocd/pyOCD/issues/151

I think pyOCD also implements a convention wherein a probe that detects the target can report this to pyOCD in the product string in square brackets.  i.e. "CMSIS-DAP [STM32F030F4]".  Is this unique to pyOCD, or is it part of the CMSIS-DAP spec?

flit, 2021-02-24:
The reason you see the u-blox board name is because of a recently found bug in the board identification code for CMSIS-DAP, fixed via #1090 (it was pending on another branch, but I moved it forward separately just now).

Board IDs comes from Mbed OS and the Mbed Enabled program. (pyocd and DAPLink were originally Mbed projects, but now pyocd is no longer associated and DAPLink is tenuously associated.)

Sadly, there is no real specification that I'm aware of. Basically, a board ID is a 4-character code that uniquely identifies a board. It also identifies an Mbed platform: the same ID is used for both (which is actually a problem). Board IDs currently are all hex numeric, but it's actually a string, though there may be problems raised in some components if you tried to use a non-hex char).

Currently these debug probe firmware support board IDs:
- DAPLink, via the first 4 chars of the USB serial number
- STLinkV2-1 and V3, via an .htm file on the associated USB mass storage volume.

You can see the full list of public Mbed platforms at https://os.mbed.com/api/v3/platforms/. The `productcode` key is the board ID.

Pyocd is itself constructing the probe description strings like "CMSIS-DAP [STM32F030F4]", not the probe. https://github.com/pyocd/pyOCD/blob/74a0b3ada8d3b76b2beb3005ccebd8a0ddf89aff/pyocd/board/mbed_board.py#L79-L81

(Btw, I'm painfully slowly working on a website for pyocd where docs can be more readily available. Once that's up my plan is to start writing docs as much as I have time.)

flit, 2021-02-24:
I created a wiki page with some of this content, and more: https://github.com/pyocd/pyOCD/wiki/Board-IDs

closed 2021-03-08 by flit