pyocd/pyOCD #1105: dependency automatic install issue

nerdralph, 2021-03-01:
I tried doing a fresh pyOCD install in Win7e/MSYS2 using pip3, and got errors on the psutil install:
```
    C:\msys64\mingw64\bin/x86_64-w64-mingw32-gcc.exe -Wno-unused-result -Wsign-compare -DNDEBUG -march=x86-64 -mtune=generic -O2 -pipe -fwrapv -D__USE_MINGW_ANSI_STDIO=1 -D_WIN32_WINNT=0x0601 -DNDEBUG -march=x86-64 -mtune=generic -O2 -pipe -fwrapv -D__USE_MINGW_ANSI_STDIO=1 -D_WIN32_WINNT=0x0601 -DNDEBUG -DPSUTIL_SIZEOF_PID_T=4 -DPSUTIL_VERSION=580 -DPSUTIL_WINDOWS=1 -D_WIN32_WINNT=0x0601 -D_AVAIL_WINVER_=0x0601 -D_CRT_SECURE_NO_WARNINGS -DPSAPI_VERSION=1 -IC:/msys64/mingw64/include/python3.8 -c psutil/_psutil_common.c -o build/temp.mingw-3.8/psutil/_psutil_common.o
    In file included from psutil/_psutil_common.h:120,
                     from psutil/_psutil_common.c:10:
    psutil/arch/windows/ntextapi.h:59:14: error: redeclaration of 'enum _KWAIT_REASON'
       59 | typedef enum _KWAIT_REASON {
          |              ^~~~~~~~~~~~~
```

After explicitly installing psutil and wheel, pyOCD installed OK.
```
psutil             5.6.7
wheel              0.36.2
pyocd              0.29.0
```

flit, 2021-03-02:
I have seen an error with psutil previously, not too long ago… 

Do you think it's because the version of wheel was too old? We can try adding explicit wheel and maybe psutil version dependencies to pyocd's `setup.py`.

Fwiw, it's having to build psutil instead of using one of the [prebuilt wheels](https://pypi.org/project/psutil/5.8.0/#files) due to the combo of OS and environment you're using. You might try normal Win32 env instead of MSYS2 if you have other issues.

nerdralph, 2021-03-02:
My python-fu isn't good enough to know exactly what was going on, but I can tell you I didn't have wheel installed when I first tried "pip3 install pyocd".  I'm pretty sure I saw a message from pip that the pyocd install was proceeding without using wheels.

I'm using both mingw32 and MSYS2 x86_64 for dual testing environments.  The main purpose was for some stuff I'm trying with clang/llvm, and since I had the MSYS2 environment already setup I decided to try pyocd.

flit, 2021-03-08:
Can you get back to a state where you can do a fresh install of pyOCD in your MSYS2?

Please try installing pyOCD from this branch where I added a setup dependency on `wheel`:

```
pip install git+https://github.com/flit/pyOCD.git@bugfix/add_wheel_setup_requires
```

nerdralph, 2021-03-09:
That still isn't working.
```
Using legacy 'setup.py install' for capstone, since package 'wheel' is not installed.
Using legacy 'setup.py install' for psutil, since package 'wheel' is not installed.
Installing collected packages: pypemicro, psutil, pyocd-pemicro, pyelftools, hidapi, capstone, pyocd
    Running setup.py install for psutil ... error
    ERROR: Command errored out with exit status 1:
```
[...]
And then the same _KWAIT_REASON error as before.

I then ran 'pip install wheel', and get another error:
```
  Complete output (8 lines):
  running bdist_wheel
  running build
  Building C extensions
  'cmake' is not recognized as an internal or external command,
  operable program or batch file.
  'nmake' is not recognized as an internal or external command,
  operable program or batch file.
  error: [Errno 2] No such file or directory: 'capstone.dll'
```

Then I tried running "pip install psutil', and got an error trying that.
I tried uninstalling wheel, and 'pip install psutil' still gives an error.  It seems my python env is now fubar'd.

I'll try removing everything including pip, try again, and let you know how it goes.

nerdralph, 2021-03-09:
I removed pip, and reinstalled python 3.8.7 and pip 20.2.2.  Your branch still fails.
```
  Downloading cmsis-pack-manager-0.2.10.tar.gz (45 kB)
     |████████████████████████████████| 45 kB 242 kB/s
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python.exe -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-n4hx9qal/cmsis-pack-manager/setup.py'"'"'; __file__='"'"'/tmp/pip-install-n4hx9qal/cmsis-pack-manager/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-7wcebnp5
         cwd: /tmp/pip-install-n4hx9qal/cmsis-pack-manager/
    Complete output (48 lines):
    WARNING: The wheel package is not available.
    WARNING: The wheel package is not available.
    WARNING: The wheel package is not available.
    WARNING: The wheel package is not available.
    WARNING: The wheel package is not available.
      ERROR: Command errored out with exit status 1:
       command: /usr/bin/python.exe -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-wheel-zg22r46w/cffi/setup.py'"'"'; __file__='"'"'/tmp/pip-wheel-zg22r46w/cffi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-8fl9h666
           cwd: /tmp/pip-wheel-zg22r46w/cffi/
      Complete output (6 lines):
      usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
         or: setup.py --help [cmd1 cmd2 ...]
         or: setup.py --help-commands
         or: setup.py cmd --help

      error: invalid command 'bdist_wheel'
      ----------------------------------------
      ERROR: Failed building wheel for cffi
    ERROR: Failed to build one or more wheels
    Traceback (most recent call last):
      File "/usr/lib/python3.8/site-packages/setuptools/installer.py", line 128, in fetch_build_egg
        subprocess.check_call(cmd)
      File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['/usr/bin/python.exe', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmpf3pr1x6u', '--quiet', 'cffi>=1.6.0']' returned non-zero exit status 1.

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-n4hx9qal/cmsis-pack-manager/setup.py", line 55, in <module>
        setup(
      File "/usr/lib/python3.8/site-packages/setuptools/__init__.py", line 160, in setup
        _install_setup_requires(attrs)
      File "/usr/lib/python3.8/site-packages/setuptools/__init__.py", line 155, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/usr/lib/python3.8/site-packages/setuptools/dist.py", line 695, in fetch_build_eggs
        resolved_dists = pkg_resources.working_set.resolve(
      File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 781, in resolve
        dist = best[req.key] = env.best_match(
      File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 1066, in best_match
        return self.obtain(req, installer)
      File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 1078, in obtain
        return installer(requirement)
      File "/usr/lib/python3.8/site-packages/setuptools/dist.py", line 754, in fetch_build_egg
        return fetch_build_egg(self, req)
      File "/usr/lib/python3.8/site-packages/setuptools/installer.py", line 130, in fetch_build_egg
        raise DistutilsError(str(e))
    distutils.errors.DistutilsError: Command '['/usr/bin/python.exe', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmpf3pr1x6u', '--quiet', 'cffi>=1.6.0']' returned non-zero exit status 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```