pyocd/pyOCD #1107: Some errors occured when starting debug with Eclipse+pyOCD+JLink

lennvn, 2021-03-02:
0005859:ERROR:gdbserver:Exception while executing remote command 'reset init': unexpected value for reset_type option ('init')
Traceback (most recent call last):
  File "c:\python\python37-32\lib\site-packages\pyocd-0.22.1.dev697-py3.7.egg\pyocd\gdbserver\gdbserver.py", line 1017, in handle_remote_command
    self._command_context.process_command_line(cmd)
  File "c:\python\python37-32\lib\site-packages\pyocd-0.22.1.dev697-py3.7.egg\pyocd\commands\execution_context.py", line 301, in process_command_line
    invoc.handler(invoc)
  File "c:\python\python37-32\lib\site-packages\pyocd-0.22.1.dev697-py3.7.egg\pyocd\commands\execution_context.py", line 370, in execute_command
    cmd_object.parse(invocation.args)
  File "c:\python\python37-32\lib\site-packages\pyocd-0.22.1.dev697-py3.7.egg\pyocd\commands\commands.py", line 352, in parse
    self.reset_type = convert_reset_type(args[0])
  File "c:\python\python37-32\lib\site-packages\pyocd-0.22.1.dev697-py3.7.egg\pyocd\utility\cmdline.py", line 143, in convert_reset_type
    raise ValueError("unexpected value for reset_type option ('%s')" % value)
ValueError: unexpected value for reset_type option ('init')
0005890:INFO:gdbserver:Attempting to load argon

flit, 2021-03-02:
Thanks for the report. What version of the Eclipse plugin are you using? I assume it's Eclipse Embedded CDT. If it's not version [6.1.1](https://eclipse-embed-cdt.github.io/blog/2021/01/20/plugins-v6.1.1-released/), can you please try testing with that version?

lennvn, 2021-03-03:
Thanks.
I updated Embedded CDT to 6.1.1 and the problem is solved. I also did some more tests with pyocd+JLink and not sure if the test result is a problem.
------------------------------------------------------------------------
I tested the latest(id: 8680ee61dcbec1f49081d17e8923526e20b27d29) master branch with windows terminal. The operations are as follows:
PS C:\Users\wuze> pyocd gdb -t hc32f46x
(......)
0002377:INFO:server:Semihost server started on port 4444 (core 0)
0002448:INFO:gdbserver:GDB server started on port 3333 (core 0)
>(Ctrl+C to abort)
0003455:ERROR:session:probe exception during disconnect:
Traceback (most recent call last):
  File "C:\Python\Python37-32\lib\site-packages\pyocd-0.29.1.dev76-py3.7.egg\pyocd\__main__.py", line 787, in do_gdbserver
    sleep(0.1)
KeyboardInterrupt

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Python\Python37-32\lib\site-packages\pyocd-0.29.1.dev76-py3.7.egg\pyocd\core\session.py", line 479, in close
    self._probe.disconnect()
  File "C:\Python\Python37-32\lib\site-packages\pyocd-0.29.1.dev76-py3.7.egg\pyocd\probe\shared_probe_proxy.py", line 73, in disconnect
    self._probe.disconnect()
  File "C:\Python\Python37-32\lib\site-packages\pyocd-0.29.1.dev76-py3.7.egg\pyocd\probe\jlink_probe.py", line 248, in disconnect
    self._invalidate_cached_registers()
AttributeError: 'JLinkProbe' object has no attribute '_invalidate_cached_registers'
0004398:INFO:server:Semihost server stopped
0004410:INFO:gdbserver:GDB server thread killed
------------------------------------------------------------------------
Then I unistalled this version, installed the latest version 0.29.0, and did the same test:
PS C:\Users\wuze> pyocd gdb -t hc32f46x
0001454:INFO:board:Target type is hc32f46x
(......)
0001942:INFO:server:Semihost server started on port 4444 (core 0)
0001988:INFO:gdbserver:GDB server started on port 3333 (core 0)
>(Ctrl+C to abort)
0032989:INFO:server:Semihost server stopped
0033000:INFO:gdbserver:GDB server thread killed
------------------------------------------------------------------------
pyocd+CMSIS_DAP has no such difference.