pyocd/pyOCD #1119: ELFSymbolProvider for structure/array elements

jimfred, 2021-03-10:
As far as I know, ELFSymbolProvider will provide addresses for simple symbols, like main(), but isn't capable of providing addresses for more complex namespace/class/structure/array elements. So, I've been calling gdb in batch mode, telling it to open my .elf file and evaluate something like &MyNamespace::MyObject.someArray[5].blah. The resulting address and type info is used in my .py GUIs to display values, similar to a live-watch window in a debugger, using pyOCD.

The gdb batch method works well to evaluate addresses but I'd like to improve on 2 problems: 
1. My app's distribution needs a gdb executable and its .dll or .so files
2. The out-of-process batch mode can be a bit slow, on the order of 10's of seconds.

Occasionally, I see questions and comments, on the internet, that 'import gdb' (using gdb as a library) might someday become a reality for stand-alone python apps and that, today, it's only possible for python running inside of gdb.

My question is:
Is there a better way of evaluating the address of expressions?

jimfred, 2021-03-25:
LLDB can evaluate expressions and is callable, in-process, from python. 

This example...
```
import lldb
dbg = lldb.SBDebugger().Create()
target = dbg.CreateTarget("my_arm.elf")
val = target.EvaluateExpression("&user_data.sub_type[2].i[3]")
print("'{0}', type={1}, name={2}, value={3}, result={4}".format(val, val.type, val.name, val.value, val.GetError()))
```
...results in this...
```
'(int *) $0 = 0x200001f0', type=int *, name=$0, value=0x200001f0, result=success
```
Installation of LLDB is not as simple as a `pip install lldb` but my example works on Linux. Still trying on Windows.

closed 2021-03-25 by jimfred

flit, 2021-03-28:
That's really cool. I knew LLDB had good Python support, but didn't realise it extended to being able to import and use it as a module.