pyocd/pyOCD #1121: cannot find my WCH-Link

kaidegit, 2021-03-13:
WCH-Link is a CMSIS-dap debugger made by WCH.

It can be found by Keil and OpenOCD when adding the command `cmsis_dap_vid_pid 0x1a86 0x8011`. 

However, it cannot be found by PyOCD. When I run `pyocd list`, it just shows `
No available debug probes are connected`.

flit, 2021-03-15:
What OS are you using?

Can you please provide a USB descriptor dump?
- `lsusb` on Linux
- [usbview.exe](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview) on Windows
- several options [here](https://stackoverflow.com/questions/17058134/is-there-an-equivalent-of-lsusb-for-os-x) for macOS.

Could you please provide the log for `pyocd list -vv`? This will output any debug logs, potentially including USB errors.

Thanks!

kaidegit, 2021-03-16:
The pyocd version is 0.30.3.dev0+dirty.

I'm using macOS Big Sur(11.0.1).

I ran `lsusb -v` on my Mac and it says:
``` shell
            WCH-Link:

              Product ID: 0x8011
              Vendor ID: 0x1a86
              Version: 1.02
              Serial Number: 0001A0000001
              Speed: Up to 12 Mb/s
              Manufacturer: wch.cn
              Location ID: 0x14a40000 / 8
              Current Available (mA): 500
              Current Required (mA): 500
              Extra Operating Current (mA): 0
```

The log for `pyocd list -vv` is 
```shell
0001355:DEBUG:session:Project directory: /Users/kai
0001370:DEBUG:pemicro:Opened PEMicro library: /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pypemicro-0.1.5-py3.8.egg/pypemicro/libs/MacOS/unitacmp-64.dylib
No available debug probes are connected
```

kaidegit, 2021-03-16:
However, i can find the wchlink on my windows laptop.....

system version: Windows  10 Pro 21H1

pyocd version: 0.29.0

``` shell
pyocd list -vv
0003547:DEBUG:session:Project directory: C:\Users\yekai
0003662:WARNING:common:STLink and CMSIS-DAPv2 probes are not supported because no libusb library was found.
  #   Probe                  Unique ID
-------------------------------------------
  0   wch.cn WCH CMSIS-DAP   0001A0000001
```

flit, 2021-03-19:
Apologies for taking a while to get back to this. Unfortunately the `lsusb` call on your Mac didn't provide the descriptors I need. Would you mind trying out this:

```
$ brew install mikhailai/misc/usbutils
$ lsusb -v -d   > usb-descriptors.txt
```

Then attach the `usb-descriptors.txt` file to a comment here. (It will be large

If you still have the `lsusb` package installed, you'll have to either uninstall it or run `brew unlink lsusb` first, as it conflicts with usbutils.

Another option would be to run usbview.exe on your Windows laptop and copy the info text dump. The data will be the same.

kaidegit, 2021-03-20:
```shell
Bus 020 Device 012: ID 1a86:8011 QinHeng Electronics 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         8
  idVendor           0x1a86 QinHeng Electronics
  idProduct          0x8011 
  bcdDevice            1.02
  iManufacturer           1 wch.cn
  iProduct                2 WCH-Link
  iSerial                 3 0001A0000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          107
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       1 AT-commands (v.25ter)
      iFunction               4 WCH CMSIS-DAP
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      0 None
      iInterface              4 WCH CMSIS-DAP
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x01
          call management
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 
      iInterface              5 (error)
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              6 (error)
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      33
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

```shell
{'interface_number': 2,
  'manufacturer_string': 'wch.cn',
  'path': b'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC@14/XH'
          b'C@14000000/HS10@14a00000/USB 2.0 Hub [MTT]@14a00000/AppleUSB20Hu'
          b'b@14a00000/AppleUSB20HubPort@14a40000/WCH-Link@14a40000/WCH CMSI'
          b'S-DAP@2/AppleUserUSBHostHIDDevice',
  'product_id': 32785,
  'product_string': 'WCH-Link',
  'release_number': 258,
  'serial_number': '0001A0000001',
  'usage': 1,
  'usage_page': 65280,
  'vendor_id': 6790},
```

flit, 2021-03-20:
Thanks! The reason it's not detected is pretty simple: there's no "CMSIS-DAP" in the device's product name string.

The CMSIS-DAP specification also allows for "CMSIS-DAP" to appear in the interface name string. But the hidapi library doesn't report the interface's name (if it has one). And, the interface name descriptors are apparently messed up on the WCH-Link: the CDC interface association and first CDC interface have the name "WCH CMSIS-DAP", but the CMSIS-DAP HID interface has an erroneous string ID 6 (according to lsusb).

However, oddly, the string "WCH CMSIS-DAP" appears in the 'path' of the hidapi device info dictionary, associated with interface 2 (the HID interface). This doesn't add up… But it does present an option for detecting CMSIS-DAPv1 devices that only have "CMSIS-DAP" in the interface string when using hidapi.

I'll make a change to look for "CMSIS-DAP" in the hidapi device path, and give you a branch to try it out with. Should be ready within a few days once I get some time to work on it.

flit, 2021-03-20:
Ok, the change is ready for testing on [this branch](https://github.com/flit/pyOCD/tree/bugfix/hidapi_uid_and_if_name) of my fork.

You can install like this:

```
$ pip install -U git+https://github.com/flit/pyOCD@bugfix/hidapi_uid_and_if_name
```

kaidegit, 2021-03-21:
OK, It can find my wch link now.

Thanks!

flit, 2021-03-21:
Great! Thanks for testing.

closed 2021-03-21 by flit

eugene-bright, 2022-12-04:
It's actual for me again currently (pyocd 0.34.3)
The problem is reproducible with both `1a86:8011` and `2a86:8011`

eugene-bright, 2022-12-05:
More info on wch-LinkE-R0-1v2
https://www.wch.cn/products/WCH-Link.html?spm=a2g0o.detail.1000023.14.56d72adcTTkc7f

eugene-bright, 2022-12-05:
It doesn't look to be based on DAP-Link based, the following command do nothing

```
openocd -f interface/cmsis-dap.cfg -f target/stm32g0x.cfg -c "cmsis_dap_vid_pid 0x1a86 0x8010"
```

Here is a modified version of openocd that can provide a hint
https://nc-pin.com/index.php/2022/04/25/openocd-for-ch32v-series/

elfmimi, 2022-12-05:
@eugene-bright 1a86:8010 means it is in RISC-V mode. Re-check its recognized product name. You should see `WCH-LinkRV` .

eugene-bright, 2022-12-05:
The docs are in Chinese only. I clicked IAP button and now I can't go out of IAP mode :hide-the-pain:
One more peace of data I found
https://www.wch.cn/bbs/thread-71088-1.html

kaidegit, 2022-12-05:
> The docs are in Chinese only. I clicked IAP button and now I can't go out of IAP mode :hide-the-pain:
> 
> One more peace of data I found
> 
> https://www.wch.cn/bbs/thread-71088-1.html

it should be work after replug it。。。do you flash the ch549's firmware？ it should be flashed the firmware marked ch32v307 and ota。

btw，the chip has no boot0 to go to chip's bootloader. So if the iap part is broken, it is a must to have an another wch link or linke to flash it.

kaidegit, 2022-12-05:
Pyocd may can only work with arm mcu. And the debugger port of v003 is not open sourced. So just use the openocd wch offered to debug v003.

eugene-bright, 2022-12-05:
👌

PR #1122 (2021-03-21): CMSIS-DAP backend synthetic UID and device detection
This PR includes some changes for the CMSIS-DAP USB backends.

The hidapi and pywinusb backends will generate a synthetic UID if the device has no USB serial number, to behave the same as the pyusb backends. The common `generate_device_unique_id()` function is modified to accept device info parameters instead of a pyusb device.

The hidapi backend checks the `path` device info key for the "CMSIS-DAP" string in addition to the product name. This is done to attempt to detect CMSIS-DAP devices with the "CMSIS-DAP" string in the interface name,  [... cut]