pyocd/pyOCD #1129: Use of pyocd to flash from powershell seems to emit an error

thedjnK, 2021-04-01:
We have had a report from someone internally using pyocd on a windows 10 system with a DAPLINK programmer trying to program an nRF52840 module whereby after it has finished, they get the error "An error has occurred that was not properly handled. Additional information is shown below. The windows powershell process will exit

[process exited with code 2]"

Upon trying to reproduce the issue on my windows 10 machine using pyocd 0.29.x I could not reproduce it, I upgraded to the latest, 0.30.3, and when flashing using powershell it seems to go OK then for me, about 2 seconds later, the powershell window closed by itself (this is a prompt I opened in a directory, not using a dedicated command just to run the pyocd command). Using command prompt inside a powershell window does not cause any issues or errors or windows closing on either of these machines so it seems to be an issue using it directly with powershell.

newbrain, 2021-04-01:
Interesting,
This happened to me (only once) yesterday, flashing with a CMSIS DAP (RT1020EVK), from a PS inside Visual Studio Code.
I imputed it to a random fluke, and had a "funny" elf file,  but I was on 0.30.1.

Since I'm going to heavily use it in the following days, I'll report here if it happens with any regularity.

flit, 2021-04-01:
Very interesting… thanks for the report. I don't use Windows much, and never PowerShell, so never would have seen this. Any ideas on what could be different between PS and cmd.exe?

Is there a way to enable crash reports with logs and stack traces in Windows?

Anyway, if it's reproducible, maybe I can run Python under Visual Studio to see if it's crashing.

newbrain, 2021-04-06:
Well, it never happened again, so I cannot really help.
I've flashed the board at least 40 times, from Powershell v7.1.3, both inside VS Code and in a Windows Terminal window.

Some background update we were unlucky to catch? Really have no idea, though.

> and never PowerShell

You should give it a go, at least in Windows.
It's an extremely powerful scripting language, with a rational syntax (sometimes annoyingly verbose) and native object orientation.
It's also available on Linux (and Mac IIRC), but it shines on Windows where one can deeply interact with the system; e.g. I use a script entirely in PS to create and send WoL packets.

flit, 2021-04-10:
@newbrain Thanks for trying to reproduce it. And yeah, linker scripts _are_ hard! I've been pretty curious about PowerShell, it does seem quite capable. And being available for Linux and Mac is nice and impressive.

@thedjnK Do you happen to know what version of PowerShell was being used when the error occurred?

BlokhinDima, 2021-08-13:
Hello. I have the same error. Powershell crashes every time when flashing or erasing process is finished. 
Power Shell Version 5.1.14393.479

thedjnK, 2021-08-16:
@flit Only just noticed the query, it was whatever version was built into windows 10 at the time

HrMitrev, 2021-11-22:
I have the same problem.  Powershell crashes every time pyOCD interracts with my debugger.

```
PS C:\Users\hrmit> Get-Host | Select-Object Version

Version
-------
5.1.22000.282

PS C:\Users\hrmit> pyocd list
  #   Probe           Unique ID
------------------------------------------------
  0   ST-Link Debug   066DFF393238424E43144037
PS C:\Users\hrmit>
[process exited with code 2]
```

elfmimi, 2021-12-05:
I'm experiencing this same problem. it's ok with cmd prompt. but somehow it kills PowerShell with weirdly delayed activation.

now, here are hints I've found.

just `pyocd` or `pyocd --version` is ok. it does no harm.

`pyocd list` or `pyocd json` or `pyocd reset` does do harm.

but `pyocd commader` is ok. list or reset or probably anything works fine in commander mode. and after exiting commander mode with exit command, it still cause no problem. PowerShell won't get killed. it just keeps working fine.

PS Version: 5.1.19041.1320
Python: 3.8.10 or 3.9.9
PyOCD: 0.29.0 or 0.32.1

elfmimi, 2021-12-05:
There are more to this.

`pyocd commander -c exit` is ok.
`pyocd commander -c help` is ok.
`pyocd commander -c list` is not ok.
`pyocd commander -c reset` is not ok.

`pyocd commander -c BOOM` is not ok. 
this one reports
> Error: unrecognized command 'boom'

and then PowerShell will get killed all the same. WHY!?

by the way, executing a sequence of commands like `list;exit` inside the commander prompt is ok.

elfmimi, 2021-12-05:
Turned out testing with [PowerShell 7.2 which is available at MicrosoftStore](https://www.microsoft.com/store/productId/9MZ1SNWT0N5D) gives seemingly useful information.
```
PS C:\Windows\System32> pyocd list
  #   Probe              Unique ID
---------------------------------------------------------------------------
  0   Arm CMSIS-DAP v1   0000008004a0247a00000000000000000000000097969902
PS C:\Windows\System32>
An error has occurred that was not properly handled. Additional information is shown below. The PowerShell process will exit.
Unhandled exception. System.ArgumentOutOfRangeException: Console key values must be between 0 and 255 inclusive. (Parameter 'key')
   at System.ConsoleKeyInfo..ctor(Char keyChar, ConsoleKey key, Boolean shift, Boolean alt, Boolean control)
   at System.ConsolePal.ReadKey(Boolean intercept)
   at System.Console.ReadKey(Boolean intercept)
   at Microsoft.PowerShell.Internal.VirtualTerminal.ReadKey()
   at Microsoft.PowerShell.PSConsoleReadLine.ReadOneOrMoreKeys()
   at Microsoft.PowerShell.PSConsoleReadLine.ReadKeyThreadProc()
   at System.Threading.Thread.StartHelper.Callback(Object state)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.Thread.StartCallback()
```

flit, 2021-12-07:
@elfmimi Thanks for trying to understand this! I don't use either Windows or PowerShell regularly (except for testing), so I have limited knowledge to try and fix it. Fwiw, I tried installing PowerShell 7.2 on macOS 11.6.1, but (of course) `pyocd list` doesn't cause a crash.

The exception log above is both interesting and strange. How does pyocd affect reading a key _after_ it exits? You didn't have to type a key before PowerShell crashed, did you?

The one that's really weird is `pyocd json`. It just does normal Python `print()`s to output.

elfmimi, 2021-12-18:
I've made some progress. I was poking around the source code and noticed that pyocd-pemicro is related in someway.
When I excluded PEMicroProbe from being used by means of modifying the source code, then pyocd worked ok within PowerShell.

To achieve that in moderate , one needs to follow these steps.
- git clone https://github.com/pyocd/pyOCD.git
- remove [this line](https://github.com/pyocd/pyOCD/blob/a7b2f00c8450f8af9c08ce66df7faa223e4699f1/setup.cfg#L54) from pyOCD/setup.cfg
  `pyocd_pemicro>=1.0.6`
- cd pyOCD and then
  `pip install .`

This is not recommended but if you are using pyocd package installed by pip command, here is how to tweak it.
uninstall pyocd-pemicro by `pip uninstall pyocd-pemicro` and then
edit site-packages/pyocd-0.32.2.dist-info/METADATA to remove the following line from it.
```Requires-Dist: pyocd-pemicro (>=1.0.6)```