pyocd/pyOCD #1130: pyocd should return non-zero status on error

ACleverDisguise, 2021-04-03:
pyocd pack --update generates one "ERRO" line in red and dozens of "WARN" lines.  It returns a status of 0 (success).  Errors and warnings should be non-zero for scripting purposes.

ACleverDisguise, 2021-04-03:
This seems to be pack related.  Other errors (including syntax errors in pack command lines) do the right thing, but `pyocd pack --install booger` also fails (for obvious reasons) without setting a correct status.

flit, 2021-04-03:
`pyocd pack --install` not returning non-zero on install failure (and similar for `pyocd pack --find`) is clearly an issue. 

However, `pyocd pack --update` is not so straightforward. The update does ultimately succeed. We could return non-zero status on any pack metadata fetch errors, but there will almost always be some of those (some silicon vendors are quite bad about this!). It's really only an error if the packs _you care about_ are not fetched. Process status codes can't cleanly indicate partial failure.

This is ultimately a problem caused by [cmsis-pack-manager](/pyocd/cmsis-pack-manager) (CPM) not returning any indication of success or failure. We need to extend the CPM API to provide this status before this can be resolved.

Btw, `--update` and `--clean` can be combined with `--find` or `--install` to fetch new metadata prior to looking for and possibly installing a pack.