pyocd/pyOCD #1133: Another non-x86 issue (dependency on pypemicro)

Spivoxity, 2021-04-07:
In addition to the known problem with installing `cmsis-package-manager` on Raspberry Pi, a new problem has appeared in version 0.30.3, caused by the dependency on `pyocd-pemicro` and `pypemicro`.  When `pyocd` is started, it tries to load these modules and the library `unitacmp-64.so` that it uses.  If fails to find the library and crashes.

As far as I can see, the `unitacmp-64.so` library is closed source and not available for Raspi OS.  But in any case, it seems a shame to require it to be installed in order to use some other debug adapter -- in my case, the built in debugging interface of the BBC micro:bit.

Perhaps a solution would be to remove the dependency of `pyocd` on `pyocd-pemicro`, and let those who need it install the add-on by hand.   My workaround is to remove the dependency from the `pyocd` metadata file, like the dependency on `cmsis-package-manager`, before installing.  The GDB server works fine without them.

With best wishes,
-- Mike Spivey

Spivoxity, 2021-04-08:
See the "installing pyocd" section here:

https://github.com/Spivoxity/baremetal-v1/blob/master/setup/install

anthrax-0, 2021-04-08:
I was wondering if one way is for `pyocd-pemicro` to depend on `pyocd` instead of other way around? Then if one needs PEMicro support it can be installed at one go?