pyocd/pyOCD #1138: Errors during installation from a fresh source clone

stephendpmurphy, 2021-04-15:
Hello :wave: 

I am attempting to make some contributions to the project but I can't seem to get the install from source to work. I am building from the master branch via:

```bash
$ cd pyocd
$ python setup.py build
$ sudo python setup.py install
```

The installation runs no problem, however when I execute ```pyocd``` I get the following error.

```bash
Traceback (most recent call last):
  File "/home/stephen/.local/bin/pyocd", line 33, in <module>
    sys.exit(load_entry_point('pyocd', 'console_scripts', 'pyocd')())
  File "/home/stephen/.local/bin/pyocd", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/home/stephen/.local/lib/python3.6/site-packages/importlib_metadata/__init__.py", line 166, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.30.4.dev15-py3.6.egg/pyocd/__main__.py", line 40, in <module>
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.30.4.dev15-py3.6.egg/pyocd/subcommands/pack_cmd.py", line 35, in <module>
  File "/usr/local/lib/python3.6/dist-packages/pyocd-0.30.4.dev15-py3.6.egg/pyocd/subcommands/pack_cmd.py", line 48, in PackSubcommandBase
TypeError: 'type' object is not subscriptable
```

newbrain, 2021-04-15:
Confirmed, you are probably using a python release before 3.9 - I used a 3.8.5 in WSL2, no problems with a 3.9.1 under Windows.

This is due to the use of the type hint in https://github.com/pyocd/pyOCD/blob/5a835e1017182975a11a4caddd83332f74b75fa4/pyocd/subcommands/pack_cmd.py#L48
That specific syntax for set ([`set[str]`](https://docs.python.org/3/library/typing.html#typing.Set) [Generic Alias Type](https://docs.python.org/3/library/stdtypes.html#generic-alias-type)) has been introduced in Python 3.9.

As the main readme states 3.6 or above, this should be replaced with just `set` (or maybe use a TypeVar?).
This seems to be the only instance of `set[str]`, but I'm not sure whether  other similar hints are elsewhere in the code, so I'm not making a PR for this.

flit, 2021-04-15:
Whoops! That would be my fault. 😬 The type annotations used in pyocd should be compatible with 3.6+. However, I'm still getting used to using type annotations and type checking with Python, so I'm bound to make mistakes. I'll fix it right away. It should be replaced with `Set[str]`, using `typing.Set`.

Also need to extend the GitHub Actions workflow to compile all the pyocd code, so language incompatibilities with earlier versions of Python can be detected.

stephendpmurphy, 2021-04-15:
Ahh, thanks for the responses guys. I've updated to Python3.9.4 and was able to build and test my changes successfully :+1: 

Thanks!

PR #1140 (2021-04-15): Fix type annotation in pack subcommand.
The new pack subcommand implementation used `set[str]` as a return type annotation, but this usage of the builtin `set` class is only supported in Python 3.9. Updated to use `typing.Set` instead. Fixes #1138.

Also added `test/import_all.py` that walks over the entire pyocd source tree and imports each module and package. This script is run in the GitHub Actions workflow, to detect any issues such as this in the future.

One additional change was made to `pyocd/probe/stlink/detect/windows.py`. Because `import_all.py` unconditionally imports all modules r [... cut]
closed 2021-04-15 by stephendpmurphy