pyocd/pyOCD #1144: cannot find any libusb even though 2 are installed

nonchip, 2021-04-24:
First things first, OS: `void Linux  5.11.15_1 #1 SMP 1618703302 x86_64 GNU/Linux`
PyOCD `0.30.2` installed through Mbed Studio (`/home/kyra/.config/Mbed\ Studio/mbed-studio-tools/python/bin/pyocd` and so on).

While Mbed Studio correctly identifies (in its target selection etc.) both my STLink (clone) and a board with builtin DFU bootloader (not using the STLink for that one, planning to go directly through the DFU mode), PyOCD doesn't even seem to be able to find a suitable `libusb` (showing a manual run there because mbed suppresses some info, stacktrace is the same tho):

```python
$ /home/kyra/.config/Mbed\ Studio/mbed-studio-tools/python/bin/pyocd cmd -vv
0000388:WARNING:pyusb_backend:CMSIS-DAPv1 probes may not be detected because no libusb library was found.
0000418:WARNING:common:STLink and CMSIS-DAPv2 probes are not supported because no libusb library was found.
0000450:DEBUG:session:Project directory: /home/kyra/dev/Julia/InfraredOverAudio
0000450:CRITICAL:__main__:No backend available
Traceback (most recent call last):
  File "/home/kyra/.config/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/__main__.py", line 402, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/kyra/.config/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/__main__.py", line 808, in do_commander
    PyOCDCommander(self._args, cmds).run()
  File "/home/kyra/.config/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/commands/commander.py", line 65, in run
    if not self.connect():
  File "/home/kyra/.config/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/commands/commander.py", line 166, in connect
    unique_id=self.args.unique_id,
  File "/home/kyra/.config/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/core/helpers.py", line 138, in choose_probe
    allProbes = ConnectHelper.get_all_connected_probes(blocking=blocking, unique_id=unique_id)
  File "/home/kyra/.config/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/core/helpers.py", line 82, in get_all_connected_probes
    allProbes = DebugProbeAggregator.get_all_connected_probes(unique_id=unique_id)
  File "/home/kyra/.config/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/probe/aggregator.py", line 64, in get_all_connected_probes
    probes += cls.get_all_connected_probes(unique_id, is_explicit)
  File "/home/kyra/.config/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/probe/picoprobe.py", line 333, in get_all_connected_probes
    return [cls(dev) for dev in PicoLink.enumerate_picoprobes()]
  File "/home/kyra/.config/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/pyocd/probe/picoprobe.py", line 108, in enumerate_picoprobes
    return [PicoLink(probe) for probe in core.find(find_all=True, custom_match=FindPicoprobe(uid))]
  File "/home/kyra/.config/Mbed Studio/mbed-studio-tools/python/lib/python3.7/site-packages/usb/core.py", line 1299, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
```

but:

```bash
$ ls -l /lib/libusb*
lrwxrwxrwx 1 root root     19 Nov 20  2018 /lib/libusb-0.1.so.4 -> libusb-0.1.so.4.4.4
-rwxr-xr-x 1 root root  22432 Nov 20  2018 /lib/libusb-0.1.so.4.4.4
-rw-r--r-- 1 root root 208486 Apr  2 19:29 /lib/libusb-1.0.a
lrwxrwxrwx 1 root root     19 Apr  2 19:29 /lib/libusb-1.0.so -> libusb-1.0.so.0.3.0
lrwxrwxrwx 1 root root     19 Apr  2 19:29 /lib/libusb-1.0.so.0 -> libusb-1.0.so.0.3.0
-rwxr-xr-x 1 root root 116824 Apr  2 19:29 /lib/libusb-1.0.so.0.3.0
-rw-r--r-- 1 root root  20626 Nov 20  2018 /lib/libusb.a
lrwxrwxrwx 1 root root     19 Nov 20  2018 /lib/libusb.so -> libusb-0.1.so.4.4.4
lrwxrwxrwx 1 root root     23 Nov 10 04:19 /lib/libusbmuxd-2.0.so.6 -> libusbmuxd-2.0.so.6.0.0
```
(with /lib, /usr/lib, ... all being symlinked together on this distro)

`libusb-1.0` being one of the names `usb.backend.libusb1` explicitely tries:
```python
def _load_library(find_library=None):
    #...
    return usb.libloader.load_locate_library(
                ('usb-1.0', 'libusb-1.0', 'usb'),
                'cygusb-1.0.dll', 'Libusb 1',
                win_cls=win_cls,
                find_library=find_library, check_symbols=('libusb_init',))
```

flit, 2021-04-24:
First thing, please update to pyocd 0.30.3. This fixes the first issue with the picoprobe related exception raised when it can't find libusb.

Short answer: I don't have any immediate ideas… 😓 pyusb has always "just worked" on every Linux system I've used. (And I'm not really much of a Linux user, mostly only using it through Docker on macOS.)

What distro are you using? I've always used Debian based distros, primarily Ubuntu. Though I know others have used pyocd successfully only other distros (Alpine comes to mind, there were a few others too).

Since pyusb works for you with another tool, let's look at what could be different between running that tool and pyocd.

- Python executable and version?
- Version of pyusb?
- `LD_LIBRARY_PATH` (or equivalent)?
- Anything else? As you've indicated, 32/64-bit shouldn't matter.

cederom, 2021-04-24:
I have observed similar problem on FreeBSD for some time/reason, reinstalling system dependencies and python modules helped :-) Let me know if that helps :-)

nonchip, 2021-04-24:
> First thing, please update to pyocd 0.30.3.

afraid i'd have to ask the mbed people to do that probably

> This fixes the first issue with the picoprobe related exception raised when it can't find libusb.

i mean letting it find libusb would "fix" the issue for me too, and pretty sure i need libusb to work anyway to have it talk to the device via dfu over usb

> What distro are you using?

void linux. used pyocd in mbed successfully a while ago on this very machine+os, they did an update to their IDE since though, probably that's what broke it.

> let's look at what could be different between running that tool and pyocd.

that's the thing, i've been running it from both the ide (with whatever wrapping magic like `LD_LIBRARY_PATH` they might be doing there) and the terminal (so it's "just" the different wrapper), same difference. so it's probably something related to either the libs delivered with that or they're doing some more advanced wrapper magic

> Version of pyusb?
```
$ cat ~/.config/Mbed\ Studio/mbed-studio-tools/python/lib/python3.7/site-packages/usb/_version.py
# coding: utf-8
# file generated by setuptools_scm
# don't change, don't track in version control
version = '1.1.1'
```

flit, 2021-04-24:
> void linux. 

😄  I saw "void Linux" in the uname output you provided, but didn't realise that was the distro.

> that's the thing, i've been running it from both the ide (with whatever wrapping magic like `LD_LIBRARY_PATH` they might be doing there) and the terminal (so it's "just" the different wrapper), same [...]

I'm not sure if running the tools included with Mbed Studio directly from a shell is the same as running from the IDE. This might be more of a packaging issue than specifically related to pyocd; at this point it's not clear.

Can you try installing pyocd outside of Mbed Studio and see if that can find libusb? Assuming you already have Python 3.6 or above installed at the system level, you can run this to install pyocd (run from some directory like your home):

```sh
python3 -m venv pyocd-venv
source pyocd-venv/bin/activate
pip install pyocd
pyocd list
```

This creates a virtualenv for pyocd, activates the virtualenv, installs pyocd, and lists connected boards. (You can get out of the virtualenv by executing `deactivate`.)

nonchip, 2021-04-25:
> ```shell
> python3 -m venv pyocd-venv
> source pyocd-venv/bin/activate
> pip install pyocd
> pyocd list
> ```

```
Successfully installed appdirs-1.4.4 capstone-4.0.2 cffi-1.14.5 cmsis-pack-manager-0.2.10 colorama-0.4.4 future-0.18.2 intelhex-2.3.0 intervaltree-3.1.0 milksnake-0.1.5 naturalsort-1.5.1 prettytable-2.1.0 psutil-5.8.0 pycparser-2.20 pyelftools-0.27 pylink-square-0.10.0 pyocd-0.30.3 pyocd-pemicro-1.0.4 pypemicro-0.1.6 pyusb-1.1.1 pyyaml-5.4.1 six-1.15.0 sortedcontainers-2.3.0 wcwidth-0.2.5

(pyocd-venv) 
# kyra @ firefly in ~/pyocdtest [7:23:08] 
$ pyocd list
No available debug probes are connected
```

cederom, 2021-05-11:
Well, I got this `pyocd.probe.pydapaccess.dap_access_api.DeviceError: No USB backend found` issue again on a fresh FreeBSD workstation, looks like something to investigate and fix, as all other applications work with USB with no problem :-)

```
(venv37zephyr) pyocd
Traceback (most recent call last):
  File "/home/x/usr/local/venv37zephyr/bin/pyocd", line 11, in <module>
    load_entry_point('pyocd==0.30.4.dev30', 'console_scripts', 'pyocd')()
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 490, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2862, in load_entry_point
    return ep.load()
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2462, in load
    return self.resolve()
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2468, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev30-py3.7.egg/pyocd/__init__.py", line 21, in <module>
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev30-py3.7.egg/pyocd/gdbserver/__init__.py", line 17, in <module>
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev30-py3.7.egg/pyocd/gdbserver/gdbserver.py", line 49, in <module>
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev30-py3.7.egg/pyocd/commands/commander.py", line 22, in <module>
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev30-py3.7.egg/pyocd/core/helpers.py", line 19, in <module>
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev30-py3.7.egg/pyocd/probe/aggregator.py", line 84, in <module>
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev30-py3.7.egg/pyocd/core/plugin.py", line 88, in load_plugin_classes_of_type
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2462, in load
    return self.resolve()
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2468, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev30-py3.7.egg/pyocd/probe/cmsis_dap_probe.py", line 24, in <module>
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev30-py3.7.egg/pyocd/probe/pydapaccess/__init__.py", line 17, in <module>
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
[... cut ...]
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev30-py3.7.egg/pyocd/probe/pydapaccess/interface/__init__.py", line 62, in <module>
pyocd.probe.pydapaccess.dap_access_api.DeviceError: No USB backend found

(venv37zephyr) uname -a
FreeBSD hexagon 12.2-RELEASE-p6 FreeBSD 12.2-RELEASE-p6 GENERIC  amd64
```