pyocd/pyOCD #1150: Pyocd Pack Update Error

SputnikTheMighty, 2021-05-11:
I am following the instructions here to install a pack for for STM32G071RB:
https://docs.zephyrproject.org/latest/boards/arm/nucleo_g071rb/doc/index.html

Running `pyocd pack -u` gives the error below:

```000289:INFO:__main__:Updating pack index...
May 11 12:25:04.188 WARN invalid utf-8 sequence of 1 bytes from index 55
May 11 12:25:04.189 WARN invalid utf-8 sequence of 1 bytes from index 55
May 11 12:25:04.189 WARN Only Comment, CDATA and DOCTYPE nodes can start with a '!'
May 11 12:25:04.237 WARN error while parsing attribute at position 13: Attribute value must start with a quote
May 11 12:25:04.294 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.294 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.294 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.294 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.294 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.297 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.300 WARN Only Comment, CDATA and DOCTYPE nodes can start with a '!'
May 11 12:25:04.308 WARN Only Comment, CDATA and DOCTYPE nodes can start with a '!'
May 11 12:25:04.318 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.351 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.351 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.351 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.363 WARN invalid digit found in string, Package: TLE984x_DFP, Vendor: Infineon
May 11 12:25:04.378 WARN Only Comment, CDATA and DOCTYPE nodes can start with a '!'
May 11 12:25:04.403 WARN Unknown file catogory preIncludeLocal, Group: Unity, Class: Test, in: Component, Package: Unity, Vendor: Arm-Packs
May 11 12:25:04.415 WARN Only Comment, CDATA and DOCTYPE nodes can start with a '!'
May 11 12:25:04.426 WARN No Core found!, Package: CMSIS, Vendor: ARM
May 11 12:25:04.438 WARN No Core found!, Package: V2M_MPS2_SSE_300_BSP, Vendor: ARM
May 11 12:25:04.445 WARN Unknown file catogory preIncludeLocal, Group: SPM HAL, Class: TFM Platform, in: Component, Package: Musca-S1_DFP, Vendor: ARM
May 11 12:25:04.446 WARN Unknown file catogory preIncludeLocal, Group: System, Class: TFM Platform, in: Component, Package: Musca-S1_DFP, Vendor: ARM
May 11 12:25:04.446 WARN Unknown file catogory preIncludeLocal, Group: Test, Class: TFM Platform, in: Component, Package: Musca-S1_DFP, Vendor: ARM
May 11 12:25:04.446 WARN Unknown file catogory preIncludeLocal, Group: Boot HAL, Class: TFM Platform, in: Component, Package: Musca-S1_DFP, Vendor: ARM
May 11 12:25:04.523 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.524 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.524 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.524 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.543 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.543 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.543 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.583 WARN expecting </link> found </head>
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.799 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.800 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.800 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.800 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.800 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.800 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.801 WARN the XML is invalid, an element was wrongly closed
May 11 12:25:04.802 WARN Unknown file catogory preIncludeLocal, Group: Bootloader, Class: TFM, in: Component, Package: TFM, Vendor: ARM
May 11 12:25:04.802 WARN Unknown file catogory preIncludeLocal, Group: Core, Class: TFM, in: Component, Package: TFM, Vendor: ARM
May 11 12:25:04.802 WARN Unknown file catogory preIncludeLocal, Group: Core, Class: TFM, in: Component, Package: TFM, Vendor: ARM
May 11 12:25:04.802 WARN Unknown file catogory preIncludeLocal, Group: Secure Service, Class: TFM, in: Component, Package: TFM, Vendor: ARM
May 11 12:25:04.802 WARN Unknown file catogory preIncludeLocal, Group: Secure Service, Class: TFM, in: Component, Package: TFM, Vendor: ARM
May 11 12:25:04.802 WARN Unknown file catogory preIncludeLocal, Group: Secure Service, Class: TFM, in: Component, Package: TFM, Vendor: ARM
May 11 12:25:04.802 WARN Unknown file catogory preIncludeLocal, Group: Secure Service, Class: TFM, in: Component, Package: TFM, Vendor: ARM
[... 61 lines cut ...]
May 11 12:25:04.845 WARN No Core found!, Package: V2M_MPS3_SSE_300_BSP, Vendor: ARM
May 11 12:25:04.845 WARN Unknown file catogory preIncludeGlobal, Group: Startup, Class: Device, in: Component, Package: V2M_MPS3_SSE_300_TFM-PF, Vendor: ARM
May 11 12:25:04.846 WARN Unknown file catogory preIncludeLocal, Group: SPM HAL, Class: TFM Platform, in: Component, Package: V2M_MPS3_SSE_300_TFM-PF, Vendor: ARM
May 11 12:25:04.846 WARN Unknown file catogory preIncludeLocal, Group: System, Class: TFM Platform, in: Component, Package: V2M_MPS3_SSE_300_TFM-PF, Vendor: ARM
May 11 12:25:04.846 WARN Unknown file catogory preIncludeLocal, Group: Test, Class: TFM Platform, in: Component, Package: V2M_MPS3_SSE_300_TFM-PF, Vendor: ARM
May 11 12:25:04.846 WARN Unknown file catogory preIncludeLocal, Group: Boot HAL, Class: TFM Platform, in: Component, Package: V2M_MPS3_SSE_300_TFM-PF, Vendor: ARM
May 11 12:25:04.913 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas
May 11 12:25:04.913 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas
May 11 12:25:04.913 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas
May 11 12:25:04.913 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas
May 11 12:25:04.913 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas
May 11 12:25:04.913 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas
May 11 12:25:04.913 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas
May 11 12:25:04.913 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas
May 11 12:25:04.913 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas
May 11 12:25:04.914 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas
May 11 12:25:04.914 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas
May 11 12:25:04.914 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas
May 11 12:25:04.914 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas
May 11 12:25:04.914 WARN cannot parse integer from empty string, Package: RA_DFP, Vendor: Renesas
May 11 12:25:04.920 WARN tried to parse element "package" from element "head"

running `pyocd pack --install stm32g071rb` gives:

```0000414:WARNING:__main__:No matching devices. Please make sure the pack index is up to date.```

rotu, 2021-05-13:
I got this too. AFAICT it still worked, but so many warnings made me think I messed things up very badly and is very confusing.

rotu, 2021-05-14:
Also, I think the pack is "[STMicroelectronics STM32G0 Series Device Support](https://keilpack.azureedge.net/pack/Keil.STM32G0xx_DFP.1.3.0.pack)" from this page
https://www.keil.com/dd2/pack/ but I don't know (1) why this isn't in the index (2) the right way to install a pack which isn't in the index.

flit, 2021-05-15:
There are several things going on here. Lots of packs from multiple vendors do really have errors and badly formatted data. The underlying cmsis-pack-manager (CPM) project unfortunately blasts out too many warnings. If there were only a few pack errors it would be ok, but it needs to be turned down a notch given how many errors there are. Related to this is that CPM doesn't understand a `preIncludeLocal` keyword. A fix for that is implemented, but we're having to totally rebuild the CPM wheel building and release CI after Travis CI changed their terms of service.

Regarding the G071 pack issue, I am able to reproduce the failure to find it stm32g071rb with pyocd 0.30.3. Interestingly, it _does_ find and install successfully with the latest in-development versions of pyocd + CPM. Not sure why the difference—the core logic of CPM hasn't changed.

To work around this, you can download the [G071 .pack](https://keilpack.azureedge.net/pack/Keil.STM32G0xx_DFP.1.3.0.pack) (as linked above) and use it with pyocd by passing `--pack <path to .pack>` on the command line. Full documentation for how to use manually downloaded packs is in the [target support docs](https://github.com/pyocd/pyOCD/blob/master/docs/target_support.md).

str4t0m, 2022-08-27:
Installing the `Keil.STM32G0xx_DFP` work since a while, so I believe this issue could be closed.

closed 2022-10-22 by flit