pyocd/pyOCD #1159: FreeBSD: ImportError: cannot import name 'ENODATA' from 'errno' (unknown location)

cederom, 2021-05-13:
If anyone encounters this problem blocking `pyocd flash` on Unix platform this comes from `pyusb` and the fix is already proposed here https://github.com/pyusb/pyusb/issues/365 :-) With this fix applied flashing works fine :-)

```
(venv37zephyr) pyocd flash build/zephyr/zephyr.hex
0001742:CRITICAL:__main__:uncaught exception: cannot import name 'ENODATA' from 'errno' (unknown location)
Traceback (most recent call last):
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev31-py3.7.egg/pyocd/__main__.py", line 150, in run
    status = cmd.invoke()
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev31-py3.7.egg/pyocd/subcommands/flash_cmd.py", line 86, in invoke
    with session:
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev31-py3.7.egg/pyocd/core/session.py", line 343, in __enter__
    self.open()
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev31-py3.7.egg/pyocd/core/session.py", line 457, in open
    self._probe.open()
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev31-py3.7.egg/pyocd/probe/cmsis_dap_probe.py", line 170, in open
    self._link.open()
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev31-py3.7.egg/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev31-py3.7.egg/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 635, in open
    self._interface.open()
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev31-py3.7.egg/pyocd/probe/pydapaccess/interface/pyusb_backend.py", line 102, in open
    if dev.is_kernel_driver_active(interface_number):
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyusb-1.1.1.post7+g90d3e05-py3.7.egg/usb/core.py", line 1107, in is_kernel_driver_active
    interface)
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyusb-1.1.1.post7+g90d3e05-py3.7.egg/usb/backend/libusb0.py", line 624, in is_kernel_driver_active
    from errno import ENODATA
ImportError: cannot import name 'ENODATA' from 'errno' (unknown location)

(venv37zephyr) uname -a
FreeBSD hexagon 12.2-RELEASE-p6 FreeBSD 12.2-RELEASE-p6 GENERIC  amd64

(venv37zephyr) which python
/home/x/usr/local/venv37zephyr/bin/python

(venv37zephyr) python --version
Python 3.7.10
```

cederom, 2021-05-13:
Well its not that simple. Turns out to be OS dependent value :-(

https://github.com/pyusb/pyusb/pull/366

Work in progress..

cederom, 2021-07-09:
Hello @flit the pyUSB 1.2.0 and today 1.2.1 was released with the fix, can we use this version as default in pyOCD? :-)

https://github.com/pyusb/pyusb/releases

flit, 2021-07-11:
Done! 😄

flit, 2021-07-11:
Please close this issue if it's fixed.

cederom, 2021-07-11:
TANK U @flit =)

closed 2021-07-11 by cederom