pyocd/pyOCD #1160: development dependencies issues

cederom, 2021-05-14:
Hello world :-)

I have noticed that recent pyOCD handles dependencies strictly. That causes problems when solving / developing multiple other dependencies - for instance I am now fixing the `pyocd-pemicro` / `pypemicro` and `pyusb` on BSD platforms. Installing test / development versions of these dependencies causes problems with running and testing pyOCD.

```
pkg_resources.ContextualVersionConflict: (pyocd 0.8.1a2.dev1620 (/home/user/venv38openbsd/lib/python3.8/site-packages/pyocd-0.8.1a2.dev1620-py3.8.egg), Requirement.parse('pyocd>=0.30.0'), {'pyocd-pemicro'})
```

flit, 2021-05-15:
Hi @cederom! 😄 

It's a tricky situation. I ended up using very specific dependency versions for certain packages after experiencing a number of bugs caused by new versions of downstream dependencies. With packages that have versions >=1.0, you can rely on semantic versioning. But versions <1.0 may make incompatible changes with a minor version bump (at least, that's the reality).

If you have specific suggestions for improving the situation, I'd like to hear them!

cederom, 2021-05-15:
Hello @flit :-) Maybe some sort of `development` switch just to disable dependencies for testing? I don't know :-) :-)

flit, 2021-05-24:
I reviewed the dependency specifications again, and it looks pretty good.

```
    capstone>=4.0,<5.0
    cmsis-pack-manager>=0.2.10
    colorama<1.0
    hidapi; platform_system != "Linux"
    intelhex>=2.0,<3.0
    intervaltree>=3.0.2,<4.0
    naturalsort>=1.5,<2.0
    prettytable>=2.0,<3.0
    pyelftools<1.0
    pylink-square>=0.8.2,<1.0
    pyocd_pemicro>=1.0.0.post2
    pyusb>=1.1.0,<2.0
    pyyaml>=5.1,<6.0
    six>=1.15.0,<2.0
```

These are all specified about as loosely as can be while still being stable (assuming SemVer is followed). All dependencies are allowed to be upgraded up to the next major revision, except `hidapi` and `pyocd_pemicro` that have no upper limit.

My strong recommendation is to use a Python virtual environment to isolate packages. This still doesn't completely resolve potential issues, but is a major improvement. I'm even planning to update the install docs to recommend using pipx.

Fyi, pyocd will be moving to [PEP 517](https://www.python.org/dev/peps/pep-0517/) sometime soon. Everything currently passed as parameters to `setup()` will move to static `pyproject.toml` and `setup.cfg` files. (The only reason I haven't switched yet is it's not clear to me how to handle the zipping of SVD files currently done in `setup.py`.)

cederom, 2022-01-24:
Hello @flit :-) Just a quick question, as Capstone will finally make a release soon that fixes out-of-the-box build on BSD, will 5.0 be compatible with 4.0 or would that require some additional work in pyOCD ? :-)

https://github.com/capstone-engine/capstone/pull/1750