pyocd/pyOCD #1163: "pyocd pack --install stm32l4a" installs "Keil::STM32L4xx_DFP::2.5.0" instead of "Keil::STM32L4xx_DFP::2.2.0". Because of this I get error "Target type 'stm32l4a6vg' not recognized."

sternocleidomastoid, 2021-05-20:


jeromecoutant, 2021-05-20:
v2.5.0 is better than v2.2.0 :-)

sternocleidomastoid, 2021-05-20:
I edited the title to be more accurate. Yes, it may be better but in my opinion it should not install automatically if it doesn't support the device supplied.

jeromecoutant, 2021-05-20:
````
$ grep 'device Dname="STM32L4A6VG' Keil.STM32L4xx_DFP.2.5.0.pdsc
        <device Dname="STM32L4A6VGTx">
        <device Dname="STM32L4A6VGTxP">
        <device Dname="STM32L4A6VGYx">
        <device Dname="STM32L4A6VGYxP">
````
try --install stm32l4a6vgtx

sternocleidomastoid, 2021-05-20:
ver 2.5.0 already supports stm32l4a6vgtx so I just changed my target to stm32l4a6vgtx instead of stm32l4a6vg to make it work. thank you so much for your help! I guess I should have understood the naming convention for stm32 first.

closed 2021-05-20 by sternocleidomastoid

flit, 2021-05-24:
The name change to add the `Tx` and other suffixes makes _partial_ sense in that _some_ STM32 devices have more than just `Tx` suffix. But why almost all of the target names were changed is unclear (and not all have been changed).

Either way, pyocd (or more accurately cmsis-pack-manager) will always use the latest available pack version, and it performs a contains search. There is not a clear logical case for searching old pack versions, as it could result in unexpected behaviour (finding incorrectly named devices fixed in newer versions), and would require maintaining metadata (PDSC files) for all existent pack versions, thus exploding the index size. You can always use `pyocd pack --find` to search for the part numbers you're looking for.