pyocd/pyOCD #1164: Which python hidapi package is used

LeeChunHei, 2021-06-03:
I would like to ask where can I install that the correct python hid API package to use the hidapi backend.
Right now I installed [this python package](https://github.com/apmorton/pyhidapi) via pip.
However, when I run pyocd, it warned me that module 'hid' has no attribute 'device'.
I looked at [pyhidapi](https://github.com/apmorton/pyhidapi) source code and noticed no attribute called device, but there is an attribute called Device.
Did I install the wrong python package or there is some typo in pyOCD when using hidapi backend?
It not likely [pyhidapi](https://github.com/apmorton/pyhidapi) changed its attribute naming as when I use git blame, the Device attribute is right there at the very beginning.

closed 2021-06-03 by LeeChunHei

LeeChunHei, 2021-06-03:
Sorry, I think I use the wrong one before.
I should use this one https://github.com/trezor/cython-hidapi

flit, 2021-06-08:
Yup, `cython-hidapi` is the one. The package name on PyPI is simply 'hidapi'. If you install pyocd via pip either from PyPI or locally from the git repo, you should get all the necessary dependencies and not have to install them manually.