pyocd/pyOCD #1187: Timeout waiting for gdb server to start

avotolato, 2021-07-20:
I am trying to debug a board with an LPC1768 processor with MBED Studio 1.4.1 but every time I try to start the debugger I get the message "Timeout waiting for gdb server to start". Yet, when I start the debugger with "pyocd --target lpc1768" it starts and keeps running. I looked up the problem and they suggested trying to run pyocd with a supplied pyocd.yaml file. So I did that and got the following results:

[pyocd_log.txt](https://github.com/pyocd/pyOCD/files/6848815/pyocd_log.txt)

Can you help me figure this out?

avotolato, 2021-07-20:
Sorry, I meant when I start the server with "pyocd gdb --target lpc1768" it keeps running.

cederom, 2021-07-20:
Did you just ask the same question 6 times in 6 separate Issues? :nerd_face:

avotolato, 2021-07-20:
> I hope not. I did have a few submits that did not seem to go through
> though.

I got a "took too long to submit" error on each of them.

flit, 2021-08-02:
Closing as duplicate

closed 2021-08-02 by flit