pyocd/pyOCD #1192: DBGMCU register addresses.

andrew-rogers, 2021-08-11:
https://github.com/pyocd/pyOCD/blob/a064d8fb99c44f32123e51d91331859b9030b5c0/pyocd/target/builtin/target_STM32L475xx.py#L29

Should the register address for APB1FZR2 be 0xE004200C?

APB1FZR2 = 0xE004200C

and

APB2FZR = 0xE0042010

I am attempting to use this driver for an STM32L496VGT3 which fails to load the flash even if I make the changes but that might be because of some difference between STM32L475 and STM32L496 so I can't confirm the above with a test. If anyone has a working STM32L496 setup I would be interested to know how you setup pyOCD for it.

flit, 2021-08-11:
Indeed you are correct. Whoops! Fixed in #1193.

For the L496, have you tried the the CMSIS pack? It doesn't write the DBGMCU registers out of the box, but the flash algo should work.

The best currently supported option for configuring DBGMCU register is a [user script](https://github.com/pyocd/pyOCD/blob/main/docs/user_scripts.md). There is an example of writing `DBGMCU_CR`, though it would probably be better to use the `did_connect()` hook instead of `did_reset()`.

For auto config of DBGMCU registers with CMSIS pack targets (and any other custom debug logic specified by a pack), I'm working on [debug sequence](https://arm-software.github.io/CMSIS_5/develop/Pack/html/debug_description.html) support. This is on the [feature/debug_sequences](https://github.com/flit/pyOCD/tree/feature/debug_sequences) branch of my fork. It's not ready for upstreaming yet, but is complete enough to support STM32 sequences to write the DBGMCU registers. If you do try it out, please let me know if you run into any issues!

andrew-rogers, 2021-08-11:
Many thanks for the fast response.
I will try the CMSIS pack. I must confess I did not read the target_support.md file and was no aware of the CMSIS pack support. Many thanks for pointers.

flit, 2021-08-11:
No worries! I realise the docs aren't all that accessible (or even remotely complete ☹️ ).

flit, 2021-08-11:
If you have questions about the L476, let's move to either a [discussion](https://github.com/pyocd/pyOCD/discussions) topic, or new issue if you run into actual bugs (except on the feature/debug_sequences branch if you try that—those are important for me to know about but somewhat expected so only warrant a discussion).

PR #1193 (2021-08-11): Fix STM32L475 DBGMCU register addresses
Fixes #1192
closed 2021-08-11 by flit