pyocd/pyOCD #1200: Fail to program big files on stm32h743

nvincent-vossloh, 2021-08-23:
Hello,

I'm trying to load a firmware on a `nucleo_h743zi` using `pyocd`.
When the firmware size is small (blinky, hello world, ~20kB) everything is fine: my firmware is downloaded onto target and runs smoothly.
However, if the firmware gets big (tried with 256k), `pyocd` says that it programmed the file correctly, however the firmware does not run or crash (fault trace from zephyr OS).

I dumped the content of my flash and saw that some portions of the flash was left unprogrammed (0xff) or set to 0.

When using `openocd` or `STM32_Programmer` I do not have that issue.

I tried to increase the size of an hello-world by adding random data at the end and tried programming it with `STM32_Programmer_CLI` (it works) and with `pyocd` (does not work).

Here are the various command line used:
- Grow small binary file (`hello.bin`):
```
head -c 200k < /dev/urandom > random.bin
cat hello.bin random.bin >> big_hello.bin
```
- programming firmware: 
```
pyocd flash -e sector -a 0x8000000 -t stm32h743zitx -O reset_type=hw  --pack Keil.STM32H7xx_DFP.2.7.0.pack big_hello.bin 
STM32_Programmer_CLI -c port=SWD -d big_hello.bin 0x08000000 -rst
```
- Dumping memory content:
```
pyocd commander -c "savemem 0x08000000 220712 dump_with_cmder.bin"
STM32_Programmer_CLI -c port=SWD  -u 0x08000000 220712 dump_from_stm32prog.bin
```

Memory read from the devices is always the same regardless of the method.
Memory written to the device is not correct when using `pyocd`.

Let me know if you need additional information.

Thanks.
ps: by now, I also get errors when programming small firmwares.

flit, 2021-08-23:
Thanks for the detailed report!

I don't have an H743, but will try to replicate on my H747.

JackNewman12, 2021-09-05:
Can confirm I see this too on a Nucleo-H743ZI2. Identical behavior.

jeromecoutant, 2021-09-06:
Hi
Make sure you are using the latest ST Link FW: https://www.st.com/en/development-tools/stsw-link007.html

nvincent-vossloh, 2021-09-06:
> Hi
> Make sure you are using the latest ST Link FW: https://www.st.com/en/development-tools/stsw-link007.html

Hi, I'm using V3J7M2 version of ST Link firmware. upgrade tool list it as the latest.

jeromecoutant, 2021-09-06:
> Hi, I'm using V3J7M2 version of ST Link firmware. upgrade tool list it as the latest.

Please upgrade firmware to the version V3J8M3

nvincent-vossloh, 2021-09-06:
Ok done (I needed to first upgrade stsw-link007 software to be able to see latest firmware...), and I have the same behavior of unwritten datas, and as @flit mentionned, data restart at a page boundary.

nvincent-vossloh, 2022-04-07:
Hello, quick update:

I tested with latest `Keil.STM32H7xx_DFP.3.0.0.pack` and I'm still having the same issue.

I tried to have a look into pyocd code to see what was going on (did not went very far, this level of code is beyond my knowledge), I could verify that what is written in memory with `write_memory_block8` in `load_page_buffer` seems to be properly written into memory since I could call `read_memory_block8` and compare read data versus written.

If you have some pointers for me to look in the "good" direction I may have some time to spent on this issue.

Thanks

flit, 2022-04-21:
Thanks for the update, and for the willingness to help out. This one has been on my priority list for too long.

I've been working on a script to fully validate an FLM-type flash algo, without the complexity of the full pyocd flash programming system. This should help narrow down the problem, at least.

https://github.com/flit/pyOCD/blob/feature/test_flash_algo_script/scripts/test_flash_algo.py

Right now it's mostly focused on verifying erases, but does check page programming incidentally. It also doesn't use double-buffered programming (writing the next page to RAM whilst another page is being programmed to flash by the target's CPU).

An initial test just now against the H743 passed all tests. Interesting, but not conclusive. 😄 

Another thing that would be helpful if you have time is to compare against [MDK-Community](https://www2.keil.com/mdk5/editions/community). Pyocd uses the same CMSIS-Pack flash algos, so it will be another good comparison to narrow down the cause.

Also, if you have a failing firmware image you could attach or send via email (see my GH profile) or Slack (see the links on [pyocd.io](https://pyocd.io)), that would be very useful.

Thanks!

nvincent-vossloh, 2022-04-29:
[Blinky_MDK.tar.gz](https://github.com/pyocd/pyOCD/files/8589763/Blinky_MDK.tar.gz)
Here is a Blinky in various formats (axf, hex, bin) generated from uVision (MDK/KEIL). With MDK I had no issue programming onto the target. With `pyocd` it fails to program properly the firmware.

If you need me to run `test_flash_algo.py` could you please provide a valid command line ?

Thanks :slightly_smiling_face:

flit, 2022-04-29:
@nvincent-vossloh Thank you for the test files!

I've run `test_flash_algo.py` and it passes. However, it doesn't directly test programming yet, it currently does that as a side effect of testing sector erases. If you would like to run it, you need to extract the STM32H7xx CMSIS-Pack as a zip file (download from [here](https://www.keil.com/dd2/pack/), change extension to `.zip` and extract). The flash algo to use is `CMSIS/Flash/STM32H7x_2048.FLM`. Pass the path to this FLM to `test_flash_algo.py` as a positional arg. For the H7xx you need to set `--ram=0x24000000` to use a large enough RAM.

I talked with the Keil MDK team, and they don't have any reports of H7xx flash programming issues.

All that, and your testing with MDK (thanks!) suggest that the issue can be fixed in pyocd itself. Which is good, since this means it's under our direct control.