pyocd/pyOCD #1202: Microchip (/Atmel) nEDBG CMSIS-DAP fail on a Curiosity Nano samd21 board.

mws-rmain, 2021-08-24:
I've made some progress getting pyocd to connect to the nEDBG on a 'Curiosity Nano SAMD21' development board, but pyocd still fails when it actually tries to connect. Please provide any suggestions / guidance you can.

Here is what I've done so far:
1. I added below to end of file '50-cmsis-dap.rules' (which was copied to '/etc/udev/rules.d'):
```
# 03eb:2175 Microchip (/Atmel) nEDBG CMSIS-DAP
SUBSYTEM="usb", Attr{idVendor}=="03eb", ATTR{idProduct}=="2175", MODE:="666"
```
(NOTE: I also added the relevant output of lsusb, and dmesg for detail at the end of this post.)

command 'pyocd list' now reports:
```
  #   Probe                                               Unique ID             
--------------------------------------------------------------------------------
  0   Microchip Technology Incorporated nEDBG CMSIS-DAP   MCHP3313031800000362  
```

The msc_cdc example builds fine, but pyocd is still unable to connect to the nEDBG CMSIS-DAP to flash the program:
```
make BOARD=curiosity_nano_samd21 flash
pyocd flash -t atsamd21g17a _build/curiosity_nano_samd21/cdc_msc.hex
0001513:CRITICAL:__main__:DAP_CONNECT failed
Traceback (most recent call last):
  File "/home/mainr/.local/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 217, in connect
    self._link.connect(port)
  File "/home/mainr/.local/lib/python3.8/site-packages/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/home/mainr/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 731, in connect
    actual_port = self._protocol.connect(port.value)
  File "/home/mainr/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/cmsis_dap_core.py", line 218, in connect
    raise DAPAccessIntf.CommandError("DAP_CONNECT failed")
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.CommandError: DAP_CONNECT failed

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/mainr/.local/lib/python3.8/site-packages/pyocd/__main__.py", line 150, in run
    status = cmd.invoke()
  File "/home/mainr/.local/lib/python3.8/site-packages/pyocd/subcommands/flash_cmd.py", line 86, in invoke
    with session:
  File "/home/mainr/.local/lib/python3.8/site-packages/pyocd/core/session.py", line 342, in __enter__
    self.open()
  File "/home/mainr/.local/lib/python3.8/site-packages/pyocd/core/session.py", line 460, in open
    self._board.init()
  File "/home/mainr/.local/lib/python3.8/site-packages/pyocd/board/board.py", line 86, in init
    self.target.init()
  File "/home/mainr/.local/lib/python3.8/site-packages/pyocd/core/soc_target.py", line 117, in init
    seq.invoke()
  File "/home/mainr/.local/lib/python3.8/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/home/mainr/.local/lib/python3.8/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/home/mainr/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 376, in _connect
    connector.connect(self._protocol)
  File "/home/mainr/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 163, in connect
    self._connect_probe(protocol)
  File "/home/mainr/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 183, in _connect_probe
    self._probe.connect(protocol)
  File "/home/mainr/.local/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 219, in connect
    raise self._convert_exception(exc) from exc
pyocd.core.exceptions.ProbeError: DAP_CONNECT failed
make: *** [../../rules.mk:173: flash-pyocd] Error 1
```

To setup the board for the build, under directory 'tinyusb/hw/bsp/samd21/boards', I created directory curiosity_nano, which has three files:
board.h:
```
/* 
 * The MIT License (MIT)
 *
[... 416 lines cut ...]

command 'dmesg' (relevant output):
```
[12091.316386] usb 2-2: new full-speed USB device number 4 using ohci-pci
[12091.898736] usb 2-2: New USB device found, idVendor=03eb, idProduct=2175, bcdDevice= 1.00
[12091.898743] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[12091.898748] usb 2-2: Product: nEDBG CMSIS-DAP
[12091.898751] usb 2-2: Manufacturer: Microchip Technology Incorporated
[12091.898755] usb 2-2: SerialNumber: MCHP3313031800000362
[12091.926337] hid-generic 0003:03EB:2175.0003: hiddev0,hidraw1: USB HID v1.11 Device [Microchip Technology Incorporated nEDBG CMSIS-DAP] on usb-0000:00:06.0-2/input0
[12091.926487] cdc_acm 2-2:1.1: ttyACM0: USB ACM device
[12091.949567] usb-storage 2-2:1.4: USB Mass Storage device detected
[12091.949711] scsi host4: usb-storage 2-2:1.4
[12092.983690] scsi 4:0:0:0: Direct-Access     MCHP     Curiosity Nano   1.00 PQ: 0 ANSI: 3
[12092.983924] sd 4:0:0:0: Attached scsi generic sg2 type 0
[12093.013095] sd 4:0:0:0: [sdb] 2048 512-byte logical blocks: (1.05 MB/1.00 MiB)
[12093.042328] sd 4:0:0:0: [sdb] Write Protect is on
[12093.042334] sd 4:0:0:0: [sdb] Mode Sense: 0f 00 80 00
[12093.071609] sd 4:0:0:0: [sdb] No Caching mode page found
[12093.071616] sd 4:0:0:0: [sdb] Assuming drive cache: write through
[12093.148421]  sdb:
[12093.271401] sd 4:0:0:0: [sdb] Attached SCSI removable disk
```
I've also attached a wireshark packet capture of USB traffic during the 'make flash' command which invokes pyocd.
[pyocd_cmsisdap_samd21_dump.zip](https://github.com/pyocd/pyOCD/files/7041733/pyocd_cmsisdap_samd21_dump.zip)

flit, 2021-08-24:
Wow, thanks for all the detail! It'll take a little while to digest.

Would you mind attaching the binary or ELF for the SAMD21 test app so I don't have to set stuff up to build it?

Fyi, there's also #1143 that is discussing an issue with the firmware version reported by the debugger from some of the EDBG firmwares (and others).

mws-rmain, 2021-08-24:
I'll send the file first thing tomorrow morning.

I did see #1143. Correct me if I'm wrong, but I think the firmware check would occur later in the initialization process than the exception that is tripping me up...(?)
I'll add the patch tomorrow & check it anyway.

I should have also mentioned - I'm running this on Ubuntu 20.04 (on windows in a VM, not that that should matter). I installed the pyocd from the Ubuntu package (not compiled from github master). Maybe I should try that too (though I didn't find any mention of closed issues that might reflect my problem, so that's a bit of a stretch).

mws-rmain, 2021-08-25:
I've attached a zip of the elf. The second zip contains the three board files (described earlier) required to rebuild.
[cdc_msc_cnano.zip](https://github.com/pyocd/pyOCD/files/7046690/cdc_msc_cnano.zip)
[curiosity_nano.zip](https://github.com/pyocd/pyOCD/files/7046691/curiosity_nano.zip)

mws-rmain, 2021-08-25:
I get the same exception after I've applied a variant of #1143 (PR #1201). 

The variation required: 
MICROCHIP_NEDBG = (0x03eb, 0x2175)

For the purposes of this test, I ignored the firmware version returned - I just filtered on the VID & PID.

mws-rmain, 2021-08-25:
I can get it to program if I force the DAPAccessIntf.PORT to SWD (instead of DEFAULT); this sets the 'mode' of the connection request to be '1' (instead of 0). See cmsis_dap_core.py:205.
Is it possible to encode this in the board.mk so make will feed it to pyocd properly?

mws-rmain, 2021-08-25:
My apologies. The board related files I've been posting in this issue are from a different project (tinyusb) that uses pyocd, so not directly related.

I finally got this to program with "dap_protocol=swd" on the pyocd command line:
`pyocd flash -t atsamd21g17a -O dap_protocol=swd cdc_msc.hex
`

For some reason, the nEDBG did not respond properly with the protocol set to 'DEFAULT'

Also, I removed the #1143 (PR #1201 ) patch, and was still able to program.
Thanks for your time!
BR

mws-rmain, 2021-08-25:
OK, apparently it goes through the motions, but actually programs nothing.
```
pyocd flash -t atsamd21g17a -O dap_protocol=swd _build/curiosity_nano/cdc_msc.hex
connect cmd: 2 1  resp: 2 1
[====================] 100%
0004049:INFO:loader:Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 24576 bytes (24 pages) at 10.11 kB/s
```

Any idea what I'm missing now?

mws-rmain, 2021-08-25:
I suspect it is MCU. I've been specifying samd21g17A (correct package & memory, but incorrect variant 'A'), since this is the closest device supported by the 'SAMD21_DFP' pack. However, the DM320119 Curiosity Nano board has a samd21g17D MCU (the 'D' variant). I expect this identity difference will be reported in the SAMD Device ID register of the 'Device Service Unit'.

The SAMD21_DFP is produced by Keil. Is an updated version available?
How can I verify this is the issue?

flit, 2021-08-25:
If it says it skipped a number of bytes, that's almost certainly because the same data was already present in those flash sectors. So it's not actually a problem. You can force it to always erase and program by setting `-Osmart_flash=0`.

Really weird that it doesn't handle the default protocol mode. At first glance that's a bug in Atmel's CMSIS-DAP implementation, though I'll read through the spec again and see if there's a way to declare default is not supported.

One option would be to have pyocd handle 'default' protocol internally and always send SWD to the probe.

flit, 2021-08-25:
Another method to force programming is to separately erase the chip with `pyocd erase --chip`, then program.

The chip variant probably isn't causing an issue. That's just a guess based on experience. But the flash controller is highly likely to be unchanged between variants. The variant could be a silicon rev with errata fixes, or different temp profile. (The datasheet or RM should say something about it, often in the ordering info section.)

flit, 2021-08-25:
Here's the variant info from page 16 of SAM-D21DA1-Family-Data-Sheet-DS40001882G.pdf.

- A = Default Variant
- B = Added RWWEE support for 32 KB and 64 KB memory options
- C = Silicon revision F for WLCSP45 package option
- L = Pinout optimized for Analog and PWM
- D = Silicon Revision G with RWWEE Support in 128KB memory options

mws-rmain, 2021-08-25:
Thanks for the tip re:smart flash.
The results LOOK better now:
```
pyocd flash -t atsamd21g17a -O smart_flash=0 -O dap_protocol=swd _build/curiosity_nano/cdc_msc.hex
[====================] 100%
0006151:INFO:loader:Erased 24576 bytes (3 sectors), programmed 24576 bytes (24 pages), skipped 0 bytes (0 pages) at 5.35 kB/s
```
I also did an erase (which I know worked, since the onboard LED stopped blinking), followed by a flash, and the LED started blinking again. This may be working now - I'll close this loop when I've added some additional code to properly verify.

I've been through the variant info in the data sheet. The most significant difference seems to be the A variant differs from BCD variants in that it lacks the RWWEE support. My concern was that the debugger might check the 'target' specified matched the actual MCU being programmed & skip actual programming if there was a mis-match. I believe the MCU _is_ actually being programmed now (my earlier tests did not erase it first).

mws-rmain, 2021-08-25:
Anyway, could you please add these lines to the end of file '50-cmsis-dap.rules':

```
# 03eb:2175 Microchip (/Atmel) nEDBG CMSIS-DAP
SUBSYTEM="usb", Attr{idVendor}=="03eb", ATTR{idProduct}=="2175", MODE:="666"
```

I'll report the other changes to the tinyusb project.
BR.

flit, 2021-09-23:
Hi @mws-rmain, can you confirm that pyocd 0.32.0 fixes this issue?

flit, 2022-01-15:
Closing this issue, problem seems to be fixed.

closed 2022-01-15 by flit