pyocd/pyOCD #1206: Fail to program LPC55S28 using pyocd

lucviala, 2021-09-01:
Hello,

Since 2 weeks, I tried to program my `lpcxpresso55s28` board using the zephyr command `west flash` which uses `pyocd` as runner, but I get the following error even if I use the `pyocd` command by myself:

`pyocd flash -e sector -a 0x10000000 -t lpc55s28 /home/luc/zephyr/build/zephyr/zephyr.hex -v`

> 0000726:INFO:board:Target type is lpc55s28
0000891:CRITICAL:__main__:No ACK received
Traceback (most recent call last):
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 312, in read_dp
    result = self._link.read_reg(reg_id, now=now)
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 896, in read_reg
    return read_reg_cb()
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 890, in read_reg_cb
    res = transfer.get_result()
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 144, in get_result
    self.daplink.flush()
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 710, in flush
    self._read_packet()
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 975, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 450, in decode_data
    data = self._decode_transfer_block_data(data)
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 419, in _decode_transfer_block_data
    self._check_response(data[3])
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 342, in _check_response
    raise DAPAccessIntf.TransferError("No ACK received")
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError: No ACK received \
\
The above exception was the direct cause of the following exception: \
\
Traceback (most recent call last):
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/__main__.py", line 150, in run
    status = cmd.invoke()
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/subcommands/flash_cmd.py", line 86, in invoke
    with session:
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/core/session.py", line 342, in __enter__
    self.open()
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/core/session.py", line 460, in open
    self._board.init()
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/board/board.py", line 86, in init
    self.target.init()
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/core/soc_target.py", line 117, in init
    seq.invoke()
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 376, in _connect
    connector.connect(self._protocol)
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 166, in connect
    self._connect_dp(protocol)
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 206, in _connect_dp
    self._idr = self.read_idr()
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 233, in read_idr
    dpidr = self._probe.read_dp(DP_IDR, now=True)
  File "/home/luc/.local/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 315, in read_dp
    raise self._convert_exception(error) from error
pyocd.core.exceptions.TransferError: No ACK received

It looks like the probe won't respond to pyocd.

Someone had already faced this issue ?

flit, 2021-09-07:
Hi there. What version of pyocd are you using?

When a "No ACK received" error is seen, it usually means the device is in a low power sleep mode. (Or the SWD pins have been muxed to another function.) Thus, you somehow need to wake the device and gain control over it. It's actually been a while since I last looked at low power recovery on the LPC55xx, so I don't remember the current status.

On most devices, this is done by setting the connect mode to `under-reset` (e.g., `--connect=under-reset`). However, I believe that this doesn't work on the LPC55xx due to how the boot ROM works.

I'm afraid I'll need to refresh my memory on the LPC55xx status and process! I'll try to get back to you quickly.

Are you using one of the zephyr samples? I'd like to be able to reproduce it.

lucviala, 2021-09-13:
Hi @flit ,

I tried with both versions `0.31.0` & `0.30.2`.

Ok, I will check if the board is in sleep mode or SWD pins are muxed elsewhere.

If you wan't to test on the main branch of `zephyr`, may the `samples/subsys/console/echo` will be enough.

Let me know if you get the same error.

lucviala, 2021-09-17:
Hi @flit,

You've right, I've found the problem with my setup, it is because one of the SWD pin have been muxed to another function as you said. I removed the configuration for this specific pin and now it works well.

Thank you for your help.

flit, 2021-09-21:
That's a relief! Glad it's working for you.

closed 2021-09-21 by flit