pyocd/pyOCD #1222: Target Detection Error

Ashleyvk, 2021-09-26:
[image]
[image]

Auto detection of the target does not take place. STM32WB55RG board is added to the target list. 

When trying to erase the target board, error comes up (image attached). Target connected to board via JTAG and has DAPLink firmware within the board.

jeromecoutant, 2021-09-27:
Maybe you should read #811

flit, 2021-09-27:
Regarding target auto detection: What build of DAPLink are you using? The last 8 digits of a DAPLink-based probe's unique ID should indicate the debug unit (aka HIC), but I don't recognise 0x46333733 == 'F373'. Also, the device name "Arm CMSIS-DAP v1" is suspicious—that text has never been used with DAPLink (but it can be changed if you modified it).

Anyway, for board auto detection with DAPLink you need to use a build of DAPLink that knows what target it expects to be used with. The builds with only the HIC name in the output files (like `lpc55s69_if`) are standalone builds with no built-in target, and thus report an all-zero board ID.

See the [Board IDs wiki page](https://github.com/pyocd/pyOCD/wiki/Board-IDs) for some more relevant info.

Ashleyvk, 2021-09-29:
I've added a custom HIC based on STM32WB55RG. How should I provide the unique ID? The device name is taken from #define USBD_HID_STRDESC.
While trying to erase, the following error is obtained:
0001109:DEBUG:dap:DP IDCODE read failed; resending SWJ sequence (use dormant=False)
0001109:DEBUG:swj:Sending deprecated SWJ sequence to select SWD

mathias-arm, 2021-11-08:
Try with another probe, check your connections, check that the target is powered on. "No ACK Received" and "DP IDCODE read failed" are errors I typically see when the probe is not able to talk properly to the target.

flit, 2021-11-08:
Those errors are most often associated with the target being in low power mode. You need to use `--connect-mode=under-reset` to forcibly wake up the device.

BibyThm, 2021-11-14:
[image]
We have checked with the above command and still obtaining the same error (refer figure). Is there anything else to be done?
@mathias-arm @flit @symeonmattes

BibyThm, 2022-01-26:
[image]

**pyocd: error: unrecognized arguments: --connect-mode=under-reset**
We got this error when we tried the above command. Any suggestions?
@mathias-arm @flit

flit, 2022-01-29:
@BibyThm The argument is `--connect=under-reset`, without "mode". Sorry for my mistake! You can always check the available command line options with `--help`.