pyocd/pyOCD #1223: `pack update` fails due to invalid certificates and 404 errors

SIGSTACKFAULT, 2021-09-28:
when I run `pyocd pack update`, this happens:

```
tether@Nervous-Carrier ~> pyocd pack clean
0000244:INFO:pack_cmd:Removing all pack data...

tether@Nervous-Carrier ~> pyocd pack update
0000255:INFO:pack_cmd:Updating pack index...
22:05:04 [WARN] Sending fatal alert BadCertificate
22:05:04 [ERROR] download of "http://www.redpinesignals.com/downloads/keil_packs/RS13100_DFP/Redpine.RS13100_DFP.pdsc" failed: https://www.redpinesignals.com/downloads/keil_packs/RS13100_DFP/Redpine.RS13100_DFP.pdsc: error trying to connect: invalid certificate: CertExpired
22:05:06 [WARN] Sending fatal alert BadCertificate
22:05:06 [ERROR] download of "http://www.redpinesignals.com/downloads/keil_packs/RS14100_DFP/Redpine.RS14100_DFP.pdsc" failed: https://www.redpinesignals.com/downloads/keil_packs/RS14100_DFP/Redpine.RS14100_DFP.pdsc: error trying to connect: invalid certificate: CertExpired
22:05:09 [WARN] Sending fatal alert BadCertificate
22:05:09 [ERROR] download of "https://www.nuvoton.com/enu/Documents/KEILSoftwarePack/Nuvoton.NuMicro_M051_BSP.pdsc" failed: https://www.nuvoton.com/enu/Documents/KEILSoftwarePack/Nuvoton.NuMicro_M051_BSP.pdsc: error trying to connect: invalid certificate: UnknownIssuer
22:05:09 [WARN] Sending fatal alert BadCertificate
22:05:09 [ERROR] download of "https://www.nuvoton.com/enu/Documents/KEILSoftwarePack/Nuvoton.NuMicro_M2351_BSP.pdsc" failed: https://www.nuvoton.com/enu/Documents/KEILSoftwarePack/Nuvoton.NuMicro_M2351_BSP.pdsc: error trying to connect: invalid certificate: UnknownIssuer
22:05:10 [WARN] Sending fatal alert BadCertificate
22:05:10 [ERROR] download of "https://www.nuvoton.com/enu/Documents/KEILSoftwarePack/Nuvoton.NuMicro_DFP.pdsc" failed: https://www.nuvoton.com/enu/Documents/KEILSoftwarePack/Nuvoton.NuMicro_DFP.pdsc: error trying to connect: invalid certificate: UnknownIssuer
22:05:10 [WARN] Sending fatal alert BadCertificate
22:05:10 [ERROR] download of "https://www.nuvoton.com/enu/Documents/KEILSoftwarePack/Nuvoton.NuVoice_ISD_DFP.pdsc" failed: https://www.nuvoton.com/enu/Documents/KEILSoftwarePack/Nuvoton.NuVoice_ISD_DFP.pdsc: error trying to connect: invalid certificate: UnknownIssuer
22:05:11 [ERROR] download of "https://s3.eu-west-2.amazonaws.com/lpccs-docs.dialog-semiconductor.com/CMSIS/DialogSemiconductor.DA1453x_DFP.pdsc" failed: https://s3.eu-west-2.amazonaws.com/lpccs-docs.dialog-semiconductor.com/CMSIS/DialogSemiconductor.DA1453x_DFP.pdsc: connection closed before message completed
22:05:11 [ERROR] download of "https://s3.eu-west-2.amazonaws.com/lpccs-docs.dialog-semiconductor.com/CMSIS/DialogSemiconductor.DA1469x_DFP.pdsc" failed: https://s3.eu-west-2.amazonaws.com/lpccs-docs.dialog-semiconductor.com/CMSIS/DialogSemiconductor.DA1469x_DFP.pdsc: connection closed before message completed
22:05:11 [WARN] Sending fatal alert BadCertificate
22:05:11 [ERROR] download of "https://www.nuvoton.com/enu/Documents/KEILSoftwarePack/Nuvoton.NuMicroM7_DFP.pdsc" failed: https://www.nuvoton.com/enu/Documents/KEILSoftwarePack/Nuvoton.NuMicroM7_DFP.pdsc: error trying to connect: invalid certificate: UnknownIssuer
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] invalid digit found in string
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] No Core found!
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Expecting </link> found </head>
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Expecting </link> found </head>
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Expecting </link> found </head>
22:05:15 [WARN] XML error: Expecting </link> found </head>
22:05:15 [WARN] XML error: Expecting </link> found </head>
22:05:15 [WARN] XML error: Expecting </link> found </head>
22:05:15 [WARN] XML error: Expecting </link> found </head>
22:05:15 [WARN] XML error: Expecting </link> found </head>
22:05:15 [WARN] XML error: Expecting </link> found </head>
22:05:15 [WARN] XML error: Expecting </link> found </head>
22:05:15 [WARN] XML error: Expecting </link> found </head>
22:05:15 [WARN] XML error: Expecting </link> found </head>
22:05:15 [WARN] XML error: Expecting </link> found </head>
22:05:15 [WARN] XML error: Expecting </link> found </head>
22:05:15 [WARN] XML error: Expecting </link> found </head>
22:05:15 [WARN] No Core found!
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] XML error: Only Comment, CDATA and DOCTYPE nodes can start with a '!'
22:05:15 [WARN] No Core found!
22:05:15 [WARN] XML error: Expecting </link> found </head>
22:05:15 [WARN] XML error: Expecting </link> found </head>
[... 138 lines cut ...]
22:05:15 [WARN] Device found without a processor ATSAME70J19
22:05:15 [WARN] Device found without a processor ATSAMS70J19
22:05:15 [WARN] Device found without a processor ATSAMV71J19
22:05:15 [WARN] Device found without a processor ATSAMV70J19
22:05:15 [WARN] cannot parse integer from empty string
22:05:15 [WARN] cannot parse integer from empty string
22:05:15 [WARN] cannot parse integer from empty string
22:05:15 [WARN] cannot parse integer from empty string
22:05:15 [WARN] cannot parse integer from empty string
22:05:15 [WARN] cannot parse integer from empty string
22:05:15 [WARN] cannot parse integer from empty string
22:05:15 [WARN] cannot parse integer from empty string
22:05:15 [WARN] cannot parse integer from empty string
22:05:15 [WARN] cannot parse integer from empty string
22:05:15 [WARN] cannot parse integer from empty string
22:05:15 [WARN] cannot parse integer from empty string
22:05:15 [WARN] cannot parse integer from empty string
22:05:15 [WARN] cannot parse integer from empty string

tether@Nervous-Carrier ~> 
```

not only that, if I open https://www.silabs.com/support/resourceskeil_packs/RS13100_DFP/Redpine.RS13100_DFP.pdsc not only is the certificate expired, it's a 404.

Please advise.

SIGSTACKFAULT, 2021-09-28:
```
tether@Nervous-Carrier ~> pyocd --version
0.32.0
```

SIGSTACKFAULT, 2021-09-28:
just tried again after `pip install -U pyocd`; no change.

flit, 2021-09-29:
Sorry about that! The vendors need to fix their presumably expired TLS certificates. We can't do that for them.

flit, 2021-09-29:
[TLS verification report for www.redpinesignals.com](https://www.ssllabs.com/ssltest/analyze.html?d=www.redpinesignals.com&latest) shows the certificate is expired. This has to be fixed by them, so please contact them if you need access to the Red Pine packs.

[Report for www.nuvoton.com.](https://www.ssllabs.com/ssltest/analyze.html?d=www.nuvoton.com) This is different, the certificate is valid. I'm not 100% certain what's causing the failure, but my guess would be the incomplete certificate chain. That may not be something we can fix in cmsis-pack-manager, certainly not easily. You may be able to manually download Nuvoton packs from the [CMSIS-Pack list](https://www.keil.com/dd2/pack/) and use them [as documented](https://github.com/pyocd/pyOCD/blob/main/docs/target_support.md) with the `--pack` argument. (Browsers tend to have far more complex TLS stacks than other tools, and are much more forgiving by default.)

flit, 2021-09-29:
(Meant to add…) About the 404 response in the files, that's a known issue in cmsis-pack-manager for which we're working on a fix.