pyocd/pyOCD #1247: error starting pyocd

ccattuto, 2021-11-25:
OSX Big Sur, Python 3.8.2 from Anaconda, freshly checked-out main branch of pyOCD. `python setup.py install` builds and installs correctly. On launching `pyocd` I get `ModuleNotFoundError: No module named 'pyocd._version`.

flit, 2021-11-28:
Thanks for reporting. This might be related to the move to PEP 517 packaging. The `pyocd._version` module is (supposed to be!) auto-generated by the `setuptools_scm` package when pyocd is built.

First, please try installing `setuptools_scm[toml]` with pip. It may be silently skipping that step if `setuptools_scm` isn't available. 

If that doesn't work, could you please try installing with `pip install .`?

(Running `setup.py` directly is supposed to be deprecated now. In fact, many projects don't even need a `setup.py` anymore; pyocd still needs it to build the SVD zip from individual SVD files checked into the repo. When running `setup.py`, the build dependencies like `setuptools_scm` don't get installed automatically.)

I will also try to reproduce.

flit, 2021-11-28:
I'm updating the developer's guide docs that still mentioned using `setup.py` directly.

ccattuto, 2021-11-29:
Thanks for the feedback. In fact, `setuptools_scm` was not installed. On manually installing it with pip, the `_version.py` file is indeed automatically generated during the build. `pip install .` also works correctly.

flit, 2021-12-01:
Glad to hear it worked. I've updated all the docs to say to use `pip install .`.

closed 2021-12-01 by flit