pyocd/pyOCD #1264: Option `enable_semihosting` is not working

caspermeijn, 2021-12-11:
Thank you for this great project. I have been using it as part of the [Zephyr project](https://www.zephyrproject.org/) and it works great.

I am using ARM semihosting for my `printf` statements and for each debug session I type `monitor arm semihosting enable`. After getting sick of doing that I wanted to enable semihosting by default for my board. When looking at the [pyOCD options documentation](https://pyocd.io/docs/options.html), I got the feeling that I needed to create the following config file to enable arm semihosting.
```
# Set to True to handle semihosting requests. Also see the semihost_console_type option. Default is False.
enable_semihosting: True
```

However with this config file the semihosting breakpoints were still not parsed. So I tried `pyocd gdbserver -O enable_semihosting=true`, which also didn't work. After trying lots of variations I finally found the following working command: `pyocd gdbserver --semihosting`.

My conclusion was that the options parsing doesn't work correctly for `enable_semihosting`.

flit, 2021-12-11:
Hi @caspermeijn, thanks for the report. I'll try to reproduce and fix.

In fact, I've been writing some detailed documentation for semihosting, so it would be nice if what's documented actually works!

flit, 2021-12-11:
Turns out the problem is that the `-S` / `--semihosting` command line argument, whether set or not, always overrides the `enable_semihosting` option's initial value from `-O` or config file. Only if set later via `arm semihosting enable` or `set option enable_semihosting=1` will it take effect. (This bug affects several other options, too.)

I won't be able to get to it right away, but it's on the list now. Thanks!