pyocd/pyOCD #1276: -f option doesn't work while using STLinkV3

llinjupt, 2021-12-23:
I have assigned the frequency with -f 24M, which is the highest avaliable freq for STLinkV3, but it seems it doesn't take any effect.

>pyocd flash -t STM32H743VITX STM32H743.bin -f 24M
also
>pyocd flash -t STM32H743VITX STM32H743.bin -f 24m 

I measured the SWCLK with an oscilloscope to check if it worked, but the result is only about 4.8MHz. THX.

flit, 2021-12-26:
Thanks for the report! I'll look into it and try to reproduce and fix.

flit, 2021-12-26:
Stupid copy paste bug… 🥲 Thanks again for reporting it.

llinjupt, 2021-12-30:
> Stupid copy paste bug… 🥲 Thanks again for reporting it.

Thanks for your patch first, but after trying it, the freq is always about 11.1MHz, no matter what the parameter is.
such as -f 1m, -f 10m or -f 24m.

I am not sure why  it happened like that. Have you tried to check the  SWCLK with an oscilloscope? BTW, I am sure it works while using keil and setting 24MHz with high performance option as well.

flit, 2021-12-31:
I wasn't checking with a scope, but I was timing reading 512 kB of SRAM. The times clearly showed the expected improvement as the frequency was increased.

I'll retest and make sure I didn't break something on the way to merging it. Thanks for following up and trying out the patch!

flit, 2022-01-03:
Would you mind running the attached script? It will iterate over all possible STLinkV3 SWD frequencies except 50 kHz and below, transfer 512 kB, and print a report with resulting transfer times. It tests all frequencies obtainable by the three different STLinkV3 HCLK prescaler values (1, 2, 4), settable by the (new) `stlink.v3_prescaler` option.

These are the results on my system (MacBook Pro x86_64 running macOS 11.6) (sorted by hand):

```
Results:
1.206 s: 24.0 MHz
2.073 s: 12.0 MHz
1.493 s: 8.0 MHz
3.696 s: 6.0 MHz
2.563 s: 4.0 MHz
2.354 s: 3.3 MHz
4.944 s: 2.0 MHz
4.242 s: 1.6 MHz
4.501 s: 1.0 MHz
5.627 s: 1.0 MHz
17.055 s: 0.2 MHz
19.465 s: 0.2 MHz
8.524 s: 0.85 MHz
11.648 s: 0.52 MHz
22.742 s: 0.2 MHz
```

There are some unexpected values, like 8 MHz being less than half 6 MHz. Some of that might be explained by the "higher performance" that prescaler=1 gives. And of course expected run variations, and interference of host processes.

The main point is that there is a clear link between SWD frequency and transfer time.

I'm particularly interested, if you _don't_ see the expected speed differences, in what the debug log messages with the actual speed reported by the STLink (logging is configured so only this debug log message is output).

Sorry I haven't tested with my scope; at the moment it's packed up for a house move in a couple weeks.

The STLinkV3 on my Nucleo-H743 is running firmware V3J8M3.

[stlinkv3_rx_times_by_swd_freq.zip](https://github.com/pyocd/pyOCD/files/7800021/stlinkv3_rx_times_by_swd_freq.zip)

llinjupt, 2022-01-03:
> Would you mind running the attached script? It will iterate over all possible STLinkV3 SWD frequencies except 50 kHz and below, transfer 512 kB, and print a report with resulting transfer times. It [...]

$ ./stlinkv3_rx_times_by_swd_freq.py
Traceback (most recent call last):
  File "F:\stlinkv3_rx_times_by_swd_freq\stlinkv3_rx_times_by_swd_freq.py", line 11, in <module>
    from pyocd.utility.color_log import ColorFormatter
ModuleNotFoundError: No module named 'pyocd.utility.color_log'

I encountered such an error message and after searching the whole source of pyOCD, I got nothing about color_log.  BTW, I tested with STLinkV3MINI running firmware V3J9M3.

flit, 2022-01-03:
`pyocd.utility.color_log` is from the `develop` branch, which is the branch containing the STLinkV3 frequency fix. I'm terribly sorry I didn't mention earlier that the `develop` branch  is required.

You can install the `develop` branch thusly:

```
pip install -U git+https://github.com/pyocd/pyOCD.git@develop
```

(Or clone the repo and install from that.)

llinjupt, 2022-01-04:
> `pyocd.utility.color_log` is from the `develop` branch, which is the branch containing the STLinkV3 frequency fix. I'm terribly sorry I didn't mention earlier that the `develop` branch is required. [...]

```
$ ./stlinkv3_rx_times_by_swd_freq.py
usage: stlinkv3_rx_times_by_swd_freq.py [-h] uid
stlinkv3_rx_times_by_swd_freq.py: error: the following arguments are required: uid
```

Badly I got another error message, but it doesn't matter and I tested using the old way, and sure it worked.

```
pyocd flash -t STM32H743VITX STM32H743.bin -f 24m
```

The stange thing is that '-f 24m' worked with real 27MHz by checking the scope. Then I did a validation on Keil MDK, and got the same result.
Another abnormal phenomenon is that if assign the -f pareameter between 10MHz and 23MHz, the real freq is pinned to 11.4MHz but below 8MHz, the real freq is more accurate close to the assigned value both on Pyocd on Keil MDK.

Such abnormal cases should be related to the clock-algorithm of STLinkV3 and don't matter to PyOCD. 
Anyway your patch worked to me.

flit, 2022-01-04:
The error message just meant that you needed to pass the probe's unique ID, eg `stlinkv3_rx_times_by_swd_freq.py 02100` for my H743.

Actually, I'm surprised that the actual frequencies are as accurate as they are!

It's interesting the (entirely expected) effect that changing the STLink's HCLK prescaler has on SWD frequencies, as reported by the STLink and visible in the `stlinkv3_rx_times_by_swd_freq.py` script output. When running pyocd directly, you can change the prescaler with `-Ostlink.v3_prescaler=x` where _x_ is 1, 2, or 4.

Because of the limited set of frequencies available at each prescaler, it would be nice at some point to implement automatic switching of the prescaler based on requested frequency.

flit, 2022-01-04:
Fyi, here's an updated version of the script. You still need to pass the probe UID, as the usage describes.

[stlinkv3_rx_times_by_swd_freq.zip](https://github.com/pyocd/pyOCD/files/7810222/stlinkv3_rx_times_by_swd_freq.zip)

llinjupt, 2022-01-05:
Thanks for your hint and it worked. FYI:

```
Setting prescaler to 1
  current=1000
  supported=[24000, 8000, 3300, 1000, 200, 50, 5]
Setting SWD clock to 24.0 MHz
0001554 D actual SWD frequency is 24000 kHz [stlink]
Read 524288 bytes
1.203 s: 24.0 MHz (prescaler 1)
Setting SWD clock to 8.0 MHz
0002766 D actual SWD frequency is 8000 kHz [stlink]
Read 524288 bytes
1.547 s: 8.0 MHz (prescaler 1)
Setting SWD clock to 3.3 MHz
0004316 D actual SWD frequency is 3300 kHz [stlink]
Read 524288 bytes
2.390 s: 3.3 MHz (prescaler 1)
Setting SWD clock to 1.0 MHz
0006705 D actual SWD frequency is 1000 kHz [stlink]
Read 524288 bytes
4.703 s: 1.0 MHz (prescaler 1)
Setting SWD clock to 0.2 MHz
0011403 D actual SWD frequency is 200 kHz [stlink]
Read 524288 bytes
17.110 s: 0.2 MHz (prescaler 1)
Setting prescaler to 2
  current=200
  supported=[12000, 4000, 1600, 1000, 200, 50, 5]
Setting SWD clock to 12.0 MHz
0028521 D actual SWD frequency is 12000 kHz [stlink]
Read 524288 bytes
2.094 s: 12.0 MHz (prescaler 2)
Setting SWD clock to 4.0 MHz
0030610 D actual SWD frequency is 4000 kHz [stlink]
Read 524288 bytes
2.687 s: 4.0 MHz (prescaler 2)
Setting SWD clock to 1.6 MHz
0033289 D actual SWD frequency is 1600 kHz [stlink]
Read 524288 bytes
4.234 s: 1.6 MHz (prescaler 2)
Setting SWD clock to 1.0 MHz
0037537 D actual SWD frequency is 1000 kHz [stlink]
Read 524288 bytes
5.625 s: 1.0 MHz (prescaler 2)
Setting SWD clock to 0.2 MHz
0043158 D actual SWD frequency is 200 kHz [stlink]
Read 524288 bytes
19.500 s: 0.2 MHz (prescaler 2)
Setting prescaler to 4
  current=200
  supported=[6000, 2000, 850, 520, 200, 50, 5]
Setting SWD clock to 6.0 MHz
0062652 D actual SWD frequency is 6000 kHz [stlink]
Read 524288 bytes
3.860 s: 6.0 MHz (prescaler 4)
Setting SWD clock to 2.0 MHz
0066519 D actual SWD frequency is 2000 kHz [stlink]
Read 524288 bytes
5.109 s: 2.0 MHz (prescaler 4)
Setting SWD clock to 0.85 MHz
0071626 D actual SWD frequency is 850 kHz [stlink]
Read 524288 bytes
8.406 s: 0.85 MHz (prescaler 4)
Setting SWD clock to 0.52 MHz
0080034 D actual SWD frequency is 520 kHz [stlink]
Read 524288 bytes
11.422 s: 0.52 MHz (prescaler 4)
Setting SWD clock to 0.2 MHz
0091449 D actual SWD frequency is 200 kHz [stlink]
Read 524288 bytes
22.500 s: 0.2 MHz (prescaler 4)
Results:
1.203 s: 24.0 MHz (prescaler 1)
1.547 s: 8.0 MHz (prescaler 1)
2.390 s: 3.3 MHz (prescaler 1)
4.703 s: 1.0 MHz (prescaler 1)
17.110 s: 0.2 MHz (prescaler 1)
2.094 s: 12.0 MHz (prescaler 2)
2.687 s: 4.0 MHz (prescaler 2)
4.234 s: 1.6 MHz (prescaler 2)
5.625 s: 1.0 MHz (prescaler 2)
19.500 s: 0.2 MHz (prescaler 2)
3.860 s: 6.0 MHz (prescaler 4)
5.109 s: 2.0 MHz (prescaler 4)
8.406 s: 0.85 MHz (prescaler 4)
11.422 s: 0.52 MHz (prescaler 4)
22.500 s: 0.2 MHz (prescaler 4)
```

flit, 2022-01-05:
Nice. 😄 Those results match mine.

I'm closing this issue now that it's fixed on `develop`.

PR #1283 (2021-12-26): STLinkV3: fix setting SWD frequency and improve max frequency.
The main point of this PR is to fix a bug setting the SWD frequency for V3 where it was always setting JTAG frequency instead. (STLink treats the two as separate frequencies.)

In addition, the STLinkV3 HCLK prescaler is set on open from a `stlink.v3_prescaler` session option. By default it's set to 1 for high performance. This increases the maximum frequency from 21.333 MHz to the maximum rating of 24 MHz. You can see the actual frequency reported by the STLink in a debug log from the `pyocd.probe.stlink.stlink` module.

Fixes #12 [... cut]
closed 2022-01-05 by flit