pyocd/pyOCD #1303: LPC5526: erase/flash failure on factory new chip

LeBlue, 2022-01-11:
pyocd version: 0.32.3

Eraseing or flashing of (factory) new chips is failing with

```
0004324:DEBUG:cortex_m:halting core 0
0004326:DEBUG:sequencer:Running task post_connect
0004327:DEBUG:sequencer:Running task post_connect_hook
0004327:DEBUG:sequencer:Running task create_flash
0004327:DEBUG:sequencer:Running task notify
0004327:INFO:load_cmd:Loading /Users/matthiaswauer/Projects/zephyr/zephyr.hex
0004522:DEBUG:builder:Analyze time: 0.000152
0004522:DEBUG:builder:Using double buffer sector erase program
[                                                  ]   0%0004522:DEBUG:cortex_m:halting core 0
0004547:DEBUG:cortex_m:reset, core 0, type=SW_SYSRESETREQ
0008570:WARNING:cortex_m:Core #0 is not accessible after reset
0008574:DEBUG:session:uninit session <pyocd.core.session.Session object at 0x11092f6d0>
0008574:DEBUG:board:uninit board <pyocd.board.board.Board object at 0x11092f8e0>
0008577:ERROR:board:link exception during target disconnect: Memory transfer fault @ 0xe000edf0-0xe000edf3
```
 
This does not happen, if the chip was already sucessfully flashed/erased before.

I have investigated the issue. When the the chip is new, the debug access is unlocked when attaching

```
0003696:DEBUG:sequencer:Running task check_locked_state
0003708:INFO:target_lpc5500:attempting unlock procedure     <----- runs only on new chips, no idea how to lock it again
0003864:INFO:target_lpc5500:re-running discovery
0003869:DEBUG:sequencer:Running task set_max_invalid_aps
0003870:DEBUG:sequencer:Running task find_aps
0003947:DEBUG:sequencer:Running task create_aps
0003947:DEBUG:sequencer:Running task create_ap.0
```

But before the flash/erase operation, the target is reset which locks the debug access again, resulting in the error above, because the debug access is not unlocked again after the reset.

A quick workaournd is using the following user script (```target``` is the ```LPC5000``` family) 
```
def did_reset():
    target.unlock(target.aps[2])
    target.halt()
```

I can send a PR but would like to know what approch would be better (both tested and fix the issue).

1. define a custom ```mass_erase``` that skips the reset for the LPC5500 family or the LPC5526 target.
2. unconditionally add the unlock from the user script into the LPC5500 family custom ```reset_and_halt``` implementation

LeBlue, 2022-01-12:
The same memory transfer fault can be reproduced after a mass/chip erase when doing a software reset

```
pyocd erase -v -v --target lpc5526 --chip
pyocd reset -v -v --target lpc5526 -m sw
```

Fixing the unlock after a reset seems to be the correct approach.

This behaviour is also observable on the ```lpc55s28``` target.

flit, 2022-01-12:
Thanks for the investigation and fixes!

LeBlue, 2022-02-18:
Fixed in v0.33.0

closed 2022-02-18 by LeBlue

flit, 2022-02-19:
Thanks for the update!

PR #1306 (2022-01-14): lpc5500 family: re-unlock debug access after reset
After a reset, debug access must be re-unlocked. 

Fixes #1303

I tested this with a LPCxpresso55s28 board (target ```lpc55s28```) and a custom board with ```lpcs5526```.

If needed, I can squash the two commits.