pyocd/pyOCD #1312: STM32H750 DBGMCU is no longer accessible after connecting with PyOCD

j4cbo, 2022-01-18:
Without having connected a debugger, I can read and write registers in the STM32H750's `DBGMCU` peripheral from firmware. After connecting with PyOCD (via an Orbtrace CMSIS-DAP), I no longer can - writes are ignored and reads come back as zero, whether done via firmware or via GDB. This persists across reset until the next power cycle and also affects other peripherals on the APB-D, like the trace funnel, buffer, and TPIU.

I found this as a possible cause: https://xor-gate.org/post/stlink-arm-ap-access/ which indicates that the cause could be a debugger clearing the `CSW.DbgSwEnable` bit in the chip's AP2. That post references the ST-Link but I suspect OpenOCD may do the same thing.

I'm not sure what the right fix is. It looks like pyocd's `AccessPort` avoids altering unintended bits in the AP's CSW. But `DBGSWENABLE` is normally clear while the chip is in reset and set by hardware when the chip _exits_ reset (https://developer.arm.com/documentation/ddi0316/d/functional-description/apb-ap/apb-ap-control-status-word-register--csw--0x00) so perhaps PyOCD is overwriting it with a cached version?

flit, 2022-01-18:
That's really bizarre. Do you know if this happens on the H743 too?

flit, 2022-01-18:
If you care connecting under reset, it could very well be that pyocd is caching the `DBGSWENABLE` bit from the reset state and using it afterwards.

j4cbo, 2022-01-18:
The only H7 that I have is H750, but I'd expect so since they're the same family (H742, 743, 753, and 750 are all covered by the same reference manual).

I'll be able to do some more investigation tonight or tomorrow.

flit, 2022-01-18:
Maybe I'm thinking of the H747 that's different?

Regardless, thanks! I'll check the code myself.

Note that only the APB-AP from CSSoC-400 (a product with a bunch of ADI and CoreSight IP) implements `DBGSWENABLE`. The attached spreadsheet (also in PDF form) is something I put together to document the variations between actual implementations of MEM-AP `CSW` registers.

￼￼
[ADI_MEM-AP_CSW_registers.xlsx](https://github.com/pyocd/pyOCD/files/7892071/ADI_MEM-AP_CSW_registers.xlsx)
[ADI_MEM-AP_CSW_registers.pdf](https://github.com/pyocd/pyOCD/files/7892082/ADI_MEM-AP_CSW_registers.pdf)
￼

j4cbo, 2022-01-18:
That matches, there are a number of references to CoreSight SoC-400 in the H742/743/753/750 reference manual RM0433.

j4cbo, 2022-01-20:
Looks like the issue isn't `_cached_csw`, it's that `AccessPort` starts off with a default value that clears that bit: https://github.com/pyocd/pyOCD/blob/main/pyocd/coresight/ap.py#L484-L485

flit, 2022-01-22:
Would you mind taking #1321 for a spin to make sure it works on your H750? You can install from my branch:

```
pip install git+https://github.com/flit/pyOCD.git@bugfix/apb_ap_dbgswenable
```

Btw, it looks like the reason that hasn't been reported before is that probably most people use an STLink with the H7s. The STLink firmware appears to set `DBGSWEN` by default, and thus masks the issue. (Partly because, currently, the CSW value from the `MEM_AP` class isn't passed to the STLink driver plugin's accelerated memory interface, thus never overriding the default STLink value. This needs to be fixed, since it prevents modifying HPROT transfer settings, but that will require an update to the memory interface API that won't come before v1.0.)

flit, 2022-01-23:
I went ahead and merged #1321 change since it's innocuous enough and testing passed. I'm still curious to hear if it does indeed fix the problem for you.

j4cbo, 2022-01-24:
Yep, that fixes it - thanks! :D

PR #1321 (2022-01-23): coresight: ap: set CSW.DBGSWEN for CSSoC-400 APB-AP.
If this bit in CSW is not set on this particular APB-AP, software running on the device will not be able to access the memory space downstream of the AP.

This is a particular issue for some STM32H7xx devices such as the H743 and H750 that use this APB-AP. However, the problem is only visible when using a debug probe other than an STLink, since the STLink masks the problem by ensuring `DBGSWEN` is set.

Fixes #1312