pyocd/pyOCD #1313: pyocd and picoprobe issues

mrgsts, 2022-01-19:
Hi, I have been trying to get pyocd to work with picoprobe and have had some problems, however it works great with the latest version of openocd. 

any sugestions?

--------------------
~# pyocd -V
0.32.3

~# arm-none-eabi-gdb -v
GNU gdb (GNU Arm Embedded Toolchain 9-2020-q2-update) 8.3.1.20191211-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Please check TERMINAL tab (gdb-server) for output from pyocd
Launching server: "pyocd" "gdbserver" "--port" "50000" "--telnet-port" "50001" "-v" "--target" "ATSAMD21G18A" "-u" "E6612483CB409221"
Launching GDB: "/home/work/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gdb" "-q" "--interpreter=mi2"
 "/home/work/dev/build/zephyr/zephyr.elf"Set "showDevDebugOutput": true in your "launch.json" to see verbose GDB transactions here. Helpful to debug issues or report problems
Reading symbols from /home/work/dev/build/zephyr/zephyr.elf...

...

[2022-01-19T16:10:48.483Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
0000677:DEBUG:session:Project directory: /home/work/dev/build/zephyr/
0000990:INFO:board:Target type is atsamd21g18a
0000992:DEBUG:sequencer:Running task load_svd
0001015:DEBUG:sequencer:Running task pre_connect
0001016:DEBUG:sequencer:Running task dp_init
0001016:DEBUG:sequencer:Running task lock_probe
0001016:DEBUG:sequencer:Running task get_probe_capabilities
0001016:DEBUG:sequencer:Running task connect
0001016:DEBUG:dap:Default wire protocol selected; using SWD
0001022:DEBUG:swj:Sending deprecated SWJ sequence to select SWD
0001065:INFO:dap:DP IDR = 0x0bc11477 (v1 MINDP rev0)
0001067:DEBUG:sequencer:Running task clear_sticky_err
0001068:DEBUG:sequencer:Running task power_up_debug
0001069:DEBUG:sequencer:Running task check_version
0001069:DEBUG:sequencer:Running task unlock_probe
0001069:DEBUG:sequencer:Running task create_discoverer
0001069:DEBUG:sequencer:Running task discovery
0001069:DEBUG:sequencer:Running task find_aps
0001072:DEBUG:sequencer:Running task create_aps
0001072:DEBUG:sequencer:Running task create_ap.0
0001073:INFO:ap:AHB-AP#0 IDR = 0x04770031 (AHB-AP var3 rev0)
0001075:DEBUG:ap:AHB-AP#0 default HPROT=3 HNONSEC=0
0001076:DEBUG:ap:AHB-AP#0 implemented HPROT=3 HNONSEC=0
0001077:DEBUG:sequencer:Running task find_components
0001078:DEBUG:sequencer:Running task init_ap.0
0001083:INFO:rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0x41003000 (designer=01f part=cd0)
0001087:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c0>
0001087:INFO:rom_table:  AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c0)
0001092:INFO:rom_table:  [0]<e000e000:SCS v6-M class=14 designer=43b part=008>
0001096:INFO:rom_table:  [1]<e0001000:DWT v6-M class=14 designer=43b part=00a>
0001099:INFO:rom_table:  [2]<e0002000:BPU v6-M class=14 designer=43b part=00b>
0001102:INFO:rom_table:[1]<41006000:MTB M0+ class=9 designer=43b part=932 devtype=31 archid=0a31 devid=0:0:0>
0001102:DEBUG:sequencer:Running task create_cores
0001102:DEBUG:discovery:Creating SCS component
0001104:INFO:cortex_m:CPU core #0 is Cortex-M0+ r0p1
0001104:DEBUG:sequencer:Running task set_default_reset_type
0001104:DEBUG:sequencer:Running task create_components
0001104:DEBUG:discovery:Creating DWT component
0001105:INFO:dwt:2 hardware watchpoints
0001107:DEBUG:discovery:Creating BPU component
0001108:INFO:fpb:4 hardware breakpoints, 0 literal comparators
0001109:DEBUG:fpb:fpb has been disabled
0001112:DEBUG:sequencer:Running task check_for_cores
0001112:DEBUG:sequencer:Running task halt_on_connect
0001112:DEBUG:cortex_m:halting core 0
0001113:DEBUG:sequencer:Running task post_connect
0001113:DEBUG:sequencer:Running task post_connect_hook
0001113:DEBUG:sequencer:Running task create_flash
0001113:DEBUG:sequencer:Running task notify
0001114:DEBUG:cortex_m:Setting vector catch to 0x00000001
0001118:INFO:server:Semihost server started on port 50001 (core 0)
0001161:INFO:gdbserver:GDB server started on port 50000 (core 0)
0001211:DEBUG:packet_io:Starting GDB server packet I/O thread
0001211:DEBUG:cortex_m:halting core 0
0001212:INFO:gdbserver:Client connected to port 50000!
0001212:DEBUG:gdbserver:GDB received query: [b'Supported', b'multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+#df']
0001213:DEBUG:gdbserver:GDB general set: b'StartNoAckMode'
0001213:DEBUG:gdbserver:GDB received query: [b'Xfer', b'features', b'read', b'target.xml', b'0,7fb#4a']
0001214:DEBUG:gdbserver:GDB query b'read_feature': offset: 0, size: 2043
0001216:DEBUG:gdbserver:GDB general set: b'NonStop:0'
0001217:DEBUG:gdbserver:GDB received query: [b'TStatus#49']
0001217:DEBUG:gdbserver:Current thread 0 is no longer valid, switching context to target
0001220:DEBUG:register:no accesses
0001220:DEBUG:register:out of date run token; invalidating cache
0001222:DEBUG:gdbserver:Tresponse=b'T05thread:1;'
0001222:DEBUG:gdbserver:GDB received query: [b'Xfer', b'threads', b'read', b'', b'0,7fb#d0']
0001222:DEBUG:gdbserver:GDB query b'threads': offset: 0, size: 2043
0001225:DEBUG:gdbserver:GDB received query: [b'Attached#8f']
0001225:DEBUG:gdbserver:GDB received query: [b'Offsets#4b']
0001226:DEBUG:context_facade:GDB getting register context
0001229:DEBUG:context_facade:GDB get_reg_context: r0 = None -> b'xxxxxxxx'
0001229:DEBUG:context_facade:GDB get_reg_context: r1 = None -> b'xxxxxxxx'
0001229:DEBUG:context_facade:GDB get_reg_context: r2 = None -> b'xxxxxxxx'
0001229:DEBUG:context_facade:GDB get_reg_context: r3 = None -> b'xxxxxxxx'
0001229:DEBUG:context_facade:GDB get_reg_context: r4 = None -> b'xxxxxxxx'
0001229:DEBUG:context_facade:GDB get_reg_context: r5 = None -> b'xxxxxxxx'
0001229:DEBUG:context_facade:GDB get_reg_context: r6 = None -> b'xxxxxxxx'
0001230:DEBUG:context_facade:GDB get_reg_context: r7 = None -> b'xxxxxxxx'
0001230:DEBUG:context_facade:GDB get_reg_context: r8 = None -> b'xxxxxxxx'
0001230:DEBUG:context_facade:GDB get_reg_context: r9 = None -> b'xxxxxxxx'
0001230:DEBUG:context_facade:GDB get_reg_context: r10 = None -> b'xxxxxxxx'
0001230:DEBUG:context_facade:GDB get_reg_context: r11 = None -> b'xxxxxxxx'
0001230:DEBUG:context_facade:GDB get_reg_context: r12 = None -> b'xxxxxxxx'
0001230:DEBUG:context_facade:GDB get_reg_context: sp = None -> b'xxxxxxxx'
0001230:DEBUG:context_facade:GDB get_reg_context: lr = None -> b'xxxxxxxx'
0001230:DEBUG:context_facade:GDB get_reg_context: pc = None -> b'xxxxxxxx'
0001230:DEBUG:context_facade:GDB get_reg_context: msp = None -> b'xxxxxxxx'
0001230:DEBUG:context_facade:GDB get_reg_context: psp = None -> b'xxxxxxxx'
0001230:DEBUG:context_facade:GDB get_reg_context: primask = None -> b'xxxxxxxx'
0001230:DEBUG:context_facade:GDB get_reg_context: xpsr = None -> b'xxxxxxxx'
0001230:DEBUG:context_facade:GDB get_reg_context: control = None -> b'xxxxxxxx'
0001231:DEBUG:packet_io:GDB packet thread: other side closed connection
0001231:DEBUG:packet_io:GDB packet thread stopping
[2022-01-19T16:10:49.774Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
---------

newbrain, 2022-01-19:
Hi, I wrote the picoprobe support code, here are some considerations:

At first sight, it would seem that the probe starts correctly - it finds the core and enumerates its AP.
Then something goes wrong, but I'm not able to identify where.

As a test, could you please add `-Opicoprobe.safeswd=True` to pyocd invocation?
This will slow it down _a lot_ but might catch some SWD fault:
picoprobe is a very basic probe, just translating bits to and from SWD line protocol, the option disables the minimal internal queuing of reads and writes.

Also, is the code using zephyr RTOS?

@flit, I might not be able to follow this up that much (away from my picos).

mrgsts, 2022-01-19:
Just discover that with "-f 400000" it seems to work well, also replug the usb seems to help when problems are shown

newbrain, 2022-01-19:
Possible signal integrity problems then.
To speed things up I assumed positive acknowledges from the target in the code, this can be disabled with the option mentioned above.
The picoprobe FW code leaves the pins at the default (fast) slew rate, AFAICS - not that the data sheet (🤮 I find it abysmal) tells us anything about the actual values, but too fast and signal integrity will suffer.

flit, 2022-01-21:
@newbrain No problem, I understand. 😄

flit, 2022-06-08:
Closing issue.

closed 2022-06-08 by flit