pyocd/pyOCD #1314: On windows 10, pyocd eats the first character of my CLI

gerrywangqd, 2022-01-20:
The CLI might "eat" what I typed, especially for the first character I typed,
For example, if I typed the CLI `show cores`, the session will be,

```
pyocd> how cores
Error: unrecognized command 'how'
```
I tried both the old "command.exe" and Powershell.

I don't find this issue in Linux (Ubuntu 20.4.3, Centos 8,4)

flit, 2022-01-21:
Oh, that's really strange! I test on Windows pretty regularly, and I haven't seen this. I'll try again this weekend. Thanks for the bug report.

cederom, 2022-01-21:
Sorry for the offtopic but you inspired me to set `how` alias to `man` :sunglasses:

elfmimi, 2022-01-22:
I know I have experienced this behaviour. But now my installation of pyocd does not show this behaviour. Why is it?
The answer is [here #1129](https://github.com/pyocd/pyOCD/issues/1129#issuecomment-997165403). After applying the treatment I mentioned in that comment, this behaviour has been gone.

To try it with ease, I made a branch.
First do: `pip uninstall pyocd pyocd-pemicro`
Then do: `pip install git+https://github.com/elfmimi/pyOCD.git@remove-pemicro-dependency`

gerrywangqd, 2022-01-24:
Thanks @elfmimi , I tried your fix, it works well on my machine.
@flit

flit, 2022-01-25:
Thanks! Fyi, there's another issue #1319 to make pyocd-pemicro optional for a different reason.