pyocd/pyOCD #1315: "No cores were discovered" with a new cortex-m33 target

TreeNewbeer, 2022-01-20:
I am working on adding a new target which base on cm33 into pyocd, and got an issue about  "No cores were discovered！", the debug log show as:

```
(venv) ➜  pyOCD git:(develop) ✗ pyocd flash test/data/binaries/csk6001.bin -t csk6001 -L 'pyocd.probe.*=debug'                         
0000794 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=046d PID=c53f). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000794 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=5986 PID=212b). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000795 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=046d PID=c53f). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0000796 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=5986 PID=212b). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0000796 D trace: open [cmsis_dap_probe]
0000797 D Detaching Kernel Driver of Interface 3 from USB device (VID=0d28 PID=0204). [pyusb_backend]
0000799 D   USB OUT> (2) 00 fe [pyusb_backend]
0000802 D   USB IN < (64) 00 01 04 [pyusb_backend]
0000802 D   USB RD < (64) 00 01 04 [pyusb_backend]
0000802 D   USB OUT> (2) 00 04 [pyusb_backend]
0000804 D   USB IN < (64) 00 05 31 2e 31 30 [pyusb_backend]
0000804 D   USB RD < (64) 00 05 31 2e 31 30 [pyusb_backend]
0000804 D CMSIS-DAP v1 probe 07000000002600204700000e4e503054a5a5a5a597969908: protocol version 1.1.0 [dap_access_cmsis_dap]
0000804 D   USB OUT> (2) 00 ff [pyusb_backend]
0000806 D   USB IN < (64) 00 02 40 [pyusb_backend]
0000806 D   USB RD < (64) 00 02 40 [pyusb_backend]
0000806 D   USB OUT> (2) 00 f0 [pyusb_backend]
0000808 D   USB IN < (64) 00 01 13 [pyusb_backend]
0000808 D   USB RD < (64) 00 01 13 [pyusb_backend]
0000808 D [cmd:0] New _Command [dap_access_cmsis_dap]
0000808 D flush: sending cmd:0; reading 0 outstanding [dap_access_cmsis_dap]
0000808 D   USB OUT> (2) 00 f0 [pyusb_backend]
0000810 D   USB IN < (64) 00 01 13 [pyusb_backend]
0000810 D   USB RD < (64) 00 01 13 [pyusb_backend]
0000810 D trace: set_clock(freq=1000000) [cmsis_dap_probe]
0000810 D flush: sending cmd:0; reading 0 outstanding [dap_access_cmsis_dap]
0000810 D   USB OUT> (5) 11 40 42 0f 00 [pyusb_backend]
0000812 D   USB IN < (64) 11 00 42 0f [pyusb_backend]
0000812 D   USB RD < (64) 11 00 42 0f [pyusb_backend]
0000812 D trace: connect(DEFAULT) [cmsis_dap_probe]
0000812 D   USB OUT> (2) 02 01 [pyusb_backend]
0000814 D   USB IN < (64) 02 01 [pyusb_backend]
0000814 D   USB RD < (64) 02 01 [pyusb_backend]
0000814 D   USB OUT> (5) 11 40 42 0f 00 [pyusb_backend]
0000816 D   USB IN < (64) 11 00 42 0f [pyusb_backend]
0000816 D   USB RD < (64) 11 00 42 0f [pyusb_backend]
0000816 D   USB OUT> (6) 04 02 50 00 00 00 [pyusb_backend]
0000818 D   USB IN < (64) 04 00 50 [pyusb_backend]
0000818 D   USB RD < (64) 04 00 50 [pyusb_backend]
0000818 D flush: sending cmd:0; reading 0 outstanding [dap_access_cmsis_dap]
0000818 D   USB OUT> (2) 13 00 [pyusb_backend]
0000820 D   USB IN < (64) 13 [pyusb_backend]
0000820 D   USB RD < (64) 13 [pyusb_backend]
0000820 D trace: swj_sequence(length=51, bits=ffffffffffffff) [cmsis_dap_probe]
0000820 D flush: sending cmd:0; reading 0 outstanding [dap_access_cmsis_dap]
0000820 D   USB OUT> (9) 12 33 ff ff ff ff ff ff ff [pyusb_backend]
0000822 D   USB IN < (64) 12 00 ff ff ff ff ff ff ff [pyusb_backend]
0000822 D   USB RD < (64) 12 00 ff ff ff ff ff ff ff [pyusb_backend]
0000822 D Sending deprecated SWJ sequence to select SWD [swj]
0000822 D trace: swj_sequence(length=16, bits=e79e) [cmsis_dap_probe]
0000822 D flush: sending cmd:0; reading 0 outstanding [dap_access_cmsis_dap]
0000822 D   USB OUT> (4) 12 10 9e e7 [pyusb_backend]
0000824 D   USB IN < (64) 12 00 9e e7 [pyusb_backend]
0000824 D   USB RD < (64) 12 00 9e e7 [pyusb_backend]
0000824 D trace: swj_sequence(length=51, bits=ffffffffffffff) [cmsis_dap_probe]
0000824 D flush: sending cmd:0; reading 0 outstanding [dap_access_cmsis_dap]
0000824 D   USB OUT> (9) 12 33 ff ff ff ff ff ff ff [pyusb_backend]
0000826 D   USB IN < (64) 12 00 ff ff ff ff ff ff ff [pyusb_backend]
0000826 D   USB RD < (64) 12 00 ff ff ff ff ff ff ff [pyusb_backend]
0000826 D trace: swj_sequence(length=8, bits=0) [cmsis_dap_probe]
0000826 D flush: sending cmd:0; reading 0 outstanding [dap_access_cmsis_dap]
0000826 D   USB OUT> (3) 12 08 00 [pyusb_backend]
0000828 D   USB IN < (64) 12 [pyusb_backend]
0000828 D   USB RD < (64) 12 [pyusb_backend]
0000828 D [cmd:0] get_request_space(1, 02:r)[wc=0, rc=0, ba=1->1] -> (sz=1, free=15) [dap_access_cmsis_dap]
0000828 D [cmd:0] add(1, 02:r) -> [wc=0, rc=1, ba=1] [dap_access_cmsis_dap]
[... 102 lines cut ...]
0000844 D   USB RD < (64) 03 [pyusb_backend]
0000844 D trace: close [cmsis_dap_probe]
0000844 D flush: sending cmd:7; reading 0 outstanding [dap_access_cmsis_dap]
0000844 D closing interface [pyusb_backend]
0000845 C No cores were discovered! [__main__]
Traceback (most recent call last):
  File "/home/xxxx/Repo/github/pyocd-test/pyOCD/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
  File "/home/xxxx/Repo/github/pyocd-test/pyOCD/pyocd/subcommands/load_cmd.py", line 96, in invoke
    with session:
  File "/home/xxxx/Repo/github/pyocd-test/pyOCD/pyocd/core/session.py", line 386, in __enter__
    self.open()
  File "/home/xxxx/Repo/github/pyocd-test/pyOCD/pyocd/core/session.py", line 523, in open
    self._board.init()
  File "/home/xxxx/Repo/github/pyocd-test/pyOCD/pyocd/board/board.py", line 124, in init
    self.target.init()
  File "/home/xxxx/Repo/github/pyocd-test/pyOCD/pyocd/core/soc_target.py", line 147, in init
    seq.invoke()
  File "/home/xxxx/Repo/github/pyocd-test/pyOCD/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/home/xxxx/Repo/github/pyocd-test/pyOCD/pyocd/coresight/coresight_target.py", line 256, in check_for_cores
    raise exceptions.DebugError("No cores were discovered!")
pyocd.core.exceptions.DebugError: No cores were discovered!
```
And when I monitored the SWD bus with a logic analyzer and found that I got `error` after `W CTRL/STAT` command flow, is it any idea to solve this issue? Thanks a lot!

TreeNewbeer, 2022-01-21:
And I found that `ADIv5Discovery._find_aps()` function could not find a valid AP though I passed `scan_all_aps` option.(I found that by printing `ap_list` inside the `_find_aps()` founction and it's value was `0`)

flit, 2022-01-21:
Hi @TreeNewbeer, thanks for the detailed trace log.

The very first transfer after the SWD switch sequence is a read of the DP IDR register. That's this line:

```
0000830 D trace: read_dp(addr=0x00000000) -> 0x1be12aeb [cmsis_dap_probe]
```

The result of this read (0x1be12aeb) doesn't make any sense. For an Arm-developed DP, the low order 12 bits should be 0x477. For instance, and ADIv5 DP r2p6 has an IDR of 0x6ba02477, or an ADIv6 DP r3p4 has an IDR of 0x4c013477.

Could there be an SWD connection or signal integrity issue?

I'll review the rest of the trace this weekend to see if anything stands out. But if reading the DP's ID register isn't working, not much else will work either.

TreeNewbeer, 2022-01-23:
Hi @flit , thanks for your reply and it really made sense that let me know I failed to read the IDR.
I am understanding the code sequence called by pyocd and I will do more test about the connection/signal integrity issue.
If anything updated I will push into here, thanks a lot again!

TreeNewbeer, 2022-01-23:
> 
[image]

Here is the command flow from Jlink(and Jlink works well) I monitored by logic analyzer, and I also found the `0x1be12aeb`, but I am not sure why it appeared. So is it any possible that the target I am using has a non-standard  Arm-developed DP?

TreeNewbeer, 2022-01-24:
Hi @flit, I think I have solved my problem and this issue could be closed.
The point is that the target I am using is not a common cortex-m33 based target, it's ARM Star (Maybe "designed" by Arm china :) ), which caused failed to create components at coresight discovery, I add some necessary code into `COMPONENT_MAP` and `AP_TYPE_MAP` table and it works well now.

Thanks for your magical project(PYOCD) and  debug suggestions again!

flit, 2022-01-29:
Ah, the STAR! This is the first time someone has reported trying to use pyocd with it. 😄 I'd like to add support for it. Do you already have the changes to `COMPONENT_MAP` and `AP_TYPE_MAP`?

flit, 2022-01-29:
The 0x1be12aeb DP IDR breaks down to:

- [0] (RAO)
- [11:1] DESIGNER = 0x575
    - JEP106 continuation codes = 10
    - JEP106 ID = 0x75 = "Arm Technology (China) Co Ltd"
- [15:12] VERSION = 0x2
- [16] MIN = 1
- [19:17] (RAZ)
- [27:20] PARTNO = 0xbe
- [31:28] REVISION = 0x1

My sincere apologies that I didn't recognise this as a valid IDR! 

Fwiw, here's the decoding of the AP#0 IDR (0x1aeb0015):

- [3:0] TYPE = 5 (AHB5 MEM-AP)
- [7:4] VARIANT = 1
- [12:8] (RES0)
- [16:13] CLASS = 0
- [27:17] DESIGNER = 0x575 (Arm China)
- [31:28] REVISION = 1

TreeNewbeer, 2022-02-09:
Hi @flit , very sorry for the late reply since I was on the LNY vocation, happy Lunar New Year :smile: and no need to apologies for the IDR issue.

Yes I have changed the `COMPONENT_MAP` and `AP_TYPE_MAP` and add the flash algo that let `pyocd flash` command works. But now I meet a new problem that gdbserver not work now, and the gdbclient has some exception log shown as:

```
Remote debugging using :3333
arch_cpu_idle () at /home/xxxx/zephyrproject/zephyr/arch/arm/core/aarch32/cpu_idle.S:107
107             cpsie   i
Loading section rom_start, size 0x100 lma 0x18000000
Loading section text, size 0x731c lma 0x18000100
Loading section .ARM.exidx, size 0x8 lma 0x1800741c
Loading section initlevel, size 0x38 lma 0x18007424
Loading section devices, size 0x60 lma 0x1800745c
Loading section sw_isr_table, size 0x100 lma 0x180074bc
Loading section device_handles, size 0x2a lma 0x180075bc
--Type <RET> for more, q to quit, c to continue without paging--
Loading section rodata, size 0 lma 40
Loading section datas, size 0x5c lma 0x18007760
Loading section device_states, size 0x10 lma 0x180077bc
Loading section .itcm, size 0xd74 lma 0x180077cc
Loading section csk6001_private, size 0x40 lma 0x18008540
Start address 0x18001e18, load size 34174
Transfer rate: 4 KB/sec, 1265 bytes/write.
(gdb) n
Note: automatically using hardware breakpoints for read-only addresses.
Warning:
Cannot insert hardware breakpoint 0.
Could not insert hardware breakpoints:
You may have requested too many hardware breakpoints/watchpoints.
```

Is it any idea about this? Thanks a lot!

BTW,  I am using the zephyrproject as the target's RTOS and happy to see that pyocd has some code for zephyr RTOS.

flit, 2022-02-13:
Glad the gdbserver is working now! I have access to the TRM and other information for the Star code, so I'll be adding full support in the near term.

First, make sure pyocd is logging a "Zephyr RTOS loaded successfully" message when you connect gdb. For Zephyr, you should also see some debug log messages "_kernel_thread_info_size_t_size = x" and "version = x, _all_threads = 0xy, _curr_thread = 0xz". If these messages aren't in the log (make sure to enable more verbose logging), then you'll need to enable Zephyr's [`CONFIG_DEBUG_THREAD_INFO`](https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_DEBUG_THREAD_INFO.html) Kconfig setting. 

Pyocd has some "read from target" logic that temporarily disables thread awareness until the first resume (`continue` command in the gdb) after a reset or flash programming. This is used to deal with situations where there is a mismatch between memory contents (such as old firmware or blank memory) and RTOS symbol addresses for kernel and thread info, because either new firmware is going to be programmed or RAM still has old contents after a reset.

You can check and control thread awareness with the [`threads` monitor command](https://pyocd.io/docs/command_reference.html#threads-1). The `status`, `enable`, and `disable` actions report the status and allow you to override the above mentioned logic. For example, the gdb command `monitor threads enable` will forcibly bypass the "read from target" check.

Fyi, there are some small fixes to Zephyr support in v0.33.1, mostly reading of thread names.

flit, 2022-03-19:
Hi @TreeNewbeer, if you have a chance could you please test pull request #1315? You can install directly from the branch for this PR using `pip`:

```
pip install git+https://github.com/flit/pyOCD.git@feature/arm_china_star
```

flit, 2022-03-22:
Star-MC1 support has been merged to the `develop` branch. So the install command changes to this:

```
pip install git+https://github.com/flit/pyOCD.git@develop
```

flit, 2022-04-18:
Closing. If you do have issues with the Star-MC1 support added in pyocd, please re-open.

PR #1348 (2022-03-20): Arm China Star-MC1 processor support
Resolves #1315
PR #1349 (2022-03-21): coresight: fix part number for Star MC-1
The TRM says the part numbers are 0xD21, same as CM33. However, data from other sources including #1315 indicate the part number is 0x132. Will double check before merging.

Fixes #1315
closed 2022-04-18 by flit