pyocd/pyOCD #1319: Make pyocd-pemicro an optional dependency

dvzrv, 2022-01-22:
Hi! As I'm currently looking into packaging pyocd for Arch Linux, I'm evaluating the project's dependencies.
I noticed that pypemicro is a dependency via pyocd-pemicro. Unfortunately there are a few open questions around the use of pypemicro in a packaging context (see https://github.com/NXPmicro/pypemicro/issues/10 and https://github.com/NXPmicro/pypemicro/issues/11) and pyocd-pemicro (see https://github.com/pyocd/pyocd-pemicro/issues/7).

Would it be possible to make pyocd-pemicro an optional dependency? It appears to only be required for very specific hardware (I can't find any direct mentions in the source code otherwise):

https://github.com/pyocd/pyOCD/blob/7d164d99e816c4ef6c99f257014543188a0ca5a6/docs/debug_probes.md?plain=1#L23

It would help in getting pyocd packaged sooner while the above issues are being settled and it would additionally allow for users not to install dependencies they do not need.

dvzrv, 2022-01-22:
Somewhat related: https://github.com/pyocd/pyOCD/issues/1158 and https://github.com/pyocd/pyOCD/issues/1160

flit, 2022-01-22:
cc @Gargy007 

Ok, I'll need to review the referenced issues.

You won't find any mentions in the source since pyocd-pemicro is a plugin. And yes, it's only required for specific hardware, the PEMicro debug probes. That's true for all the built-in probe plugins, too, though.

Very unfortunately, there is no way to declare an optional dependency for Python packages which gets installed by default.

One option would be to move pyocd-pemicro to an "extras" dependency, so pyocd would be installed like `pip install pyocd[pemicro]` if you wanted the plugin.

dvzrv, 2022-02-11:
> One option would be to move pyocd-pemicro to an "extras" dependency, so pyocd would be installed like `pip install pyocd[pemicro]` if you wanted the plugin.

Given the unclear licensing and source code around those upstream shared libraries (which could contain backdoored code and therefore can not be trusted) that would be a good approach I believe.

Gargy007, 2022-02-11:
Hello,
next release of pypemicro, makes the license for binaries much more clear.
There is no reason makes changes in SPSDK.

Please be patient for some days.

dvzrv, 2022-02-11:
> Hello, next release of pypemicro, makes the license for binaries much more clear. There is no reason makes changes in SPSDK.

@Gargy007 Thanks for the update on the licensing question.
Please note that even if the license question is settled, some distributions will not consider packaging pypemicro based on the fact that it ships with precompiled shared libraries that are of unknown origin. Personally I also do not feel like this would be a good idea and will abstain from packaging it unless there is access to the sources (in which case this should even be split out to a different project/package as a shared library may be used by other applications as well). Users may still install the package using pip in their user's context though, if they choose so.

flit, 2022-02-19:
@Gargy007 I'm leaning strongly towards making pyocd-pemicro optional, and will probably make the change for the next release (v0.34 since it's more than a simple bug fix). You'd still be able to install at the same time as pyocd with `pip install pyocd[pemicro]`.

Aside from the license/binaries concern, there is #1129. If pyocd-pemicro is installed, running pyocd in PowerShell causes the shell to crash when pyocd exits.

Btw, one way to work around the binaries concern is to remove the binary libs from pypemicro and require the user to install the PEMicro drivers themselves. This is the approach taken for J-Link. While it's an extra step, it's probably (?) reasonable to assume the user will want to install the PEMicro tools anyway. Or you could separate out the libs into yet another package (e.g. pypemicro-libraries), and potentially use an optional install to get them (e.g., `pip install pypemicro[libs]`).

flit, 2022-05-07:
Update: The change to make pyocd-pemicro optional has been merged in #1359. It's available in the `pemicro` install extra, so it's just a small change to install simultaneously with pyocd. The docs have been updated to inform users about this if they need PEMicro probe support.

Closing this issue now that the change has been implemented.

PR #1359 (2022-04-30): setup: make pyocd-pemicro optional
The main purpose of this PR is to move `pyocd-pemicro` to an optional extra install. There are two reasons for this:

1. The PEMicro shared libraries contained in the `pypemicro` Python package are closed source. That's an issue for a number of Linux distros. (The built-in Segger J-Link probe driver also uses a closed source shared library at runtime, but that library is not contained in or installed via a Python package.)

2. For some reason, the PEMicro shared libraries cause Microsoft PowerShell to crash when pyocd exits. See #1129.

Making `pyocd-pemicro` [... cut]
closed 2022-05-07 by flit