pyocd/pyOCD #1324: Debugging Zephyr app on reel board see some python error.

smrtos, 2022-02-03:
Host environment:

- Ubuntu: 20.04.3 LTS (Focal Fossa)
- Python: 3.8.10
- PyOCD: 0.32.0
- Zephyr SDK: 0.13.1

Steps:

- Build zephyr blinky sample application
	cmake -B build -S samples/basic/blinky -GNinja -DBOARD=reel_board
	cd build
	ninja

- Start gdbserver:
	pyocd gdbserver --target=nrf52840 --frequency=4000000
	Output:
```
	  0000275:INFO:board:Target type is nrf52840
	  0000321:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
	  0000337:INFO:ap:AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
	  0000394:INFO:ap:AP#1 IDR = 0x02880000 (AP var0 rev0)
	  0000396:INFO:target_nRF52:NRF52840 not in secure state
	  0000406:INFO:rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=244 part=008)
	  0000412:INFO:rom_table:[0]<e000e000:SCS v7-M class=14 designer=43b part=00c>
	  0000416:INFO:rom_table:[1]<e0001000:DWT v7-M class=14 designer=43b part=002>
	  0000420:INFO:rom_table:[2]<e0002000:FPB v7-M class=14 designer=43b part=003>
	  0000424:INFO:rom_table:[3]<e0000000:ITM v7-M class=14 designer=43b part=001>
	  0000428:INFO:rom_table:[4]<e0040000:TPIU M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=ca1:0:0>
	  0000432:INFO:rom_table:[5]<e0041000:ETM M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
	  0000434:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
	  0000442:INFO:cortex_m:FPU present: FPv4-SP-D16-M
	  0000446:INFO:dwt:4 hardware watchpoints
	  0000448:INFO:fpb:6 hardware breakpoints, 4 literal comparators
	  0000463:INFO:server:Semihost server started on port 4444 (core 0)
	  0000472:INFO:gdbserver:GDB server started on port 3333 (core 0)
```

- Start gdb  client:
	zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb zephyr/build/zephyr/zephyr.elf

- Connect to gdbserver:
```
	>>> target remote :3333
```

- gdb shows below python value error:
[image]

And btw it seems this very python exception will break the [Native Debug extension](https://marketplace.visualstudio.com/items?itemName=webfreak.debug) for the Visual Studio Code. See below:

[image]

flit, 2022-02-06:
Hi @smrtos! I also use both GDB Dashboard for command line debugging, and VSCode for GUI debugging.

The `ValueError: max() arg is an empty sequence` exception you see is a bug in GDB Dashboard. It's not completely happy with the Cortex-M architecture.

I'm not sure where the "Exception has occurred" error comes from. I don't think that's a result of the GDB Dashboard error, since that's a Python exception (and you can see Dashboard continues to print the rest of the dashboard sections). This error is probably related to the Native Debug extension not being meant for embedded debugging.

So basically, the Native Debug extension should not be used with Cortex-M debugging. Please use the [cortex-debug](https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug) extension instead. It works well with pyocd.

Here's an example cortex-debug launch configuration I use for debugging [DAPLink](https://github.com/ARMmbed/DAPLink).

```json
{
    "name": "DAPLink lpc55s69_bl",
    "executable": "${workspaceRoot}/projectfiles/make_gcc_arm/lpc55s69_bl/build/lpc55s69_bl.elf",
    "cwd": "${workspaceRoot}",
    "request": "launch",
    "type": "cortex-debug",
    "servertype": "pyocd",
    "serverpath": "${workspaceRoot}/venv/mac39/bin/pyocd",
    "targetId": "lpc55s69",
    "serverArgs": [
        "--uid=L02932983A",
        "--core=0",
    ],
    "svdFile": "/xxx/_cmsis_packs/NXP.LPC55S69_DFP.12.3.0/LPC55S69_cm33_core0.xml",
    "showDevDebugOutput": false,
    "gdbPath": "/opt/gcc-arm-none-eabi-10.3-2021.07/bin/arm-none-eabi-gdb-py",
},

```

Notes:
- On multicore devices you must include `--core=N` in the `serverArgs` or cortex-debug will try to use clashing server ports for the multiple cores.
- Use pyocd's `-u`/`--uid` argument in `serverArgs` to specify the probe's unique ID, if you have multiple boards connected.
- The `targetId` key is where you put pyocd's `-t`/`--target` argument, if needed.
- cortex-debug config attribute documentation [is here](https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md).

(I plan on adding a full guide to debugging with VSCode+cortex-debug to the pyocd docs soon.)

smrtos, 2022-02-07:
Thanks @flit  I will try the Cortex Debug extension. I remember I disabled the Native Debug extension but the "Exception has occured" still showed up.  I will double check on that and provide further details.

flit, 2022-02-10:
Hmmm, that's a bit frustrating indeed.