pyocd/pyOCD #1327: pyocd 0.33.1 Mass erase failed on NXP IMXRT series

Hoohaha, 2022-02-09:
Tested v0.33.1 version on evkbimxrt1050, the mass erase always failed.

- OS: Windows 10
- Python 3.9

Added `print(reset_type)` in `family/target_imxrt.py line:99`, found that reset_type always is `ResetType.SW_SYSRESETREQ` no matter i changed the default reset type.

If force set `reset_type=self.ResetType.SW` in `def set_reset_catch(self, reset_type=None):` seems it works.

```
"C:\Python39\python.exe" -m pyocd erase --mass -v --no-wait -u 02380000082e00d800000000000000000000000097969905 -Oresume_on_disconnect=false -Oallow_no_cores=true
0001264 I Target type is mimxrt1024 [board]
0001297 I DP IDR = 0x0bd11477 (v1 MINDP rev0) [dap]
0001328 I AHB-AP#0 IDR = 0x04770041 (AHB-AP var4 rev0) [ap]
0001375 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fd000 (designer=00e part=88c) [rom_table]
0001375 I [0]<e00fe000:ROM class=1 designer=43b part=4c8> [rom_table]
0001375 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00fe000 (designer=43b part=4c8) [rom_table]
0001394 I   [0]<e00ff000:ROM class=1 designer=43b part=4c7> [rom_table]
0001394 I     AHB-AP#0 Class 0x1 ROM table #2 @ 0xe00ff000 (designer=43b part=4c7) [rom_table]
0001398 I     [0]<e000e000:SCS v7-M class=14 designer=43b part=00c> [rom_table]
0001400 I     [1]<e0001000:DWT v7-M class=14 designer=43b part=002> [rom_table]
0001403 I     [2]<e0002000:FPB v7-M class=14 designer=43b part=00e> [rom_table]
0001403 I     [3]<e0000000:ITM v7-M class=14 designer=43b part=001> [rom_table]
0001403 I   [1]<e0041000:ETM M7 class=9 designer=43b part=975 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0001403 I   [2]<e0042000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0001453 I [1]<e0040000:TPIU M7 class=9 designer=43b part=9a9 devtype=11 archid=0000 devid=ca1:0:0> [rom_table]
0001453 I [2]<e0043000:TSGEN class=15 designer=43b part=101> [rom_table]
0001469 I IMXRT Boot Mode: Serial Download Mode [target_imxrt]
0001469 I CPU core #0 is Cortex-M7 r1p1 [cortex_m]
0001484 I FPU present: FPv5-D16-M [cortex_m]
0001505 I 4 hardware watchpoints [dwt]
0001512 I 8 hardware breakpoints, 1 literal comparators [fpb]
0001687 I Mass erasing device... [eraser]
0001687 I Erasing chip... [eraser]
0001719 I IMXRT Boot Mode: Serial Download Mode [target_imxrt]
0001719 I IMXRT Boot Device: 0 [target_imxrt]
ResetType.SW_SYSRESETREQ
0055364 I Done [eraser]
0055364 E Mass erase failed. [eraser]
0001264 I Target type is mimxrt1024 [board]
0001297 I DP IDR = 0x0bd11477 (v1 MINDP rev0) [dap]
0001328 I AHB-AP#0 IDR = 0x04770041 (AHB-AP var4 rev0) [ap]
0001375 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fd000 (designer=00e part=88c) [rom_table]
0001375 I [0]<e00fe000:ROM class=1 designer=43b part=4c8> [rom_table]
0001375 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00fe000 (designer=43b part=4c8) [rom_table]
0001394 I   [0]<e00ff000:ROM class=1 designer=43b part=4c7> [rom_table]
0001394 I     AHB-AP#0 Class 0x1 ROM table #2 @ 0xe00ff000 (designer=43b part=4c7) [rom_table]
0001398 I     [0]<e000e000:SCS v7-M class=14 designer=43b part=00c> [rom_table]
0001400 I     [1]<e0001000:DWT v7-M class=14 designer=43b part=002> [rom_table]
0001403 I     [2]<e0002000:FPB v7-M class=14 designer=43b part=00e> [rom_table]
0001403 I     [3]<e0000000:ITM v7-M class=14 designer=43b part=001> [rom_table]
0001403 I   [1]<e0041000:ETM M7 class=9 designer=43b part=975 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0001403 I   [2]<e0042000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0001453 I [1]<e0040000:TPIU M7 class=9 designer=43b part=9a9 devtype=11 archid=0000 devid=ca1:0:0> [rom_table]
0001453 I [2]<e0043000:TSGEN class=15 designer=43b part=101> [rom_table]
0001469 I IMXRT Boot Mode: Serial Download Mode [target_imxrt]
0001469 I CPU core #0 is Cortex-M7 r1p1 [cortex_m]
0001484 I FPU present: FPv5-D16-M [cortex_m]
0001505 I 4 hardware watchpoints [dwt]
0001512 I 8 hardware breakpoints, 1 literal comparators [fpb]
0001687 I Mass erasing device... [eraser]
0001687 I Erasing chip... [eraser]
0001719 I IMXRT Boot Mode: Serial Download Mode [target_imxrt]
0001719 I IMXRT Boot Device: 0 [target_imxrt]
ResetType.SW_SYSRESETREQ
0055364 I Done [eraser]
0055364 E Mass erase failed. [eraser]
```

mathias-arm, 2022-02-09:
Based on your logs `Target type is mimxrt1024 [board]` and the probe ID, it's probably a MIMXRT1024-EVK board and not `evkbimxrt1050` (or something is wrong with the probe firmware).

Hoohaha, 2022-02-09:
Sorry, pasted wrong log.  

Tested on rt1024, rt1050, rt1060, rt1064, confirmed they all have this issue.
And downgrade to v0.32.1, pyocd works fine with these evk boards. 

RT1050-EVKB
```
"C:\Python39\python.exe" -m pyocd erase --mass -v --no-wait -u 0227000047784e45001b9003d7450045ddb1000097969900 -Oresume_on_disconnect=false -Oallow_no_cores=true
0001127 I Target type is mimxrt1050_hyperflash [board]
0001187 I DP IDR = 0x0bd11477 (v1 MINDP rev0) [dap]
0001201 I AHB-AP#0 IDR = 0x04770041 (AHB-AP var4 rev0) [ap]
0001250 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fd000 (designer=00e part=88c) [rom_table]
0001250 I [0]<e00fe000:ROM class=1 designer=43b part=4c8> [rom_table]
0001250 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00fe000 (designer=43b part=4c8) [rom_table]
0001265 I   [0]<e00ff000:ROM class=1 designer=43b part=4c7> [rom_table]
0001265 I     AHB-AP#0 Class 0x1 ROM table #2 @ 0xe00ff000 (designer=43b part=4c7) [rom_table]
0001265 I     [0]<e000e000:SCS v7-M class=14 designer=43b part=00c> [rom_table]
0001265 I     [1]<e0001000:DWT v7-M class=14 designer=43b part=002> [rom_table]
0001281 I     [2]<e0002000:FPB v7-M class=14 designer=43b part=00e> [rom_table]
0001285 I     [3]<e0000000:ITM v7-M class=14 designer=43b part=001> [rom_table]
0001328 I   [1]<e0041000:ETM M7 class=9 designer=43b part=975 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0001328 I   [2]<e0042000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0001344 I [1]<e0040000:TPIU M7 class=9 designer=43b part=9a9 devtype=11 archid=0000 devid=ca1:0:0> [rom_table]
0001344 I [2]<e0043000:TSGEN class=15 designer=43b part=101> [rom_table]
0001344 I IMXRT Boot Mode: Serial Download Mode [target_imxrt]
0001344 I CPU core #0 is Cortex-M7 r1p1 [cortex_m]
0001360 I FPU present: FPv5-D16-M [cortex_m]
0001437 I 4 hardware watchpoints [dwt]
0001468 I 8 hardware breakpoints, 1 literal comparators [fpb]
0001562 I Mass erasing device... [eraser]
0001562 I Erasing chip... [eraser]
0001593 I IMXRT Boot Mode: Serial Download Mode [target_imxrt]
0001609 I IMXRT Boot Device: 0 [target_imxrt]
0005297 I Done [eraser]
0005297 E Mass erase failed. [eraser]
0001127 I Target type is mimxrt1050_hyperflash [board]
0001187 I DP IDR = 0x0bd11477 (v1 MINDP rev0) [dap]
0001201 I AHB-AP#0 IDR = 0x04770041 (AHB-AP var4 rev0) [ap]
0001250 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fd000 (designer=00e part=88c) [rom_table]
0001250 I [0]<e00fe000:ROM class=1 designer=43b part=4c8> [rom_table]
0001250 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00fe000 (designer=43b part=4c8) [rom_table]
0001265 I   [0]<e00ff000:ROM class=1 designer=43b part=4c7> [rom_table]
0001265 I     AHB-AP#0 Class 0x1 ROM table #2 @ 0xe00ff000 (designer=43b part=4c7) [rom_table]
0001265 I     [0]<e000e000:SCS v7-M class=14 designer=43b part=00c> [rom_table]
0001265 I     [1]<e0001000:DWT v7-M class=14 designer=43b part=002> [rom_table]
0001281 I     [2]<e0002000:FPB v7-M class=14 designer=43b part=00e> [rom_table]
0001285 I     [3]<e0000000:ITM v7-M class=14 designer=43b part=001> [rom_table]
0001328 I   [1]<e0041000:ETM M7 class=9 designer=43b part=975 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0001328 I   [2]<e0042000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0001344 I [1]<e0040000:TPIU M7 class=9 designer=43b part=9a9 devtype=11 archid=0000 devid=ca1:0:0> [rom_table]
0001344 I [2]<e0043000:TSGEN class=15 designer=43b part=101> [rom_table]
0001344 I IMXRT Boot Mode: Serial Download Mode [target_imxrt]
0001344 I CPU core #0 is Cortex-M7 r1p1 [cortex_m]
0001360 I FPU present: FPv5-D16-M [cortex_m]
0001437 I 4 hardware watchpoints [dwt]
0001468 I 8 hardware breakpoints, 1 literal comparators [fpb]
0001562 I Mass erasing device... [eraser]
0001562 I Erasing chip... [eraser]
0001593 I IMXRT Boot Mode: Serial Download Mode [target_imxrt]
0001609 I IMXRT Boot Device: 0 [target_imxrt]
0005297 I Done [eraser]
0005297 E Mass erase failed. [eraser]
```

flit, 2022-02-13:
Hi @Hoohaha. (Sorry for the delayed reply, I was moving to a new house this past week.) I'll move this to top priority.

flit, 2022-05-07:
Hi @Hoohaha.

I'm not seeing any failure to erase flash when testing this with an unmodified pyocd v0.33.1. It does report that "Mass erase failed" (which will be fixed in the next release), but the erase actually happens.

### Pre-erase check

```
$ pyocd cmd -tmimxrt1050_hyperflash -c 'rw 0x0800_0000 32'
08000000:  42464346 56010400 00000000 03030303    |BFCFV...........|
08000010:  00000000 00000000 00000000 00000000    |................|
```

### Erase

```
$ pyocd erase --mass -Oresume_on_disconnect=0 -L*cortex_m=debug -v -tmimxrt1050_hyperflash
...
0002744 D halting core 0 [cortex_m]
0002797 I Mass erasing device... [eraser]
0002797 I Erasing chip... [eraser]
0002797 D halting core 0 [cortex_m]
0002877 I IMXRT Boot Mode: Internal Boot [target_imxrt]
0002916 I IMXRT Boot Device: 0 [target_imxrt]
0002916 D set reset catch, core 0 [cortex_m]
0002916 D halting core 0 [cortex_m]
0003025 D reset, core 0, type=SW_SYSRESETREQ [cortex_m]
0003568 D clear reset catch, core 0 [cortex_m]
0005261 D resuming core 0 [cortex_m]
0005861 D resuming core 0 [cortex_m]
0010627 D resuming core 0 [cortex_m]
0010882 I Done [eraser]
0010883 E Mass erase failed. [eraser]
```

Note the line 0003025 that says the reset type is `SW_SYSRESETREQ`. The reported reset type is the same if I pass `-Oreset_type=hw`.

### Post-erase check

```
$ pyocd cmd -tmimxrt1050_hyperflash -c 'rw 0x0800_0000 32'
08000000:  ffffffff ffffffff ffffffff ffffffff    |................|
08000010:  ffffffff ffffffff ffffffff ffffffff    |................|
```

## Reset type bug

You are completely correct that the specified reset type is not used in the erase. When the flash algo is initialised, this line from 3 years ago tells it to always uses software reset:
https://github.com/pyocd/pyOCD/blob/b870b8ba2c11810581d5ab9d776b0cf2b63112fa/pyocd/flash/flash.py#L232

I have no idea why it was forced to software reset. Maybe because that's supposedly "always" available?

This will be fixed in the next release to use the default reset type, or the `reset_type` option if that is set.

flit, 2022-07-05:
Closing; should be fixed now.

closed 2022-07-05 by flit