pyocd/pyOCD #1338: pyOCD fails without cmsis-pack-manager despite docs suggesting it's optional

kcp-chewie, 2022-02-15:
The docs suggest that pyOCD can be used without cmsis-pack-manager (https://pyocd.io/docs/installing_on_non_x86.html), but I've observed several failures when it isn't installed.

The first error is due to an issue with type annotations
```
$ python
Python 3.9.2 (default, Mar 12 2021, 04:06:34) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyocd
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/pyocd/__init__.py", line 21, in <module>
    from . import gdbserver
  File "/usr/local/lib/python3.9/dist-packages/pyocd/gdbserver/__init__.py", line 17, in <module>
    from .gdbserver import GDBServer
  File "/usr/local/lib/python3.9/dist-packages/pyocd/gdbserver/gdbserver.py", line 37, in <module>
    from .syscall import GDBSyscallIOHandler
  File "/usr/local/lib/python3.9/dist-packages/pyocd/gdbserver/syscall.py", line 18, in <module>
    from ..debug.semihost import SemihostIOHandler
  File "/usr/local/lib/python3.9/dist-packages/pyocd/debug/semihost.py", line 26, in <module>
    from ..core import (exceptions, session)
  File "/usr/local/lib/python3.9/dist-packages/pyocd/core/session.py", line 30, in <module>
    from ..board.board import Board
  File "/usr/local/lib/python3.9/dist-packages/pyocd/board/board.py", line 23, in <module>
    from ..target.pack import pack_target
  File "/usr/local/lib/python3.9/dist-packages/pyocd/target/pack/pack_target.py", line 55, in <module>
    class ManagedPacksImpl:
  File "/usr/local/lib/python3.9/dist-packages/pyocd/target/pack/pack_target.py", line 64, in ManagedPacksImpl
    def get_installed_packs(cache: Optional[cmsis_pack_manager.Cache] = None) -> List["CmsisPackRef"]: # type:ignore
NameError: name 'cmsis_pack_manager' is not defined
```

This issue can be fixed by adding `from __future__ import annotations` to pyocd/target/pack/pack_target.py as suggested here: https://mypy.readthedocs.io/en/stable/runtime_troubles.html#future-annotations-import-pep-563

After fixing that error, the following issue is observed
```
$ python -m pyocd
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/kevin/code/pyOCD/pyocd/__init__.py", line 21, in <module>
    from . import gdbserver
  File "/Users/kevin/code/pyOCD/pyocd/gdbserver/__init__.py", line 17, in <module>
    from .gdbserver import GDBServer
  File "/Users/kevin/code/pyOCD/pyocd/gdbserver/gdbserver.py", line 41, in <module>
    from ..rtos import RTOS
  File "/Users/kevin/code/pyOCD/pyocd/rtos/__init__.py", line 29, in <module>
    load_plugin_classes_of_type('pyocd.rtos', RTOS, ThreadProvider)
  File "/Users/kevin/code/pyOCD/pyocd/core/plugin.py", line 97, in load_plugin_classes_of_type
    plugin = entry_point.load()()
  File "/Users/kevin/code/pyOCD/venv/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2464, in load
    self.require(*args, **kwargs)
  File "/Users/kevin/code/pyOCD/venv/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2487, in require
    items = working_set.resolve(reqs, env, installer, extras=self.extras)
  File "/Users/kevin/code/pyOCD/venv/lib/python3.9/site-packages/pkg_resources/__init__.py", line 777, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'cmsis-pack-manager<1.0,>=0.4.0' distribution was not found and is required by the application
```

The docs suggest it should be runnable through Python

> Once pyOCD is successfully installed, you will need to run it as a module using the python executable as python -mpyocd rather than running the pyocd executable directly. This is because the pyocd executable, which is auto-generated by the install process, verifies dependencies and will error out due to the missing cmsis-pack-manager. But when run through python, dependencies are not checked and pyOCD can handle the lack of cmsis-pack-manager gracefully.

linfan68, 2022-05-29:
I got the same error when running this on a RPi,

LeBlue, 2022-06-11:
Calling as a module (`python3 -mpyocd -h`) still seems to verify dependencies.

For just getting it to work one can remove the cmsis-pack-manager dependency in `pyocd.egg-info/requires.txt` in the `site-packages` (and add the `from __future__ import annotations` in `pyocd/target/pack_target.py`).