pyocd/pyOCD #1340: Vega VT-Link II (formerly known as Essemi ES-Link II) Support

Entropy98, 2022-02-21:
# Description
  
pyOCD does not support the CMSIS compliant debugger VT-Link II. The debugger is CMSIS compliant but does not contain CMSIS in the product string
  
### Justification
  
Expanding the compatibility of pyOCD to work with more debuggers especially in this time of chip shortages where go-to MCUs are not available

flit, 2022-02-21:
Interesting, I had not heard of this probe. (To be pedantic, if the string "CMSIS-DAP" isn't in either the USB product name or interface name strings, then it's technically not CMSIS-DAP compliant. 😉)

Which version of CMSIS-DAP does it support? More specifically, v1 with a HID interface or v2 with vendor-specific bulk interface?

Can you create a pull request to add support? Otherwise it will have to be low priority since I can't test and there are plenty of other issues affecting more users.

You'll need to modify the code in the [`pyocd.probe.pydapaccess.interface`](https://github.com/pyocd/pyOCD/tree/main/pyocd/probe/pydapaccess/interface) package to support matching against USB VID/PID pairs. Which backends depends on HID vs bulk.

There is already a function `is_known_cmsis_dap_vid_pid()`: https://github.com/pyocd/pyOCD/blob/6a6fa2834d3aed120125bbef04e69f9f6118dda2/pyocd/probe/pydapaccess/interface/common.py#L86-L88

The different backends currently use this to decide whether to log more detailed error logs, mostly for Linux dealing with udev permissions. The `KNOWN_CMSIS_DAP_IDS` table should be updated as appropriate, and the function integrated into the pyusb match callbacks.

Entropy98, 2022-02-21:
I've made a hot fix in the Find_DAP function to make it compatible. You can look at it here: https://github.com/Entropy98/pyOCD/commit/6b6e1a66b0566b86b086ca509da2db64fe6f7320 but I'm going to clean it up before making a formal pull request. Do you have a suggestion on the best way to implement this? I can show test data of it working.

flit, 2022-02-21:
I tend to prefer generic changes if possible, so I would modify the match callback to also accept a device if `is_known_cmsis_dap_vid_pid()` returns True. Also, need to implement the same change in `hidapi_backend.py` and `pywinusb_backend.py` to support all OSes. Thanks!

closed 2022-03-12 by flit