pyocd/pyOCD #1342: write_memory does not change the memory state

vacagonzalo, 2022-02-26:
# write_memory does not change the memory state

Intended use: By using *commander* or *API* make bit flips in the core registers and SDRAM in order to simulate cosmic radiation.

I was not able to achieve the simulation in the SDRAM. **target.write_memory(addr, data)** does not change the value in memory.

## Aditional information

### PyOCD

Version 0.33.1

### Board

NUCLEO-F429ZI

### Computer

OS: Linux Mint 20.3 x86_64 
Host: MAX G0101 MAX G0101 
Kernel: 5.4.0-100-generic 
Uptime: 4 hours, 17 mins 
Packages: 2759 (dpkg), 8 (flatpak) 
Shell: bash 5.0.17 
Resolution: 1366x768, 1600x900 
DE: Cinnamon 
WM: Mutter (Muffin) 
WM Theme: Mint-Y (Mint-Y) 
Theme: Mint-Y [GTK2/3] 
Icons: Mint-Y [GTK2/3] 
Terminal: gnome-terminal 
CPU: Intel i7-4710MQ (8) @ 3.500GHz 
GPU: Intel 4th Gen Core Processor 
Memory: 2110MiB / 11887MiB

flit, 2022-02-27:
Hi @vacagonzalo,

Questions:
- Are you using the on-board STLink2 probe?
- Have you updated the STLink firmware to the latest? (The memory transfer details are handled by the STLink firmware.)
- It fails both with commander and the Python API?
- Is a transfer fault exception raised?
- How are you checking whether the write was successful?
- If you repeat the same write many times (10s-100s or more), does it eventually take effect? If so, this implies caching.

Keep in mind that in order to use a (any) debugger with more complex memory architectures such as SDRAM, you'll need to thoroughly understand how the memory and bus architecture of the MCU works in detail. I'm happy to help as much as possible, but I don't know the F429 in depth.

Things to check:
- Is there a cache or FIFO that needs to be flushed?
- Is the FMC peripheral configured properly for the SDRAM at the time you are writing? (Always worth checking! 😄)

Lastly, we can verify that the write memory command is actually sent correctly to the STLink by adding `-L*stlink.usb.trace=debug` to the pyocd command line. This will dump raw USB transfer data. The write memory command will start with `f2 08` for a 32-bit write, followed by the 4 address bytes in little endian order. (You can also attach the relevant part of the log output to a comment here and I can help decode it.)

vacagonzalo, 2022-02-27:
# SOLVED

I was using the wrong address to write in the SDRAM. I am sorry to make you waste time. This ticket should be closed.

## Questions:

> Are you using the on-board STLink2 probe?

Yes, I am.

> Have you updated the STLink firmware to the latest? (The memory transfer details are handled by the STLink firmware.)

Yes, It is in the last version with the following details:

  * Version: V2J39M27
  * Type: STM32 Debug + Mass storage + VCP

> It fails both with commander and the Python API?

Yes, it fails in both

> Is a transfer fault exception raised?

No, there are no exceptions

> How are you checking whether the write was successful?

with a halted target I do the following:

  1. read the memory value and store it in **old** variable.
  2. create a **flip** variable with the value of **old** and flip one bit.
  3. write **flip** value in the memory addr.
  4. read the memory value and store it in **new** variable.
  5. return a tuple (old, new)

This is my implementation:

```python
def bitFlipMemory(self, addr:int, bit:int) -> tuple:
    mask = 1 << bit
    with Halted(self.target):
        old = self.target.read_memory(addr)
        flip = old ^ mask
        self.target.write_memory(addr, flip)
        new = self.target.read_memory(addr)
    return (old, new)
```

Halted class:

```python
class Halted():
    def __init__(self, target):
        self.target = target

    def __enter__(self):
        self.target.halt()

    def __exit__(self, exc_type, exc_val, traceback):
        self.target.resume()
```

> If you repeat the same write many times (10s-100s or more), does it eventually take effect? If so, this implies caching.

> Keep in mind that in order to use a (any) debugger with more complex memory architectures such as SDRAM, you'll need to thoroughly understand how the memory and bus architecture of the MCU works in detail. I'm happy to help as much as possible, but I don't know the F429 in depth.

**THIS WAS THE SOLUTION, I WAS USING THE WRONG ADDRESS**

## Things to check

> Is there a cache or FIFO that needs to be flushed?
No, there is not.

> Is the FMC peripheral configured properly for the SDRAM at the time you are writing? (Always worth checking! 😄)
Yes, it was.

closed 2022-02-27 by vacagonzalo

flit, 2022-02-27:
Thanks for the update. Glad it's working for you now!