pyocd/pyOCD #1351: Support for STM32U5

daniel-merrick, 2022-03-21:
Are there any plans to add support for STM32U5 NUCLEO boards?

https://www.st.com/en/evaluation-tools/nucleo-u575zi-q.html

flit, 2022-03-21:
Those will be supported via CMSIS-Packs unless there is a strong reason to have built-in support, like the Packs don't work. (CMSIS-Packs are the preferred way to add new targets unless there are barriers to doing so.)

daniel-merrick, 2022-03-21:
Ah I see - So I download the **Keil.STM32U5xx_DFP.1.1.1.pack** then pass ''--pack=Keil.STM32U5xx_DFP.1.1.1.pack'' to the PYOCD command. 

Thank you!

closed 2022-03-21 by daniel-merrick

flit, 2022-03-21:
Actually you can use `pyocd pack install u575` to install the pack and not have to download it manually. This also makes the target types available without needing the `--pack` argument.

The [target support documentation](https://pyocd.io/docs/target_support.html) has all the details.

daniel-merrick, 2022-03-21:
Installing like that doesn't seem to work properly: This is the error I receive 

```
pyocd pack install u575
0000741 W No matching devices. Please make sure the pack index is up to date. [pack_cmd]
```

flit, 2022-03-21:
That's odd. I just tried it again and it worked fine, even without have a pack index installed previously. I guess it's possible you have an out of date pack index from before the U5xx pack was added?

Please try adding `-u` or `--update` to the command line: `pyocd pack install -u u575`. This will ensure the pack index is up to date before checking (you'll get a bunch of warnings for XML formatted errors in the numerous packs).

You can also try: `pyocd pack clean ; pyocd pack update`, then run the install command. This will completely erase and re-download the pack index before attempting to install. (It should be more or less equivalent to the `-u` argument, but could fix an index that was somehow corrupted.)

flit, 2022-03-21:
Very interesting! Thanks for trying and I'm glad it worked eventually…

One strange thing is that the warnings are different between the 2 pack updates, and also different from my own tests.