pyocd/pyOCD #1356: [question] How to download a binary file to external flash

xray-bit, 2022-03-30:
I have used [pyOCD/FlashAlgo](https://github.com/pyocd/FlashAlgo) to generate an FLM file for external flash. How to download a binary file to external flash using CLI or API? I've searched on docs, issues and google, but couldn't find any relevant questions.

xray-bit, 2022-03-31:
After I cloned pyOCD, I added a new target referring to the [docs/adding_new_targets.md](https://github.com/pyocd/pyOCD/blob/main/docs/adding_new_targets.md) documentation. External flash can now be programmed using the CLI. 
Now my question is, after opening a `session` using API, is it possible to use one algorithm to program the internal flash and then switch to another algorithm to program the external flash?

xray-bit, 2022-04-06:
[docs/user_scripts.md](https://github.com/pyocd/pyOCD/blob/main/docs/user_scripts.md)

closed 2022-04-06 by xray-bit

elfmimi, 2022-04-07:
Maybe link to pyocd.io is better for sharing.
https://pyocd.io/docs/user_scripts.html