pyocd/pyOCD #1357: nRF5340 net core flash

Vulpiculus, 2022-03-31:
I am doing some trials with PyOCD + DAP link (probe is Max32625PICO) with the dual core nRF5340. I installed the pack nrf5340_xxaa.
Flashing the APP core is no issue and everything goes well. But when trying to flash the NET core I get :
`Failed to add data chunk: no memory region defined for address 0x01000000 [file_programmer]`

I dived a bit in the device pack and the parser and found out the ROM region for NET core is beeing removed and not replaced by a FLASH region as it should (And as it does for the APP core)
NET:
[image]
APP:
[image]

Problem seems to be in _split_flash_region_by_sector_size where we do `start = region.start + offset` and `start` ends up equal `end` and as a result the FLASH region is not added. Changing this line then gives an error about the NET core RAM region:
`pyocd.core.exceptions.TransferFaultError: Memory transfer fault @ 0x21000200-0x210003ff`

- Is nRF5340 really fully supported by PyOCD ?
- If yes, What am I doing something wrong ?

Thank you !

peterniebert, 2022-06-09:
I wonder about this issue too. The device pack exposes two devices, one with _APP suffix and the other with _NET, but pyocd lists just one device without suffix? And with that device it is possible to flash the APP cpu.

GatCode, 2023-03-18:
Are there any updates? I also have the same issue.

maxd-nordic, 2024-01-01:
I'm working on it: https://github.com/pyocd/pyOCD/pull/1656