pyocd/pyOCD #1358: Stuck in probe debugger after packaging script with pyinstaller

xray-bit, 2022-04-01:
A python script written using the pyOCD API is packaged into an executable file using pyinstaller. After the program runs, it has been waiting for the debugger to connect (console output `Waiting for a debug probe to be connected...`), but the debugger (DAP Link) is always connected.
However, by running the python script directly, the connected debugger can be obtained normally.

my simple script:
```python
from pyocd.core.helpers import ConnectHelper

def main():
    debuggers = ConnectHelper.get_all_connected_probes(blocking=True)
    print(debuggers[0].description)

if __name__ == '__main__':
    main()
```

and `.spec` file:
```spec
# -*- mode: python ; coding: utf-8 -*-
# reference: [flit: feature/binary_build](https://github.com/flit/pyOCD/tree/feature/binary_build)
import glob
import platform
from PyInstaller.utils.hooks import (get_package_paths, collect_dynamic_libs)

block_cipher = None

is_windows = (platform.system() == "Windows")

# Get binary extension libraries...

# Capstone for disassembly.
capstone_libs = collect_dynamic_libs("capstone")

# CPM's native lib doesn't match the patterns that collect_dynamic_libs() expects.
cpm_path = get_package_paths('cmsis_pack_manager')[1]
if is_windows:
    # Example: _native__lib.cp37-win_amd64.pyd
    ##############################################################
    # modified. because my python310 is 64bits or matches[0] will return a win32 lib
    ##############################################################
    matches = glob.glob(os.path.join(cpm_path, "*amd64.pyd"))
    print("matches", matches)
    if matches:
        cpm_lib_name = matches[0]
    else:
        raise Exception("failed to find cmsis-pack-manager native library")
else:
    cpm_lib_name = "_native__lib.so"
cpm_libs = [(os.path.join(cpm_path, cpm_lib_name), "cmsis_pack_manager")]

binaries = capstone_libs + cpm_libs
# and add libusb
binaries += [(r'G:\Program\Python3\Lib\site-packages\libusb_package\libusb-1.0.dll', 'libusb_package')]

print("binaries", binaries)

a = Analysis(['flash.py'],
             pathex=[],
             binaries=binaries,
             datas=[],
             hiddenimports=[],
             hookspath=[],
             hooksconfig={},
             runtime_hooks=[],
             excludes=[],
             win_no_prefer_redirects=False,
             win_private_assemblies=False,
             cipher=block_cipher,
             noarchive=False)
pyz = PYZ(a.pure, a.zipped_data,
             cipher=block_cipher)

exe = EXE(pyz,
          a.scripts, 
          #a.binaries,
          #a.zipfiles,
          #a.datas,  
          [],
          exclude_binaries=True, # add
          name='flash',
          debug=False,
          bootloader_ignore_signals=False,
          strip=False,
          upx=True,
          #upx_exclude=[],
          #runtime_tmpdir=None,
          console=True,
          disable_windowed_traceback=False,
          target_arch=None,
          codesign_identity=None,
          entitlements_file=None )

coll = COLLECT(exe,
               a.binaries,
               a.zipfiles,
               a.datas, 
               strip=False,
               upx=True,
               upx_exclude=[],
               name='flash')

```

xray-bit, 2022-04-01:
I guessed it was a libusb problem earlier. So I tested libusb_package.
I copied `Usage example for find_library():` from https://github.com/pyocd/libusb-package:
```python
import libusb_package
import usb.core
import usb.backend.libusb1

libusb1_backend = usb.backend.libusb1.get_backend(find_library=libusb_package.find_library)
print(list(usb.core.find(find_all=True, backend=libusb1_backend)))
```

1. The `binaries` in `.spec` keep the default empty array. Then use pyinstaller to package and run, and a "usb.core.NoBackendError: No backend available" error occurs.
2. Change the `binaries` in `.spec` to `binaries = [("C:\\Windows\\SysWOW64\\libusb-1.0.dll", ".")]`, then use pyinstaller to package and run, then it can get the USB Device List

and the `.spec`:
```python
# -*- mode: python ; coding: utf-8 -*-

block_cipher = None

binaries = [("C:\\Windows\\SysWOW64\\libusb-1.0.dll", ".")]

a = Analysis(['usbtest.py'],
             pathex=[],
             binaries=binaries,
             datas=[],
             hiddenimports=[],
             hookspath=[],
             hooksconfig={},
             runtime_hooks=[],
             excludes=[],
             win_no_prefer_redirects=False,
             win_private_assemblies=False,
             cipher=block_cipher,
             noarchive=False)
pyz = PYZ(a.pure, a.zipped_data,
             cipher=block_cipher)

exe = EXE(pyz,
          a.scripts,
          a.binaries,
          a.zipfiles,
          a.datas,  
          [],
          name='usbtest',
          debug=False,
          bootloader_ignore_signals=False,
          strip=False,
          upx=True,
          upx_exclude=[],
          runtime_tmpdir=None,
          console=True,
          disable_windowed_traceback=False,
          target_arch=None,
          codesign_identity=None,
          entitlements_file=None )
```

xray-bit, 2022-04-06:
[discussions/1274](https://github.com/pyocd/pyOCD/discussions/1274)

xinkunlong, 2022-06-20:
[image]
I tried your method, but I still couldn't.

chencen-cz, 2022-06-22:
Hello, @xray-bit. I had the same problem, running the executable file is always "Waiting for a debug probe to be connected...". 

**Environment:**
OS: Windows 10 Enterprise 1909
Python: 3.10.0 (64-bit)
pyOCD: tag v0.34.0 (0.33.1 can not work normally by packing an the executable file)
pyinstaller: 4.10

qujilai, 2023-01-17:
> have you solve this? i have the same problem

chencen-cz, 2023-03-07:
> 

Hi @qujilai, I'm so sorry to late to see, I had solved this by .spec modification.

USTHzhanglu, 2023-03-15:
> > 
> 
> Hi @qujilai, I'm so sorry to late to see, I had solved this by .spec modification.

Hi, i have the same problem. how do you solve this? specify libusb.dll in spec?

autumn-qmj, 2024-04-19:
> [image] I tried your method, but I still couldn't.

Have you fixed this issue >?

autumn-qmj, 2024-04-19:
> > 
> 
> Hi @qujilai, I'm so sorry to late to see, I had solved this by .spec modification.

can you share the details how to fix it   ? Thanks

autumn-qmj, 2024-04-19:
> > > 
> > 
> > 
> > Hi @qujilai, I'm so sorry to late to see, I had solved this by .spec modification.
> 
> can you share the details how to fix it ? Thanks

I already added the libusb-1.0-dll in my .spec file, and it is packed into the .exe, but still the same issue as below:
[image]