pyocd/pyOCD #1362: ATSAMD51N20A has no default RAM defined

nedimlisica, 2022-04-09:
I'm trying to debug ATSAMD51N20A using J-Link Plus as a probe, but I'm getting `CMSIS-Pack device ATSAMD51N20A has no default RAM defined, cannot program flash [cmsis_pack]`. Here is the full CLI output.
```
> pyocd gdb

0001123 I Using config options for probe xxxxxxxxx [session]
0001128 W Overlapping memory regions in file Microchip.SAMD51_DFP.3.6.120.pack (ATSAMD51N20A); deleting outer region. Further warnings will be suppressed for this file. [cmsis_pack]
0001128 W CMSIS-Pack device ATSAMD51N20A has no default RAM defined, cannot program flash [cmsis_pack]
0001128 I Target type is atsamd51n20a [board]
Exception in thread load-svd:
Traceback (most recent call last):
  File "C:\Program Files\Python310\lib\threading.py", line 1009, in _bootstrap_inner
    self.run()
  File "C:\Program Files\Python310\lib\site-packages\pyocd\debug\svd\loader.py", line 71, in run
    self._svd_location.load()
  File "C:\Program Files\Python310\lib\site-packages\pyocd\debug\svd\loader.py", line 47, in load
    self.device = SVDParser.for_xml_file(self.filename).get_device()
  File "C:\Program Files\Python310\lib\site-packages\pyocd\debug\svd\parser.py", line 74, in for_xml_file
    return cls(ET.parse(path), remove_reserved)
  File "C:\Program Files\Python310\lib\xml\etree\ElementTree.py", line 1229, in parse
    tree.parse(source, parser)
  File "C:\Program Files\Python310\lib\xml\etree\ElementTree.py", line 569, in parse
    source = open(source, "rb")
TypeError: expected str, bytes or os.PathLike object, not NoneType
0001163 I DP IDR = 0x2ba01477 (v1 rev2) [dap]
0001166 E Exception while probing AP#2: Memory transfer fault (Unspecified error.) [discovery]
Traceback (most recent call last):
  File "C:\Program Files\Python310\lib\site-packages\pyocd\probe\jlink_probe.py", line 311, in read_ap
    value = self._link.coresight_read((addr & self.A32) // 4, ap=True)
  File "C:\Program Files\Python310\lib\site-packages\pylink\jlink.py", line 204, in wrapper
    return func(self, *args, **kwargs)
  File "C:\Program Files\Python310\lib\site-packages\pylink\jlink.py", line 3384, in coresight_read
    raise errors.JLinkException(res)
pylink.errors.JLinkException: Unspecified error.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Program Files\Python310\lib\site-packages\pyocd\coresight\discovery.py", line 128, in _find_aps
    isValid = AccessPort.probe(self.dp, apsel)
  File "C:\Program Files\Python310\lib\site-packages\pyocd\coresight\ap.py", line 318, in probe
    idr = dp.read_ap((ap_num << APSEL_SHIFT) | AP_IDR)
  File "C:\Program Files\Python310\lib\site-packages\pyocd\coresight\dap.py", line 838, in read_ap
    result_cb = self.probe.read_ap(addr, now=False)
  File "C:\Program Files\Python310\lib\site-packages\pyocd\probe\jlink_probe.py", line 313, in read_ap
    raise self._convert_exception(exc) from exc
pyocd.core.exceptions.TransferFaultError: Memory transfer fault (Unspecified error.)
0001183 I AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2) [ap]
0001187 I AHB-AP#1 IDR = 0x74770001 (AHB-AP var0 rev7) [ap]
0001196 I AHB-AP#0 Class 0x1 ROM table #0 @ 0x41003000 (designer=01f part=cd0) [rom_table]
0001202 I [0]<e00ff000:ROM class=1 designer=43b part=4c4> [rom_table]
0001202 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c4) [rom_table]
0001209 I   [0]<e000e000:SCS v7-M class=14 designer=43b part=00c> [rom_table]
0001212 I   [1]<e0001000:DWT v7-M class=14 designer=43b part=002> [rom_table]
0001216 I   [2]<e0002000:FPB v7-M class=14 designer=43b part=003> [rom_table]
0001220 I   [3]<e0000000:ITM v7-M class=14 designer=43b part=001> [rom_table]
0001228 I   [4]<e0040000:TPIU M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=ca1:0:0> [rom_table]
0001233 I   [5]<e0041000:ETM M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0> [rom_table]
0001238 I   [6]<e0042000:ETB class=9 designer=43b part=907 devtype=21 archid=0000 devid=0:0:0> [rom_table]
0001243 I CPU core #0 is Cortex-M4 r0p1 [cortex_m]
0001246 I FPU present: FPv4-SP-D16-M [cortex_m]
0001248 I 4 hardware watchpoints [dwt]
0001251 I 6 hardware breakpoints, 4 literal comparators [fpb]
0001266 I Semihost server started on port 4444 (core 0) [server]
0001410 I GDB server started on port 3333 (core 0) [gdbserver]
```
The OS is Windows 10, while pyOCD version is at `0.33.1`

```
# pyocd.yaml
probes:
  xxxxxxxxx:
    target_override: ATSAMD51N20A

pack:
  Microchip.SAMD51_DFP.3.6.120.pack
```
I have examined the packs's .pdsc file, and there is a default RAM defined.
`<memory id="IRAM1" start="0x20000000" size="0x40000" default="1"/>`

elfmimi, 2022-04-11:
RAM regions for ATSAMD51N20A are defined like this.
```
            <memory id="IRAM1" start="0x20000000" size="0x40000" default="1"/>
            <memory id="IRAM2" start="0x20000000" size="0x8000"/>
            <memory id="IRAM3" start="0x20000000" size="0x8000"/>
            <memory id="IRAM4" start="0x47000000" size="0x2000"/>
```
The problem is IRAM1 , IRAM2 and IRAM3 are overlapping and only IRAM3 is being listed by [_extract_memories()](https://github.com/pyocd/pyOCD/blob/88222188de5ee553ada083efbdd54eb8f15974b0/pyocd/target/pack/cmsis_pack.py#L245-L307)

nedimlisica, 2022-04-11:
If I delete IRAM2 & IRAM3 entries (does that make any sense?), 'no default RAM defined' is gone, but I'm left with the exception above. 

Here is the full output, this time from MacOS since I won't have access to the Win PC until morning.
```
> pyocd gdb

0000706 I Using config options for probe xxxxxxxxx [session]
0000742 W Overlapping memory regions in file Microchip.SAMD51_DFP.3.6.120.pack (ATSAMD51G18A); deleting outer region. Further warnings will be suppressed for this file. [cmsis_pack]
0000746 I Target type is atsamd51n20a [board]
0000959 C Memory transfer fault (Unspecified error.) [__main__]
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.9/site-packages/pyocd/probe/jlink_probe.py", line 293, in read_dp
    value = self._link.coresight_read(addr // 4, ap=False)
  File "/opt/homebrew/lib/python3.9/site-packages/pylink/jlink.py", line 204, in wrapper
    return func(self, *args, **kwargs)
  File "/opt/homebrew/lib/python3.9/site-packages/pylink/jlink.py", line 3384, in coresight_read
    raise errors.JLinkException(res)
pylink.errors.JLinkException: Unspecified error.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.9/site-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
  File "/opt/homebrew/lib/python3.9/site-packages/pyocd/subcommands/gdbserver_cmd.py", line 190, in invoke
    with session:
  File "/opt/homebrew/lib/python3.9/site-packages/pyocd/core/session.py", line 386, in __enter__
    self.open()
  File "/opt/homebrew/lib/python3.9/site-packages/pyocd/core/session.py", line 523, in open
    self._board.init()
  File "/opt/homebrew/lib/python3.9/site-packages/pyocd/board/board.py", line 124, in init
    self.target.init()
  File "/opt/homebrew/lib/python3.9/site-packages/pyocd/core/soc_target.py", line 147, in init
    seq.invoke()
  File "/opt/homebrew/lib/python3.9/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/opt/homebrew/lib/python3.9/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/opt/homebrew/lib/python3.9/site-packages/pyocd/coresight/dap.py", line 406, in _connect
    connector.connect(self._protocol)
  File "/opt/homebrew/lib/python3.9/site-packages/pyocd/coresight/dap.py", line 180, in connect
    self._connect_dp(protocol)
  File "/opt/homebrew/lib/python3.9/site-packages/pyocd/coresight/dap.py", line 221, in _connect_dp
    self._idr = self.read_idr()
  File "/opt/homebrew/lib/python3.9/site-packages/pyocd/coresight/dap.py", line 248, in read_idr
    dpidr = self._probe.read_dp(DP_IDR, now=True)
  File "/opt/homebrew/lib/python3.9/site-packages/pyocd/probe/jlink_probe.py", line 295, in read_dp
    raise self._convert_exception(exc) from exc
pyocd.core.exceptions.TransferFaultError: Memory transfer fault (Unspecified error.)
```

elfmimi, 2022-04-11:
This time it looks it's trying to read DP_IDR and failing.
In the previous log there was `0001163 I DP IDR = 0x2ba01477 (v1 rev2) [dap]` , but not this time.
So probably it is wiring issue?

nedimlisica, 2022-04-12:
Now I tried it on Windows PC, and so far so good! Basic gdb stuff works, though there still is an exception.

```
> pyocd gdb

0001846 I Using config options for probe 601007506 [session]
0001862 W Overlapping memory regions in file Microchip.SAMD51_DFP.3.6.120.pack (ATSAMD51G18A); deleting outer region. Further warnings will be suppressed for this file. [cmsis_pack]
0001870 I Target type is atsamd51n20a [board]
0002022 I DP IDR = 0x2ba01477 (v1 rev2) [dap]
0002026 E Exception while probing AP#2: Memory transfer fault (Unspecified error.) [discovery]
Traceback (most recent call last):
  File "C:\Program Files\Python310\lib\site-packages\pyocd\probe\jlink_probe.py", line 311, in read_ap
    value = self._link.coresight_read((addr & self.A32) // 4, ap=True)
  File "C:\Program Files\Python310\lib\site-packages\pylink\jlink.py", line 204, in wrapper
    return func(self, *args, **kwargs)
  File "C:\Program Files\Python310\lib\site-packages\pylink\jlink.py", line 3384, in coresight_read
    raise errors.JLinkException(res)
pylink.errors.JLinkException: Unspecified error.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Program Files\Python310\lib\site-packages\pyocd\coresight\discovery.py", line 128, in _find_aps
    isValid = AccessPort.probe(self.dp, apsel)
  File "C:\Program Files\Python310\lib\site-packages\pyocd\coresight\ap.py", line 318, in probe
    idr = dp.read_ap((ap_num << APSEL_SHIFT) | AP_IDR)
  File "C:\Program Files\Python310\lib\site-packages\pyocd\coresight\dap.py", line 838, in read_ap
    result_cb = self.probe.read_ap(addr, now=False)
  File "C:\Program Files\Python310\lib\site-packages\pyocd\probe\jlink_probe.py", line 313, in read_ap
    raise self._convert_exception(exc) from exc
pyocd.core.exceptions.TransferFaultError: Memory transfer fault (Unspecified error.)
0002063 I AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2) [ap]
0002068 I AHB-AP#1 IDR = 0x74770001 (AHB-AP var0 rev7) [ap]
0002078 I AHB-AP#0 Class 0x1 ROM table #0 @ 0x41003000 (designer=01f part=cd0) [rom_table]
0002086 I [0]<e00ff000:ROM class=1 designer=43b part=4c4> [rom_table]
0002086 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c4) [rom_table]
0002093 I   [0]<e000e000:SCS v7-M class=14 designer=43b part=00c> [rom_table]
0002096 I   [1]<e0001000:DWT v7-M class=14 designer=43b part=002> [rom_table]
0002101 I   [2]<e0002000:FPB v7-M class=14 designer=43b part=003> [rom_table]
0002105 I   [3]<e0000000:ITM v7-M class=14 designer=43b part=001> [rom_table]
0002109 I   [4]<e0040000:TPIU M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=ca1:0:0> [rom_table]
0002115 I   [5]<e0041000:ETM M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0> [rom_table]
0002120 I   [6]<e0042000:ETB class=9 designer=43b part=907 devtype=21 archid=0000 devid=0:0:0> [rom_table]
0002121 I CPU core #0 is Cortex-M4 r0p1 [cortex_m]
0002125 I FPU present: FPv4-SP-D16-M [cortex_m]
0002127 I 4 hardware watchpoints [dwt]
0002130 I 6 hardware breakpoints, 4 literal comparators [fpb]
0002149 I Semihost server started on port 4444 (core 0) [server]
0002251 I GDB server started on port 3333 (core 0) [gdbserver]
0093928 I Client connected to port 3333! [gdbserver]
0481122 I Client detached [gdbserver]
0481122 I Client disconnected from port 3333! [gdbserver]
0481563 I Semihost server stopped [server]
```

I suppose AP2 doesn't exist in this MCU?
Regarding the previous failure, I guess it is wiring related, and I will try again to confirm.