pyocd/pyOCD #1364: Problems connecting to an HC32l110x6 chip

jeffreyabecker, 2022-04-12:
I'm attempting to connect to a hc32l110 chip using a picoprobe and seeing an error message `Protocol fault [__main__]` regardless of how I try. Some reading of the google translated data sheet leads me to believe this chip may not require SWJ switching which seems to be whats failing. is there any way to disable this step? Has anyone had success connecting to this chip with a specific debug probe other than a picoprobe?
the command I'm invoking is `pyocd gdbserver -t hc32l110 -L *=debug`
and the output produced is:
```
0002070 D Project directory: C:\ode [session]
0004746 D Opened PEMicro library: ...\pypemicro\libs\Windows\unitacmp-64.dll, b'Interface Libraries Version 8.12.00.00 (http://www.pemicro.com)'. [pemicro]
0004747 D Opened PEMicro library: unitacmp-64.dll, b'Interface Libraries Version 8.12.00.00 (http://www.pemicro.com)'. [pemicro]
0004829 D Not sending notification because no subscribers: event=target_override [notification]
0004830 D Project directory: C:\ode [session]
0004834 I Target type is hc32l110 [board]
0004837 D Running task load_svd [sequencer]
0004837 D Running task pre_connect [sequencer]
0004838 D Running task dp_init [sequencer]
0004838 D Running task lock_probe [sequencer]
0004841 D Running task get_probe_capabilities [sequencer]
0004843 D Running task connect [sequencer]
0004845 D Default wire protocol selected; using SWD [dap]
0004847 D Sending deprecated SWJ sequence to select SWD [swj]
0004867 D DP IDCODE read failed; resending SWJ sequence (use dormant=False) [dap]
0004868 D Sending deprecated SWJ sequence to select SWD [swj]
0004869 D DP IDCODE read failed; resending SWJ sequence (use dormant=False) [dap]
0004870 D Sending SWJ sequence to select SWD; using dormant state [swj]
0004872 D DP IDCODE read failed; resending SWJ sequence (use dormant=False) [dap]
0004873 D Sending SWJ sequence to select SWD; using dormant state [swj]
0004875 D DP IDCODE read failed; resending SWJ sequence (use dormant=False) [dap]
0004875 D uninit session <pyocd.core.session.Session object at 0x0000017938C18790> [session]
0004876 C Picoprobe: Protocol fault [__main__]
Traceback (most recent call last):
  File "...\pyocd\__main__.py", line 161, in run
    status = cmd.invoke()
  File "...\pyocd\subcommands\gdbserver_cmd.py", line 190, in invoke
    with session:
  File "...\pyocd\core\session.py", line 386, in __enter__
    self.open()
  File "...\pyocd\core\session.py", line 523, in open
    self._board.init()
  File "...\pyocd\board\board.py", line 124, in init
    self.target.init()
  File "...\pyocd\core\soc_target.py", line 147, in init
    seq.invoke()
  File "...\pyocd\utility\sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "...\pyocd\utility\sequencer.py", line 208, in invoke
    resultSequence = call()
  File "...\pyocd\coresight\dap.py", line 406, in _connect
    connector.connect(self._protocol)
  File "...\pyocd\coresight\dap.py", line 180, in connect
    self._connect_dp(protocol)
  File "...\pyocd\coresight\dap.py", line 221, in _connect_dp
    self._idr = self.read_idr()
  File "...\pyocd\coresight\dap.py", line 248, in read_idr
    dpidr = self._probe.read_dp(DP_IDR, now=True)
  File "...\pyocd\probe\picoprobe.py", line 490, in read_dp
    val = self._read_reg(addr, self.DP)
  File "...\pyocd\probe\picoprobe.py", line 605, in _read_reg
    self._read_check_swd_ack()
  File "...\pyocd\probe\picoprobe.py", line 652, in _read_check_swd_ack
    self._check_swd_acks(ack)
  File "...\pyocd\probe\picoprobe.py", line 670, in _check_swd_acks
    raise e
  File "...\pyocd\coresight\dap.py", line 221, in _connect_dp
    self._idr = self.read_idr()
  File "...\pyocd\coresight\dap.py", line 248, in read_idr
    dpidr = self._probe.read_dp(DP_IDR, now=True)
  File "...\pyocd\probe\picoprobe.py", line 490, in read_dp
    val = self._read_reg(addr, self.DP)
  File "...\pyocd\probe\picoprobe.py", line 605, in _read_reg
    self._read_check_swd_ack()
  File "...\pyocd\probe\picoprobe.py", line 652, in _read_check_swd_ack
    self._check_swd_acks(ack)
  File "...\pyocd\probe\picoprobe.py", line 670, in _check_swd_acks
    raise e
  File "...\pyocd\coresight\dap.py", line 221, in _connect_dp
    self._idr = self.read_idr()
  File "...\pyocd\coresight\dap.py", line 248, in read_idr
    dpidr = self._probe.read_dp(DP_IDR, now=True)
  File "...\pyocd\probe\picoprobe.py", line 490, in read_dp
    val = self._read_reg(addr, self.DP)
  File "...\pyocd\probe\picoprobe.py", line 605, in _read_reg
    self._read_check_swd_ack()
  File "...\pyocd\probe\picoprobe.py", line 652, in _read_check_swd_ack
    self._check_swd_acks(ack)
  File "...\pyocd\probe\picoprobe.py", line 670, in _check_swd_acks
    raise e
  File "...\pyocd\coresight\dap.py", line 221, in _connect_dp
    self._idr = self.read_idr()
  File "...\pyocd\coresight\dap.py", line 248, in read_idr
    dpidr = self._probe.read_dp(DP_IDR, now=True)
  File "...\pyocd\probe\picoprobe.py", line 490, in read_dp
    val = self._read_reg(addr, self.DP)
  File "...\pyocd\probe\picoprobe.py", line 605, in _read_reg
    self._read_check_swd_ack()
  File "...\pyocd\probe\picoprobe.py", line 652, in _read_check_swd_ack
    self._check_swd_acks(ack)
  File "...\pyocd\probe\picoprobe.py", line 670, in _check_swd_acks
    raise e
pyocd.core.exceptions.TransferError: Picoprobe: Protocol fault
```

flit, 2022-04-12:
You are probably the first person to try a picoprobe + HDSC MCU. 😄  Unfortunately I don't have any HDSC EVKs so I can't try to reproduce. The HDSC support in pyOCD has been added directly by that company, so presumably it was successfully tested. Not sure what probe they used.

Afaik, all the HDSC devices have CM0+ cores. The CM0+ doesn't support JTAG, so the SWJ sequences aren't actually necessary. But they do not hurt, either. Still, you can disable by passing `-Odap_swj_enable=0` (or setting `dap_swj_enable` in a config file).

Suggestions:
- Double and triple check the SWD signal connection. Make sure you have GND solidly connected between boards.
- Try a CMSIS-DAP based probe that runs on the Pico instead of picoprobe. It will be faster, anyway! See [Dapper Mime](https://github.com/majbthrd/DapperMime).
- Make sure the target MCU is not in sleep or some state where it won't response to SWD (like having SWD pins muxed to GPIO or some other peripheral).

cc @lennvn

jeffreyabecker, 2022-04-12:
Thanks! I've got a pitaya link probe (please tell me this isnt trash) in the mail and will double check the physical / code items you've suggested. Currently, I've tried adding `-Odap_swj_enable=0`, sadly this just makes it fail faster.

jeffreyabecker, 2022-04-12:
So I've double checked my pin configs, connections, and loaded up a copy of Dapper Mime which is producing much more detailed errors. I can post a full log if you like but the crux of it seems to be `error(Unexpected ACK value (5) returned by probe)`.

jeffreyabecker, 2022-04-12:
Sadly I dont. I do have a few more things I'm going to try:
1) holding the reset pin low -- there seems to be something about doing this in their flash programming utility docs
2) enabling all peripherals -- Being a low-power chip there's a peripheral enable register and one diagram indicates that SWD is controlled. However there's no docs  on which bit controls it.

jeffreyabecker, 2022-04-12:
Sadly, it looks like I'm going to have to wait for the pitaya to show up and free up the pico for use with pulseview.

jeffreyabecker, 2022-04-13:
Not sure if this helps any but their DDL example code has a bunch of jlink ini files in it which all look like this:
```
[BREAKPOINTS]
ForceImpTypeAny = 0
ShowInfoWin = 1
EnableFlashBP = 2
BPDuringExecution = 0
[CFI]
CFISize = 0x00
CFIAddr = 0x00
[CPU]
MonModeVTableAddr = 0xFFFFFFFF
MonModeDebug = 0
MaxNumAPs = 0
LowPowerHandlingMode = 0
OverrideMemMap = 0
AllowSimulation = 1
ScriptFile=""
[FLASH]
CacheExcludeSize = 0x00
CacheExcludeAddr = 0x00
MinNumBytesFlashDL = 0
SkipProgOnCRCMatch = 1
VerifyDownload = 1
AllowCaching = 1
EnableFlashDL = 2
Override = 1
Device="Cortex-M0+"
[GENERAL]
WorkRAMSize = 0x00
WorkRAMAddr = 0x00
RAMUsageLimit = 0x00
[SWO]
SWOLogFile=""
[MEM]
RdOverrideOrMask = 0x00
RdOverrideAndMask = 0xFFFFFFFF
RdOverrideAddr = 0xFFFFFFFF
WrOverrideOrMask = 0x00
WrOverrideAndMask = 0xFFFFFFFF
WrOverrideAddr = 0xFFFFFFFF
```

flit, 2022-04-13:
Sorry, I know frustrating this can be!

flit, 2022-04-13:
I'm afraid the JLink INI file doesn't have anything unexpected. It looks like all the default values.

jeffreyabecker, 2022-04-13:
I mean at some level, what should I expect from a $0.40 MCU? I've solved my underlying interrupt issues. I've got a couple more things to try on the debugger front this evening:
1) breaking out a pristine device to see if somehow the way I've been flashing them (UART based utility from HDSC) is breaking stuff
2) hdsc make an unlock/decrypt/reset tool which (via google translate) proports to unlock the swd port 
3) trying to get in touch with a couple people who've had success connecting via openocd to see if they had any specific tools or experiences

flit, 2022-04-13:
If you do make progress, let's be sure to add some notes to the [target family usage notes](https://pyocd.io/docs/target_family_notes.html) docs.

jeffreyabecker, 2022-04-16:
So, the pitaya link works. I'm currently working on testing why but my main operating theory is that the pico based hardware is operating at 5V, rather than 3.3V for the pitaya, and thats affecting things.

jeffreyabecker, 2022-04-25:
Strike that -- the Pitaya link was able to program the chips but debugging is still non-functional. I'm going to redo my board design to include some components the docs say are optional and see if that helps.

closed 2022-04-25 by jeffreyabecker

buttim, 2023-07-29:
[jeffreyabecker](https://github.com/jeffreyabecker), can you give us some link to the tool to unlock the SWD port you mentioned?