pyocd/pyOCD #1368: STM32H730: error parsing coresight table

diggit, 2022-04-16:
When connecting to STM32H730VB, there is problem with coresight rom table readout.
In fact, even records which are found do not seem to match [ref. manual](https://www.st.com/resource/en/reference_manual/rm0468-stm32h723733-stm32h725735-and-stm32h730-value-line-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) (65.5.1 page 3140).
Debugging and breakpoints work.
As TPIU is not found, SVW does not work and requesting to do so, crashes pyOCD (last log).

Verbose logs follow. 

pyOCD version: 0.33.1

stlink v2-1
```
pyocd gdbserver -vvvvv --port 50000 --telnet-port 50001 --target stm32h730vbtx  -f=1000000                                                                                                          
0000822 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=04f2 PID=b2da). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000823 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=2516 PID=0048). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000953 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=145f PID=02b3). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000955 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=04f2 PID=b2da). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0000956 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=2516 PID=0048). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0000961 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=145f PID=02b3). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0000962 D Project directory: /home/diggit [session]
0001025 D STLink probe 48FF74065077515756461187 firmware version: V2J39M7 [stlink]
0001045 D Project directory: /home/diggit [session]
0001370 I Target type is stm32h730vbtx [board]
0001375 D STLink probe 48FF74065077515756461187 firmware version: V2J39M7 [stlink]
0001376 D Running task load_svd [sequencer]
0001420 D Running task pre_connect [sequencer]
0001421 D Running task dp_init [sequencer]
0001421 D Running task lock_probe [sequencer]
0001421 D Running task get_probe_capabilities [sequencer]
0001421 D Running task connect [sequencer]
0001423 D Default wire protocol selected; using SWD [dap]
0001423 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0001424 D Running task clear_sticky_err [sequencer]
0001424 D Running task power_up_debug [sequencer]
0001425 D Running task check_version [sequencer]
0001425 D Running task unlock_probe [sequencer]
0001425 D Running task create_discoverer [sequencer]
0001425 D Running task discovery [sequencer]
0001425 D Running task find_aps [sequencer]
0001428 D Running task create_aps [sequencer]
0001428 D Running task create_ap.0 [sequencer]
0001606 D Using accelerated memory access interface [ap]
0001612 I AHB-AP#0 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
0001639 D AHB-AP#0 default HPROT=b HNONSEC=0 [ap]
0001666 D AHB-AP#0 implemented HPROT=f HNONSEC=1 [ap]
0001693 D Running task create_ap.1 [sequencer]
0001721 D Using accelerated memory access interface [ap]
0001726 I AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
0001753 D AHB-AP#1 default HPROT=3 HNONSEC=1 [ap]
0001780 D AHB-AP#1 implemented HPROT=f HNONSEC=1 [ap]
0001807 D Running task create_ap.2 [sequencer]
0001834 D Using accelerated memory access interface [ap]
0001839 I APB-AP#2 IDR = 0x54770002 (APB-AP var0 rev5) [ap]
0001879 D APB-AP#2 default HPROT=0 HNONSEC=0 [ap]
0001881 D APB-AP#2 implemented HPROT=0 HNONSEC=0 [ap]
0001883 D Running task find_components [sequencer]
0001883 D Running task init_ap.0 [sequencer]
0001888 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fe000 (designer=020 part=483) [rom_table]
0001891 I [0]<e00ff000:ROM class=1 designer=43b part=4c7> [rom_table]
0001891 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c7) [rom_table]
0001894 I   [0]<e000e000:SCS v7-M class=14 designer=43b part=00c> [rom_table]
0001897 I   [1]<e0001000:DWT v7-M class=14 designer=43b part=002> [rom_table]
0001899 I   [2]<e0002000:FPB v7-M class=14 designer=43b part=00e> [rom_table]
0001900 I   [3]<e0000000:ITM v7-M class=14 designer=43b part=001> [rom_table]
0001900 D   [4]<fff41002 not present> [rom_table]
0001900 D   [5]<fff42002 not present> [rom_table]
0001903 I [1]<e0041000:ETM M7 class=9 designer=43b part=975 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0001907 I [2]<e0043000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=40800:0:0> [rom_table]
0001907 D [3]<1ff02002 not present> [rom_table]
0001907 D Running task init_ap.2 [sequencer]
0001910 I APB-AP#2 Class 0x1 ROM table #0 @ 0xe00e0000 (designer=020 part=483) [rom_table]
[... 157 lines cut ...]
0002905 D halting core 0 [cortex_m]
0002906 D Running task post_connect [sequencer]
0002906 D Running task post_connect_hook [sequencer]
0002906 D Running task create_flash [sequencer]
0002906 D Running task notify [sequencer]
0002907 D Setting vector catch to 0x00000001 [cortex_m]
0002911 I Semihost server started on port 50001 (core 0) [server]
0002990 I GDB server started on port 50000 (core 0) [gdbserver]
```

If I try to enable SVW (`-O enable_swv=true -O swv_system_clock=500000000 -O swv_clock=1000000 `), pyOCD terminates with exception:
```
0003152 C uncaught exception: 'NoneType' object has no attribute 'init' [__main__]
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
  File "/usr/lib/python3.10/site-packages/pyocd/subcommands/gdbserver_cmd.py", line 221, in invoke
    gdb = GDBServer(session, core=core_number)
  File "/usr/lib/python3.10/site-packages/pyocd/gdbserver/gdbserver.py", line 216, in __init__
    self._swv_reader.init(sys_clock, swo_clock, console_file)
  File "/usr/lib/python3.10/site-packages/pyocd/trace/swv.py", line 116, in init
    tpiu.init()
AttributeError: 'NoneType' object has no attribute 'init'
```
Which is not surprising when TPIU was not found.

flit, 2022-04-28:
Hi @diggit. This is caused by the `DGBMCU.CR` register not being configured to enable certain clocks that the H7xx devices require for fully enabling debug components.

Can you please try out the branch where I'm implementing support for CMSIS-Pack [debug sequences](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/debug_description.html#pdsc_SequenceNameEnum_pg)? You can install it directly with:

```
pip install git+https://github.com/flit/pyOCD.git@feature/debug_sequences
```

In my testing, this resolves the discovery issues completely. (The STM32 packs have sequences for configuring `DBGMCU` registers upon connect.) The branch will hopefully be merged pretty soon, there are just some loose ends to tie up and more testing.

diggit, 2022-06-25:
I've finally tested with branch you linked and still no success.

```
# pyocd -V
0.28.1.dev667+dirty
```

stlink v2-1
```
# pyocd gdbserver -vvvvv --port 50000 --telnet-port 50001 --target stm32h730vbtx -O enable_swv=true -O swv_system_clock=500000000 -O swv_clock=1000000   -f=100000
0000865 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=04f2 PID=b2da). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000865 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=2516 PID=0048). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0000865 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=145f PID=02b3). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_backend]
0001058 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=04f2 PID=b2da). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0001059 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=2516 PID=0048). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0001060 D [Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=145f PID=02b3). This can probably be remedied with a udev rule. See <https://github.com/pyocd/pyOCD/tree/master/udev> for help. [pyusb_v2_backend]
0001063 D Project directory: /home/diggit [session]
0001533 D STLink probe 48FF74065077515756461187 firmware version: V2J39M7 [stlink]
0001545 D Project directory: /home/diggit [session]
0001902 D flash algo: [code=0x21c] [b1=0x220,0x620] [b2=0x620,0xa20] [stack=0x2a20; 0x2000 b] (ram=0x20000000, 0x8000 b) [flash_algo]
0001980 I Target type is stm32h730vbtx [board]
0001985 D STLink probe 48FF74065077515756461187 firmware version: V2J39M7 [stlink]
0001987 D Running task load_svd [sequencer]
0002074 D Running task pre_connect [sequencer]
0002074 D Running task dp_init [sequencer]
0002074 D Running task lock_probe [sequencer]
0002074 D Running task get_probe_capabilities [sequencer]
0002074 D Running task connect [sequencer]
0002220 D Default wire protocol selected; using SWD [dap]
0002231 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0002232 D Running task clear_sticky_err [sequencer]
0002248 D Running task power_up_debug [sequencer]
0002270 D Running task check_version [sequencer]
0002270 D Running task unlock_probe [sequencer]
0002270 D Running task unlock_device [sequencer]
0002270 D Running debug sequence 'DebugDeviceUnlock' [pack_target]
0002274 D Running debug sub-sequence 'CheckID' [functions]
0002308 D Using accelerated memory access interface for APB-AP#2 [ap]
0002353 D APB-AP#2 default HPROT=0 HNONSEC=0 [ap]
0002380 D APB-AP#2 implemented HPROT=0 HNONSEC=0 [ap]
0002514 D Running task create_discoverer [sequencer]
0002514 D Running task discovery [sequencer]
0002514 D Running task find_aps [sequencer]
0002526 D Running task create_aps [sequencer]
0002526 D Running task create_ap.0 [sequencer]
0002530 D Using accelerated memory access interface for AHB-AP#0 [ap]
0002535 D AHB-AP#0 default HPROT=b HNONSEC=0 [ap]
0002538 D AHB-AP#0 implemented HPROT=f HNONSEC=1 [ap]
0002541 I AHB-AP#0 IDR = 0x84770001 (AHB-AP var0 rev8) [discovery]
0002541 D Running task create_ap.1 [sequencer]
0002545 D Using accelerated memory access interface for AHB-AP#1 [ap]
0002550 D AHB-AP#1 default HPROT=3 HNONSEC=1 [ap]
0002553 D AHB-AP#1 implemented HPROT=f HNONSEC=1 [ap]
0002556 I AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8) [discovery]
0002556 D Running task create_ap.2 [sequencer]
0002558 D Using accelerated memory access interface for APB-AP#2 [ap]
0002566 D APB-AP#2 default HPROT=0 HNONSEC=0 [ap]
0002569 D APB-AP#2 implemented HPROT=0 HNONSEC=0 [ap]
0002572 I APB-AP#2 IDR = 0x54770002 (APB-AP var0 rev5) [discovery]
0002572 D Running task find_components [sequencer]
0002572 D Running task init_ap.0 [sequencer]
0002591 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fe000 (designer=020:ST part=483) [rom_table]
0002607 I [0]<e00ff000:ROM class=1 designer=43b:Arm part=4c7> [rom_table]
0002607 I   AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b:Arm part=4c7) [rom_table]
0002623 I   [0]<e000e000:SCS v7-M class=14 designer=43b:Arm part=00c> [rom_table]
0002633 I   [1]<e0001000:DWT v7-M class=14 designer=43b:Arm part=002> [rom_table]
0002641 I   [2]<e0002000:FPB v7-M class=14 designer=43b:Arm part=00e> [rom_table]
0002650 I   [3]<e0000000:ITM v7-M class=14 designer=43b:Arm part=001> [rom_table]
0002650 D   [4]<fff41002 not present> [rom_table]
0002651 D   [5]<fff42002 not present> [rom_table]
[... 39 lines cut ...]
0002891 D Running task post_connect_hook [sequencer]
0002891 D Running task create_flash [sequencer]
0002891 D Running task notify [sequencer]
0002894 D Setting vector catch to 0x00000001 [cortex_m]
0002904 I Semihost server started on port 50001 (core 0) [server]
0002905 D Running debug sequence 'TraceStart' [pack_target]
0002906 D Running debug sub-sequence 'EnableTraceSWO' [functions]
[... cut ...]
  File "/usr/lib/python3.10/site-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
  File "/usr/lib/python3.10/site-packages/pyocd/subcommands/gdbserver_cmd.py", line 221, in invoke
    gdb = GDBServer(session, core=core_number)
  File "/usr/lib/python3.10/site-packages/pyocd/gdbserver/gdbserver.py", line 216, in __init__
    self._swv_reader.init(sys_clock, swo_clock, console_file)
  File "/usr/lib/python3.10/site-packages/pyocd/trace/swv.py", line 127, in init
    tpiu.init()
AttributeError: 'NoneType' object has no attribute 'init'
```

diggit, 2022-06-25:
TPIU is found, so that's fixed, but issue is also somewhere else.

flit, 2022-06-25:
Thanks for the update. It's probably because the TPIU is accessed through APB-AP#2, and is declared on that AP's ROM table (instead of the same ROM table as the CM7). That configuration likely won't work in pyocd right now. Sorry!

The AttributeError fix should be straightforward. (Actually, I thought I had already fixed an error if it didn't find the TPIU! 😖) 

I'll look into adding support for a system-level TPIU.

diggit, 2022-06-25:
> It's probably because the TPIU is accessed through APB-AP#2

yeah, only APB-AP#0: DWT, FPB, ITM are available in `session`

> That configuration likely won't work in pyocd right now. Sorry!

Does fixing this mean a lot of work?

> The AttributeError fix should be straightforward. (Actually, I thought I had already fixed an error if it didn't find the TPIU! confounded)

in `swv.py`
```python
        itm = self._session.target.get_first_child_of_type(ITM)
        if not itm:
            LOG.warning("SWV not initalized: Target does not have ITM component")
            return False
        itm = self._session.target.get_first_child_of_type(TPIU)
#       ^^ should be tpiu
        if not tpiu:
            LOG.warning("SWV not initalized: Target does not have TPIU component")
            return False
```

flit, 2022-06-25:
> Does fixing this mean a lot of work?

Not really sure! The main thing is to ensure the TPIU gets attached to the target node.

> ^^ should be tpiu

Uh, embarrassing! 🙃

flit, 2022-06-25:
>^^ should be tpiu

What branch/commit are you looking at? It's correct on all of `main`, `develop`, and `feature/debug_sequences`, and the v0.34 release (v0.33.1 was notably different).

diggit, 2022-06-25:
shoot, I tried to undo my changes and failed :facepalm: 

Should have been...
```python
        itm = self._session.target.get_first_child_of_type(ITM)
        if not itm:
            LOG.warning("SWV not initalized: Target does not have ITM component")
            return False
        tpiu = self._session.target.get_first_child_of_type(TPIU)
        if not itm:
#               ^^ should be tpiu
            LOG.warning("SWV not initalized: Target does not have TPIU component")
            return False
```

flit, 2022-06-25:
Ugh, and I totally missed the error on the next line below… 😅

diggit, 2022-06-25:
oh, first issue could be the fact, that stlink v2-1 does not offer access to any other AP than 0

flit, 2022-06-26:
Actually, the STLink V2J28 and above, including all v3, does allow access of any AP (except the version 2 APs used in ADIv6, required for Cortex-M55 and M85). That's how pyocd can read the ROM table from AP#2.