pyocd/pyOCD #1373: Run directly under VSCode aka run source code directly from command line?

buzmeg, 2022-04-21:
As things stand right now, it seems like I have to "pip install", try out pyocd, make changes to code, "pip install", try out pyocd, etc.

Is there a way to run the edited pyocd code directly from the command line without the "pip install" step?

The purpose of this is to allow me to run this under VSCode directly so things like debugging and stepping actually work.  If there is a better way to run this from VSCode than trying to run from the command line, I'm happy to do that instead.

Thanks.

buzmeg, 2022-04-23:
Strange.  I get quite different behavior when I execute when pyocd is installed vs isn't:

```
(.venv) :~/python/pyOCD$  python -mpyocd list
  #   Probe               Unique ID  
-------------------------------------
  0   Segger J-Link EDU   <serial>  

(.venv) :~/python/pyOCD$ pip uninstall pyocd
Found existing installation: pyocd 0.33.2.dev2+dirty
Uninstalling pyocd-0.33.2.dev2+dirty:
  Would remove:
    /home/andrewl/python/pyOCD/.venv/bin/pyocd
    /home/andrewl/python/pyOCD/.venv/bin/pyocd-gdbserver
    /home/andrewl/python/pyOCD/.venv/lib/python3.8/site-packages/pyocd-0.33.2.dev2+dirty.dist-info/*
    /home/andrewl/python/pyOCD/.venv/lib/python3.8/site-packages/pyocd/*
Proceed (y/n)? y
  Successfully uninstalled pyocd-0.33.2.dev2+dirty
(.venv) :~/python/pyOCD$ python -mpyocd list
No available debug probes are connected
```

elfmimi, 2022-04-23:
Hmm, please try adding these lines to the end of pyocd/probe/aggregator.py .
```python
from ..probe.jlink_probe import JLinkProbe
PROBE_CLASSES["jlink"] = JLinkProbe
```

elfmimi, 2022-04-23:
Or instead use this as the top level script. like: `python main.py list`
[main.py](https://github.com/pyocd/pyOCD/files/8546725/main.py.txt)
```python
from pyocd.probe.aggregator import PROBE_CLASSES
from pyocd.probe.cmsis_dap_probe import CMSISDAPProbe
from pyocd.probe.jlink_probe import JLinkProbe
PROBE_CLASSES["cmsisdap"] = CMSISDAPProbe
PROBE_CLASSES["jlink"] = JLinkProbe

from pyocd.__main__ import main
main()
```

elfmimi, 2022-04-25:
I admit I personally prefer above wacky method. but it looks like sane people's choice is different.
Which is discussed in https://github.com/pyocd/pyOCD/issues/938 .
That is `pip install -e .`