pyocd/pyOCD #1396: Pyocd flash error for GD32F350G8 - Memory transfer fault @ 0x08000000-0x0800001f

Thomas-Lawrence, 2022-05-26:
# DFP info

          Vendor       Pack            Version
        ----------------------------------------
          GigaDevice   GD32F3x0_DFP    2.0.0

# Command
        pyocd flash -f 1.2m "C:\Users\Documents\Bin\binary.hex" -t GD32F350G8 -W -v

#  ERROR  
        Probe                        Unique ID
        -------------------------------------------------------------------------------------
          0   ARM CMSIS-DAP v1             0000000300190051044ab7bd00000c120000000097969921
          1   Segger 3rd party CMSIS-DAP   1045103829
        
        Enter the number of the debug probe or 'q' to quit> 0
        0004384 I Target type is gd32f350g8 [board]
        Exception in thread load-svd:
        Traceback (most recent call last):
          File "c:\python37\lib\threading.py", line 926, in _bootstrap_inner
            self.run()
          File "c:\python37\lib\site-packages\pyocd\debug\svd\loader.py", line 71, in run
            self._svd_location.load()
          File "c:\python37\lib\site-packages\pyocd\debug\svd\loader.py", line 47, in load
            self.device = SVDParser.for_xml_file(self.filename).get_device()
          File "c:\python37\lib\site-packages\pyocd\debug\svd\parser.py", line 74, in for_xml_file
            return cls(ET.parse(path), remove_reserved)
          File "c:\python37\lib\xml\etree\ElementTree.py", line 1197, in parse
            tree.parse(source, parser)
          File "c:\python37\lib\xml\etree\ElementTree.py", line 598, in parse
            self._root = parser._parse_whole(source)
          File "<string>", line None
        xml.etree.ElementTree.ParseError: XML or text declaration not at start of entity: line 1, column 2
        
        0004416 I DP IDR = 0x2ba01477 (v1 rev2) [dap]
        0004431 I AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2) [ap]
        0004447 I AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=751 part=61e) [rom_table]
        0004451 I [0]<e000e000:SCS v7-M class=14 designer=43b part=00c> [rom_table]
        0004454 I [1]<e0001000:DWT v7-M class=14 designer=43b part=002> [rom_table]
        0004456 I [2]<e0002000:FPB v7-M class=14 designer=43b part=003> [rom_table]
        0004460 I [3]<e0000000:ITM v7-M class=14 designer=43b part=001> [rom_table]
        0004464 I [4]<e0040000:TPIU M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=ca0:0:0> [rom_table]
        0004468 I CPU core #0 is Cortex-M4 r0p1 [cortex_m]
        0004476 I FPU present: FPv4-SP-D16-M [cortex_m]
        0004480 I 4 hardware watchpoints [dwt]
        0004484 I 6 hardware breakpoints, 4 literal comparators [fpb]
        0004494 I Loading C:\Users\thomas.lawrence\Documents\Bin\VMGOII_v1_0_1.hex [load_cmd]
        0004747 C Memory transfer fault @ 0x08000000-0x0800001f [__main__]
        Traceback (most recent call last):
          File "c:\python37\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 463, in read_ap_repeat_callback
            values = result()
          File "c:\python37\lib\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 1011, in reg_read_repeat_cb
            res = transfer.get_result()
          File "c:\python37\lib\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 154, in get_result
            self.daplink.flush()
          File "c:\python37\lib\site-packages\pyocd\utility\concurrency.py", line 29, in _locking
            return func(self, *args, **kwargs)
          File "c:\python37\lib\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 790, in flush
            self._read_packet()
          File "c:\python37\lib\site-packages\pyocd\utility\concurrency.py", line 29, in _locking
            return func(self, *args, **kwargs)
          File "c:\python37\lib\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 1056, in _read_packet
            decoded_data = cmd.decode_data(raw_data)
          File "c:\python37\lib\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 475, in decode_data
            data = self._decode_transfer_data(data)
          File "c:\python37\lib\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 381, in _decode_transfer_data
            self._check_response(data[2])
          File "c:\python37\lib\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 359, in _check_response
            raise DAPAccessIntf.TransferFaultError()
        pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferFaultError
        
        The above exception was the direct cause of the following exception:
        
        Traceback (most recent call last):
          File "c:\python37\lib\site-packages\pyocd\__main__.py", line 161, in run
            status = cmd.invoke()
          File "c:\python37\lib\site-packages\pyocd\subcommands\load_cmd.py", line 120, in invoke
            file_format=self._args.format)
          File "c:\python37\lib\site-packages\pyocd\flash\file_programmer.py", line 170, in program
            self._loader.commit()
          File "c:\python37\lib\site-packages\pyocd\flash\loader.py", line 293, in commit
            keep_unwritten=self._keep_unwritten)
          File "c:\python37\lib\site-packages\pyocd\flash\builder.py", line 490, in program
            sector_erase_count, page_program_time = self._compute_sector_erase_pages_and_weight(fast_verify)
          File "c:\python37\lib\site-packages\pyocd\flash\builder.py", line 669, in _compute_sector_erase_pages_and_weight
            self._analyze_pages_with_partial_read()
          File "c:\python37\lib\site-packages\pyocd\flash\builder.py", line 609, in _analyze_pages_with_partial_read
            data = self.flash.target.read_memory_block8(page.addr, size)
          File "c:\python37\lib\site-packages\pyocd\core\soc_target.py", line 212, in read_memory_block8
            return self.selected_core_or_raise.read_memory_block8(addr, size)
          File "c:\python37\lib\site-packages\pyocd\coresight\cortex_m.py", line 476, in read_memory_block8
            data = self.ap.read_memory_block8(addr, size)
          File "c:\python37\lib\site-packages\pyocd\core\memory_interface.py", line 181, in read_memory_block8
            data32 = self.read_memory_block32(addr, size // 4)
          File "c:\python37\lib\site-packages\pyocd\utility\concurrency.py", line 29, in _locking
            return func(self, *args, **kwargs)
          File "c:\python37\lib\site-packages\pyocd\coresight\ap.py", line 1171, in _read_memory_block32
            resp += self._read_block32_page(addr, n//4)
          File "c:\python37\lib\site-packages\pyocd\coresight\ap.py", line 1129, in _read_block32_page
            resp = self.dp.read_ap_multiple(self.address.address + self._reg_offset + MEM_AP_DRW, size)
          File "c:\python37\lib\site-packages\pyocd\coresight\dap.py", line 934, in read_ap_multiple
            return read_ap_multiple_cb()
          File "c:\python37\lib\site-packages\pyocd\coresight\dap.py", line 924, in read_ap_multiple_cb
            return result_cb()
          File "c:\python37\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 470, in read_ap_repeat_callback
            raise self._convert_exception(exc) from exc
        pyocd.core.exceptions.TransferFaultError: Memory transfer fault @ 0x08000000-0x0800001f

Thomas-Lawrence, 2022-05-30:
Any suggestions to fix the issue?

flit, 2022-06-08:
(Apologies for taking a long time to reply! 🙇🏽)

Thanks for providing a detailed report.

The exceptions suggest that the XML from the CMSIS-Pack is malformed. I'll have to take a look and verify this—will get back to you on this.

Because the Pack wasn't successfully read, the memory map probably isn't complete and is missing the flash information. So pyocd might have tried to write to flash as if it were RAM, which clearly isn't going to end well… 😄 

If the Pack XML is really malformed, the best option is to extract a copy locally and fix the XML. Then you can use the expanded pack with pyocd's `--pack` argument. (And try to report the issue to GigaDevice.)

Thomas-Lawrence, 2022-06-09:
Hi Flit, 
I have also tried to add a built-in target for the same as per the documentation, but there also am getting the same memory transfer fault.
I have verified the memory mapping and the flash algorithm generated and seems to be correct.
When tried with the 'basic_test.py' script, it was able to run up to 'TEST RESET' without issues. see below log,

  Probe                        Unique ID
  -------------------------------------------------------------------------------------
    0   ARM CMSIS-DAP v1             0000000300190051044ab7bd00000c120000000097969921  
    1   Segger 3rd party CMSIS-DAP   1045103829

      Enter the number of the debug probe or 'q' to quit> 0
      INFO:pyocd.core.session:Using config options for probe 0000000300190051044ab7bd00000c120000000097969921
      INFO:pyocd.board.board:Target type is stm32f051
      INFO:pyocd.coresight.dap:DP IDR = 0x2ba01477 (v1 rev2)
      INFO:pyocd.coresight.ap:AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
      INFO:pyocd.coresight.rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=751 part=a15)
      INFO:pyocd.coresight.rom_table:[0]<e000e000:SCS v7-M class=14 designer=43b part=00c>
      INFO:pyocd.coresight.rom_table:[1]<e0001000:DWT v7-M class=14 designer=43b part=002>
      INFO:pyocd.coresight.rom_table:[2]<e0002000:FPB v7-M class=14 designer=43b part=003>
      INFO:pyocd.coresight.rom_table:[3]<e0000000:ITM v7-M class=14 designer=43b part=001>
      INFO:pyocd.coresight.rom_table:[4]<e0040000:TPIU M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=ca0:0:0>
      INFO:pyocd.coresight.cortex_m:CPU core #0 is Cortex-M4 r0p1
      INFO:pyocd.coresight.cortex_m:FPU present: FPv4-SP-D16-M
      INFO:pyocd.coresight.dwt:4 hardware watchpoints
      INFO:pyocd.coresight.fpb:6 hardware breakpoints, 4 literal comparators
      binary file: c:\pyOCD\test\data\binaries\VMM_v2_0_8.hex
      
      
      ------ GET Unique ID ------
      Unique ID: 0000000300190051044ab7bd00000c120000000097969921
      
      
      ------ TEST READ / WRITE CORE REGISTER ------
      initial pc: 0x20000000
      now pc: 0x3D82
      initial pc value rewritten: 0x20000000
      MSP = 0x20001000; PSP = 0x9e7e38ec
      CONTROL = 0x00; FAULTMASK = 0x01; BASEPRI = 0x00; PRIMASK = 0x00
      New PRIMASK = 0x01
      Restored PRIMASK = 0x00
      S0 = 1.40145e-08 (0x3270c48a)
      New S0 = 3.14159 (0x40490fdb)
      Restored S0 = 1.40145e-08 (0x3270c48a)
      
      
      ------ TEST HALT / RESUME ------
      resume
      halt
      HALT: pc: 0x20000000
      
      
      ------ TEST STEP ------
      reset and halt
      HALT: pc: 0xFFFFFFFE
      step
      STEP: pc: 0xFFFFFFFE
      step
      STEP: pc: 0xFFFFFFFE
      step
      STEP: pc: 0xFFFFFFFE
      step
      STEP: pc: 0xFFFFFFFE
      
      
      ------ TEST RANGE STEP ------
      range start = 0x20001fb8; range_end = 0x20001fca
      wrote range test step code to RAM successfully
      start PC: 0x20001FB8
      end PC: 0x20001FCA
      start PC: 0x20001FB8
      end PC: 0x20001FCA
      halt reason: DEBUG (should be BREAKPOINT)
      
      
      ------ TEST READ / WRITE MEMORY ------
      READ32/WRITE32
      write32 0x3FAEBE81 at 0x20000000
      read32 at 0x20000000: 0x3FAEBE81
      
      READ16/WRITE16
      write16 0x9A52 at 0x20000002
      read16 at 0x20000002: 0x9A52
      
      READ8/WRITE8
      write8 0xB at 0x20000001
      read8 at 0x20000001: 0xB
      
      
      ------ TEST READ / WRITE MEMORY BLOCK ------
      TEST PASSED
      
      
      ------ TEST RESET ------
      pc: 0xFFFFFFFE
      pc: 0xFFFFFFFE
      pc: 0xFFFFFFFE
      pc: 0xFFFFFFFE
      pc: 0xFFFFFFFE
      reset and halt

Thomas-Lawrence, 2022-07-10:
Hi Flit,
Any suggestions to fix the issue?

AAsyunkin-se, 2022-09-19:
GigaDevice.GD32F30x_DFP.2.2.0.pack had all their SVDs malformed - whitespace at the start of 1st line. Not sure why this is not an issue with Keil, but pyocd behaves correctly as in 'it is indeed a malformed xml'.
So fix unpack the pack, rename svds to xml and try to open in a browser. If that fails, need to fix svd and repack back to the pack.
This is how I fixed mine, your issue could be similar.
Hope this helps